probe-rs/probe-rs #1270: Can't connect to STM32WL55JC MCU

johannesneyer, 2022-10-28:
Probe-rs does not seem to work with `STM32WL55JC` MCUs. Segger tools and openocd work fine.

Using a `nucleo-WL55JC1` dev-kit on Arch Linux.

<details>
<summary>with on-board st-link</summary>

```
> RUST_LOG=debug probe-rs-cli reset --chip STM32WL55JCIx --speed 200
 DEBUG probe_rs::config::registry > Searching registry for chip with name STM32WL55JCIx
 DEBUG probe_rs::config::registry > Exact match for chip name: STM32WL55JCIx
 DEBUG probe_rs::probe::cmsisdap::tools > Searching for CMSIS-DAP probes using libusb
 DEBUG probe_rs::probe::cmsisdap::tools > Found 0 CMSIS-DAP probes using libusb, searching HID
 DEBUG probe_rs::probe::cmsisdap::tools > Found 0 CMSIS-DAP probes total
 DEBUG jaylink                          > libusb 1.0.26.11724
 DEBUG jaylink                          > libusb has capability API: true
 DEBUG jaylink                          > libusb has HID access: true
 DEBUG jaylink                          > libusb has hotplug support: true
 DEBUG jaylink                          > libusb can detach kernel driver: true
 DEBUG probe_rs::probe::cmsisdap::tools > Attempting to open 0483:374e in CMSIS-DAP v1 mode
 DEBUG probe_rs::probe::stlink::usb_interface > Acquired libusb context.
 DEBUG probe_rs::probe::stlink::usb_interface > Aquired handle for probe
 DEBUG probe_rs::probe::stlink::usb_interface > Active config descriptor: ConfigDescriptor { bLength: 9, bDescriptorType: 2, wTotalLength: 128, bNumInterfaces: 4, bConfigurationValue: 1, iConfiguration: 4, bmAttributes: 128, bMaxPower: 250, extra: [] }
 DEBUG probe_rs::probe::stlink::usb_interface > Device descriptor: DeviceDescriptor { bLength: 18, bDescriptorType: 1, bcdUSB: 512, bDeviceClass: 239, bDeviceSubClass: 2, bDeviceProtocol: 1, bMaxPacketSize: 64, idVendor: 1155, idProduct: 14158, bcdDevice: 256, iManufacturer: 1, iProduct: 2, iSerialNumber: 3, bNumConfigurations: 1 }
 DEBUG probe_rs::probe::stlink::usb_interface > Claimed interface 0 of USB device.
 DEBUG probe_rs::probe::stlink::usb_interface > Succesfully attached to STLink.
 DEBUG probe_rs::probe::stlink                > Initializing STLink...
 DEBUG probe_rs::probe::stlink                > Current device mode: MassStorage
 DEBUG probe_rs::probe::stlink                > STLink version: (3, 9)
 DEBUG probe_rs::probe::stlink                > attach(Swd)
 DEBUG probe_rs::probe::stlink                > Current device mode: MassStorage
 DEBUG probe_rs::probe::stlink                > Switching protocol to SWD
 INFO  probe_rs::probe::stlink                > Target voltage (VAPP): 3.28 V
 DEBUG probe_rs::probe::stlink                > Successfully initialized SWD.
 DEBUG probe_rs::architecture::arm::ap        > Reading register IDR
 DEBUG probe_rs::probe::stlink                > Opening AP 0
 DEBUG probe_rs::architecture::arm::ap        > Read register    IDR, value=0x24770011
 DEBUG probe_rs::architecture::arm::ap        > Reading register IDR
 DEBUG probe_rs::probe::stlink                > Opening AP 1
 DEBUG probe_rs::architecture::arm::ap        > Read register    IDR, value=0x84770001
 DEBUG probe_rs::architecture::arm::ap        > Reading register IDR
 DEBUG probe_rs::probe::stlink                > Opening AP 2
 DEBUG probe_rs::architecture::arm::ap        > Read register    IDR, value=0x0
 DEBUG probe_rs::architecture::arm::ap        > Reading register IDR
 DEBUG probe_rs::architecture::arm::ap        > Read register    IDR, value=0x24770011
 DEBUG probe_rs::architecture::arm::ap        > Reading register BASE
 DEBUG probe_rs::architecture::arm::ap        > Read register    BASE, value=0xe00ff003
 DEBUG probe_rs::architecture::arm::ap        > Reading register BASE2
 DEBUG probe_rs::architecture::arm::ap        > Read register    BASE2, value=0x0
 DEBUG probe_rs::architecture::arm::ap        > Reading register CSW
 DEBUG probe_rs::architecture::arm::ap        > Read register    CSW, value=0x23000052
 DEBUG probe_rs::architecture::arm::ap        > Writing register CSW, value=CSW { DbgSwEnable: 1, HNONSEC: 1, PROT: 6, CACHE: 3, SPIDEN: 0, _RES0: 0, MTE: 0, Type: 0, Mode: 0, TrinProg: 0, DeviceEn: 0, AddrInc: Single, _RES1: 0, SIZE: U8 }
 DEBUG probe_rs::architecture::arm::ap        > Reading register CSW
 DEBUG probe_rs::architecture::arm::ap        > Read register    CSW, value=0x23000050
 DEBUG probe_rs::architecture::arm::ap        > Writing register CSW, value=CSW { DbgSwEnable: 0, HNONSEC: 0, PROT: 2, CACHE: 3, SPIDEN: 0, _RES0: 0, MTE: 0, Type: 0, Mode: 0, TrinProg: 0, DeviceEn: 1, AddrInc: Single, _RES1: 0, SIZE: U32 }
 DEBUG probe_rs::architecture::arm::communication_interface > HNONSEC supported: false
 DEBUG probe_rs::architecture::arm::ap                      > Reading register CFG
 DEBUG probe_rs::architecture::arm::ap                      > Read register    CFG, value=0x0
 DEBUG probe_rs::probe::stlink                              > AP ApAddress {
    dp: Default,
    ap: 0x0,
}: MemoryAp(MemoryApInformation { address: ApAddress { dp: Default, ap: 0 }, only_32bit_data_size: false, debug_base_address: 3759140864, supports_hnonsec: false, has_large_address_extension: false, has_large_data_extension: false })
 DEBUG probe_rs::architecture::arm::ap                      > Reading register IDR
 DEBUG probe_rs::architecture::arm::ap                      > Read register    IDR, value=0x84770001
 DEBUG probe_rs::architecture::arm::ap                      > Reading register BASE
 DEBUG probe_rs::architecture::arm::ap                      > Read register    BASE, value=0xf0000003
 DEBUG probe_rs::architecture::arm::ap                      > Reading register BASE2
 DEBUG probe_rs::architecture::arm::ap                      > Read register    BASE2, value=0x0
 DEBUG probe_rs::architecture::arm::ap                      > Reading register CSW
 DEBUG probe_rs::architecture::arm::ap                      > Read register    CSW, value=0x43800042
[... 346 lines cut ...]
    cdbgpwrupack: true,
    cdbgpwrupreq: true,
    cdbgrstack: true,
    c_dbg_rst_req: false,
    trn_cnt: 0,
    mask_lane: 15,
    w_data_err: false,
    read_ok: false,
    sticky_err: false,
    stick_cmp: false,
    trn_mode: 0,
    sticky_orun: true,
    orun_detect: true,
}
 DEBUG probe_rs::probe::jlink::arm                          > Performing 1 transfers (0 additional transfers)
 DEBUG probe_rs::probe::jlink::arm                          > Transfer result 0: Ok(0)
Error: Connecting to the chip was unsuccessful.

Caused by:
    0: A core architecture specific error occurred
    1: Failed to read register DRW at address 0x0000000c
    2: An error specific to the selected architecture occurred
    3: Target device responded with a FAULT response to the request.
```
</details>

newAM, 2022-10-30:
What version of `probe-rs` are you using?  There have been a number of bugs in the past (ref https://github.com/stm32-rs/stm32wlxx-hal/issues/74).  All the bugs are fixed in version >= 0.12.0

Yatekii, 2022-10-30:
Could also be that software reset simply does not work and segger/openocd just do a hardware reset.

johannesneyer, 2022-10-31:
@newAM I tried `v0.12.0`, `v0.13.0` and master.

@Yatekii how do I figure that out? `--connect-under-reset` does not help.

noppej, 2022-11-07:
@johannesneyer I do not have one of these boards to validate, but I have worked with a number of other ST devices, and do not see any obvious reason why this should fail. Also, I am assuming that @newAM is still able to connect to the STM32WL55JC and that this is therefore not a universal issue with support of these boards (@newAM, any chance you can confirm?). 

Just to eliminate the possibility of strange timing errors, can you try a faster speed like 2000?

newAM, 2022-11-07:
> Also, I am assuming that @newAM is still able to connect to the STM32WL55JC and that this is therefore not a universal issue with support of these boards (@newAM, any chance you can confirm?).

I can connect using CMSIS-DAP probes (hs-probe, orbtrace) and STLinks (v2, v3).  I don't own a segger to try and help debug this issue sadly.

noppej, 2022-11-07:
@newAM I believe he is running into this with onboard STLink v3. I appreciate you confirming that you can still connect ... it helps eliminate some stuff.

newAM, 2022-11-08:
I just re-checked this, I made a bad assumption that all the `probe-rs` based tools behave the same.

With `probe-run` I can flash & use defmt (frequently used for CI in [stm32wlxx-hal](https://github.com/stm32-rs/stm32wlxx-hal)), but connecting with `probe-rs-cli` results in the same problem as @johannesneyer.

> Just to eliminate the possibility of strange timing errors, can you try a faster speed like 2000?

Same problem at various frequencies.

Let me know if you want me to run any other things for debug.

johannesneyer, 2022-11-08:
> I made a bad assumption that all the `probe-rs` based tools behave the same.

I made the same assumption because `probe-run` does not work for me as well:
<details>
<summary>probe-run output with on-board st-link</summary>

```
> probe-run -vv --speed 200 --chip STM32WL55JCIx /tmp/cargo/target/thumbv7em-none-eabihf/debug/lora-embassy-playground
(HOST) DEBUG vector table: VectorTable { initial_stack_pointer: 20003000, hard_fault: 800f4e5 }
└─ probe_run::elf @ /home/johannes/.cargo/registry/src/github.com-1ecc6299db9ec823/probe-run-0.3.4/src/elf.rs:34
(HOST) DEBUG Searching registry for chip with name STM32WL55JCIx
└─ probe_rs::config::registry @ /home/johannes/.cargo/registry/src/github.com-1ecc6299db9ec823/probe-rs-0.13.0/src/config/registry.rs:148
(HOST) DEBUG Exact match for chip name: STM32WL55JCIx
└─ probe_rs::config::registry @ /home/johannes/.cargo/registry/src/github.com-1ecc6299db9ec823/probe-rs-0.13.0/src/config/registry.rs:159
(HOST) DEBUG RAM region: 0x20000000-0x2000FFFF
└─ probe_run::target_info @ /home/johannes/.cargo/registry/src/github.com-1ecc6299db9ec823/probe-run-0.3.4/src/target_info.rs:115
(HOST) DEBUG section `.data` is in RAM at 0x20000000..=0x20000047
└─ probe_run::target_info @ /home/johannes/.cargo/registry/src/github.com-1ecc6299db9ec823/probe-run-0.3.4/src/target_info.rs:153
(HOST) DEBUG section `.bss` is in RAM at 0x20000048..=0x200001CB
└─ probe_run::target_info @ /home/johannes/.cargo/registry/src/github.com-1ecc6299db9ec823/probe-run-0.3.4/src/target_info.rs:153
(HOST) DEBUG valid SP range: 0x200001CC..=0x20002FFC
└─ probe_run::target_info @ /home/johannes/.cargo/registry/src/github.com-1ecc6299db9ec823/probe-run-0.3.4/src/target_info.rs:167
(HOST) DEBUG Searching for CMSIS-DAP probes using libusb
└─ probe_rs::probe::cmsisdap::tools @ /home/johannes/.cargo/registry/src/github.com-1ecc6299db9ec823/probe-rs-0.13.0/src/probe/cmsisdap/tools.rs:16
(HOST) DEBUG Found 0 CMSIS-DAP probes using libusb, searching HID
└─ probe_rs::probe::cmsisdap::tools @ /home/johannes/.cargo/registry/src/github.com-1ecc6299db9ec823/probe-rs-0.13.0/src/probe/cmsisdap/tools.rs:25
(HOST) DEBUG Found 0 CMSIS-DAP probes total
└─ probe_rs::probe::cmsisdap::tools @ /home/johannes/.cargo/registry/src/github.com-1ecc6299db9ec823/probe-rs-0.13.0/src/probe/cmsisdap/tools.rs:46
(HOST) DEBUG libusb 1.0.26.11724
└─ jaylink @ /home/johannes/.cargo/registry/src/github.com-1ecc6299db9ec823/jaylink-0.3.0/src/lib.rs:1557
(HOST) DEBUG libusb has capability API: true
└─ jaylink @ /home/johannes/.cargo/registry/src/github.com-1ecc6299db9ec823/jaylink-0.3.0/src/lib.rs:1566
(HOST) DEBUG libusb has HID access: true
└─ jaylink @ /home/johannes/.cargo/registry/src/github.com-1ecc6299db9ec823/jaylink-0.3.0/src/lib.rs:1567
(HOST) DEBUG libusb has hotplug support: true
└─ jaylink @ /home/johannes/.cargo/registry/src/github.com-1ecc6299db9ec823/jaylink-0.3.0/src/lib.rs:1568
(HOST) DEBUG libusb can detach kernel driver: true
└─ jaylink @ /home/johannes/.cargo/registry/src/github.com-1ecc6299db9ec823/jaylink-0.3.0/src/lib.rs:1569
(HOST) DEBUG found 1 probes
└─ probe_run::probe @ /home/johannes/.cargo/registry/src/github.com-1ecc6299db9ec823/probe-run-0.3.4/src/probe.rs:25
(HOST) TRACE Attempting to open device matching 0483:374e:0026001D5553500820393256
└─ probe_rs::probe::cmsisdap::tools @ /home/johannes/.cargo/registry/src/github.com-1ecc6299db9ec823/probe-rs-0.13.0/src/probe/cmsisdap/tools.rs:257
(HOST) TRACE Trying device Bus 004 Device 001: ID 1d6b:0003
└─ probe_rs::probe::cmsisdap::tools @ /home/johannes/.cargo/registry/src/github.com-1ecc6299db9ec823/probe-rs-0.13.0/src/probe/cmsisdap/tools.rs:273
(HOST) TRACE Error opening: Access
└─ probe_rs::probe::cmsisdap::tools @ /home/johannes/.cargo/registry/src/github.com-1ecc6299db9ec823/probe-rs-0.13.0/src/probe/cmsisdap/tools.rs:286
(HOST) TRACE Trying device Bus 003 Device 003: ID 0d8c:0014
└─ probe_rs::probe::cmsisdap::tools @ /home/johannes/.cargo/registry/src/github.com-1ecc6299db9ec823/probe-rs-0.13.0/src/probe/cmsisdap/tools.rs:273
(HOST) TRACE Error opening: Access
└─ probe_rs::probe::cmsisdap::tools @ /home/johannes/.cargo/registry/src/github.com-1ecc6299db9ec823/probe-rs-0.13.0/src/probe/cmsisdap/tools.rs:286
(HOST) TRACE Trying device Bus 003 Device 004: ID 0483:374e
└─ probe_rs::probe::cmsisdap::tools @ /home/johannes/.cargo/registry/src/github.com-1ecc6299db9ec823/probe-rs-0.13.0/src/probe/cmsisdap/tools.rs:273
(HOST) TRACE Trying device Bus 003 Device 002: ID feed:6061
└─ probe_rs::probe::cmsisdap::tools @ /home/johannes/.cargo/registry/src/github.com-1ecc6299db9ec823/probe-rs-0.13.0/src/probe/cmsisdap/tools.rs:273
(HOST) TRACE Error opening: Access
└─ probe_rs::probe::cmsisdap::tools @ /home/johannes/.cargo/registry/src/github.com-1ecc6299db9ec823/probe-rs-0.13.0/src/probe/cmsisdap/tools.rs:286
(HOST) TRACE Trying device Bus 003 Device 001: ID 1d6b:0002
└─ probe_rs::probe::cmsisdap::tools @ /home/johannes/.cargo/registry/src/github.com-1ecc6299db9ec823/probe-rs-0.13.0/src/probe/cmsisdap/tools.rs:273
(HOST) TRACE Error opening: Access
└─ probe_rs::probe::cmsisdap::tools @ /home/johannes/.cargo/registry/src/github.com-1ecc6299db9ec823/probe-rs-0.13.0/src/probe/cmsisdap/tools.rs:286
(HOST) TRACE Trying device Bus 002 Device 001: ID 1d6b:0003
└─ probe_rs::probe::cmsisdap::tools @ /home/johannes/.cargo/registry/src/github.com-1ecc6299db9ec823/probe-rs-0.13.0/src/probe/cmsisdap/tools.rs:273
(HOST) TRACE Error opening: Access
[... cut ...]
└─ probe_rs::probe::stlink::usb_interface @ /home/johannes/.cargo/registry/src/github.com-1ecc6299db9ec823/probe-rs-0.13.0/src/probe/stlink/usb_interface.rs:282
(HOST) DEBUG Current device mode: Jtag
└─ probe_rs::probe::stlink @ /home/johannes/.cargo/registry/src/github.com-1ecc6299db9ec823/probe-rs-0.13.0/src/probe/stlink/mod.rs:379
Error: An error with the usage of the probe occurred

Caused by:
    0: An error specific to a probe type occurred
    1: Command failed with status SwdApFault
```
</details>

@newAM am I doing something wrong?

newAM, 2022-11-09:
Oh!  I know the problem.

The STM32WL55 is a dual-core target and likely uses commands for multi-drop targets which are still in development (I think, it has been a while since I read the code).

I have been using the single core chip definition, `--chip STM32WLE5JCIx` for all my testing, even if I am using the dual-core`STM32WL55JCIx`.

If I change my probe-run command to use `STM32WL55JCIx` I get the same behavior as you.

johannesneyer, 2022-11-09:
Ah I see. You are right `STM32WLE5JCIx` works. Now I can finally use defmt :grinning: 

Can we add a note about this somewhere? I wasn't even aware that there were single-core chip definitions for the dual-core chips.

Tiwalun, 2022-11-09:
> The STM32WL55 is a dual-core target and likely uses commands for multi-drop targets which are still in development (I think, it has been a while since I read the code).

Most dual core targets don't need multi-drop, multi-drop is usually only used if you have multiple physical chips, but only a single debug connection. 

The Raspberry Pi Pico uses it for multi-core, but that's a bit weird. AFAIK it's the only chip where the probe-rs mult-drop support is used so far.

ceekdee, 2022-12-09:
Attempt to deploy an Embassy blinky executable to a STM32WL55 (Nucleo-64) board from USB on a Linux PC (using STM32WLE5JCIx to avoid multi-core issues):

probe-run --chip STM32WLE5JCIx target/thumbv7em-none-eabihf/debug/blinky

Error: An error with the usage of the probe occurred

Caused by:
    0: An error specific to a probe type occurred
    1: Command failed with status JtagGetIdcodeError
 
 RUST_LOG=probe_rs=debug probe-rs-cli info
 DEBUG probe_rs::probe::cmsisdap::tools > Searching for CMSIS-DAP probes using libusb
 DEBUG probe_rs::probe::cmsisdap::tools > Found 0 CMSIS-DAP probes using libusb, searching HID
 DEBUG probe_rs::probe::cmsisdap::tools > Found 0 CMSIS-DAP probes total
 DEBUG probe_rs::probe::cmsisdap::tools > Attempting to open 0483:374e in CMSIS-DAP v1 mode
 DEBUG probe_rs::probe::stlink::usb_interface > Acquired libusb context.
 DEBUG probe_rs::probe::stlink::usb_interface > Aquired handle for probe
 DEBUG probe_rs::probe::stlink::usb_interface > Active config descriptor: ConfigDescriptor { bLength: 9, bDescriptorType: 2, wTotalLength: 128, bNumInterfaces: 4, bConfigurationValue: 1, iConfiguration: 4, bmAttributes: 128, bMaxPower: 250, extra: [] }
 DEBUG probe_rs::probe::stlink::usb_interface > Device descriptor: DeviceDescriptor { bLength: 18, bDescriptorType: 1, bcdUSB: 512, bDeviceClass: 239, bDeviceSubClass: 2, bDeviceProtocol: 1, bMaxPacketSize: 64, idVendor: 1155, idProduct: 14158, bcdDevice: 256, iManufacturer: 1, iProduct: 2, iSerialNumber: 3, bNumConfigurations: 1 }
 DEBUG probe_rs::probe::stlink::usb_interface > Claimed interface 0 of USB device.
 DEBUG probe_rs::probe::stlink::usb_interface > Succesfully attached to STLink.
 DEBUG probe_rs::probe::stlink                > Initializing STLink...
 DEBUG probe_rs::probe::stlink                > Current device mode: MassStorage
 DEBUG probe_rs::probe::stlink                > STLink version: (3, 7)
 DEBUG probe_rs::probe::stlink                > attach(Jtag)
 DEBUG probe_rs::probe::stlink                > Current device mode: MassStorage
 DEBUG probe_rs::probe::stlink                > Switching protocol to JTAG
 INFO  probe_rs::probe::stlink                > Target voltage (VAPP): 3.26 V
 WARN  probe_rs::probe::stlink                > send_jtag_command 242 failed: JtagGetIdcodeError
 DEBUG probe_rs::probe::stlink                > Detaching from STLink.
 DEBUG probe_rs::probe::stlink                > Current device mode: MassStorage
 WARN  probe_rs_cli::info                     > Error identifying target using protocol JTAG: An error with the usage of the probe occurred
 DEBUG probe_rs::probe::stlink                > attach(Swd)
 DEBUG probe_rs::probe::stlink                > Current device mode: MassStorage
 DEBUG probe_rs::probe::stlink                > Switching protocol to SWD
 INFO  probe_rs::probe::stlink                > Target voltage (VAPP): 3.26 V
 WARN  probe_rs::probe::stlink                > send_jtag_command 242 failed: JtagGetIdcodeError
 DEBUG probe_rs::probe::stlink                > Detaching from STLink.
 DEBUG probe_rs::probe::stlink                > Current device mode: MassStorage
 WARN  probe_rs_cli::info                     > Error identifying target using protocol SWD: An error with the usage of the probe occurred
 DEBUG probe_rs::probe::stlink                > Current device mode: MassStorage

ceekdee, 2022-12-09:
The aforementioned issue was addressed by --connect-under-reset

probe-run --chip STM32WLE5JCIx --connect-under-reset target/thumbv7em-none-eabihf/debug/blinky.

This was identified by reading through other JtagGetIdcodeError issues.

Tiwalun, 2023-01-18:
@johannesneyer, could you please try this again? We just got a bug fixed regarding DP wait errors on the ST-Link, so this should work better now. (cc #1406)

johannesneyer, 2023-01-20:
> @johannesneyer, could you please try this again?

Does not work for me on current master:

<details>
<summary>RUST_LOG=debug ./target/release/probe-rs-cli reset --chip STM32WL55JCIx</summary>

```
 INFO probe_rs_cli: Writing log to "/home/neye/.local/share/probe-rs/1674228041567.log"
DEBUG probe_rs::config::registry: Searching registry for chip with name STM32WL55JCIx
DEBUG probe_rs::config::registry: Exact match for chip name: STM32WL55JCIx
DEBUG list_all:list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: Searching for CMSIS-DAP probes using libusb
DEBUG list_all:list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: Found 1 CMSIS-DAP probes using libusb, searching HID
DEBUG list_all:list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: Found 1 CMSIS-DAP probes total
DEBUG list_all:list_jlink_devices: jaylink: libusb 1.0.26.11724
DEBUG list_all:list_jlink_devices: jaylink: libusb has capability API: true
DEBUG list_all:list_jlink_devices: jaylink: libusb has HID access: true
DEBUG list_all:list_jlink_devices: jaylink: libusb has hotplug support: true
DEBUG list_all:list_jlink_devices: jaylink: libusb can detach kernel driver: true
DEBUG open: probe_rs::probe::cmsisdap::tools: Could not open 0d28:0204 in CMSIS-DAP v2 mode
DEBUG open: probe_rs::probe::cmsisdap::tools: Attempting to open 0d28:0204 in CMSIS-DAP v1 mode
DEBUG open: probe_rs::probe::cmsisdap::commands: Draining probe of any pending data.
DEBUG open: probe_rs::probe::cmsisdap::commands: Attempt 1 to find packet size
DEBUG open: probe_rs::probe::cmsisdap::commands: Success: packet size is 64
DEBUG open: probe_rs::probe::cmsisdap::commands: Configuring probe to use packet size 64
DEBUG open: probe_rs::probe::cmsisdap: Detected probe capabilities: Capabilities { _swd_implemented: true, _jtag_implemented: false, swo_uart_implemented: false, swo_manchester_implemented: false, _atomic_commands_implemented: true, _test_domain_timer_implemented: true, swo_streaming_trace_implemented: false, _uart_communication_port_implemented: false, uart_com_port_implemented: false }
DEBUG attach: probe_rs::probe::cmsisdap: Attaching to target system (clock = 1000kHz)
 INFO attach: probe_rs::probe::cmsisdap: Using protocol SWD
DEBUG debug_port_setup: probe_rs::probe::cmsisdap: Adding command to batch: Read(port=DebugPort, addr=0)
DEBUG debug_port_setup: probe_rs::probe::cmsisdap: 1 items in batch
DEBUG debug_port_setup: probe_rs::probe::cmsisdap: Attempting batch of 1 items
DEBUG debug_port_setup: probe_rs::probe::cmsisdap: 1 of batch of 1 items suceeded
DEBUG debug_device_unlock: probe_rs::architecture::arm::sequences: debug_device_unlock - empty by default
DEBUG probe_rs::architecture::arm::communication_interface: Selecting DP Default
DEBUG debug_port_start: probe_rs::architecture::arm::dp: Writing DP register ABORT, value=0x0000001e
DEBUG debug_port_start: probe_rs::probe::cmsisdap: Adding command to batch: Write(port=DebugPort, addr=0, data=0x0000001e
DEBUG debug_port_start: probe_rs::architecture::arm::dp: Writing DP register SELECT, value=0x00000000
DEBUG debug_port_start: probe_rs::probe::cmsisdap: Adding command to batch: Write(port=DebugPort, addr=8, data=0x00000000
DEBUG debug_port_start: probe_rs::architecture::arm::dp: Reading DP register CTRL/STAT
DEBUG debug_port_start: probe_rs::probe::cmsisdap: Adding command to batch: Read(port=DebugPort, addr=4)
DEBUG debug_port_start: probe_rs::probe::cmsisdap: 3 items in batch
DEBUG debug_port_start: probe_rs::probe::cmsisdap: Attempting batch of 3 items
DEBUG debug_port_start: probe_rs::probe::cmsisdap: 3 of batch of 3 items suceeded
DEBUG debug_port_start: probe_rs::architecture::arm::dp: Read    DP register CTRL/STAT, value=0x08000000
DEBUG debug_port_start: probe_rs::architecture::arm::dp: Writing DP register CTRL/STAT, value=0x50000000
DEBUG debug_port_start: probe_rs::probe::cmsisdap: Adding command to batch: Write(port=DebugPort, addr=4, data=0x50000000
DEBUG debug_port_start: probe_rs::architecture::arm::dp: Reading DP register CTRL/STAT
DEBUG debug_port_start: probe_rs::probe::cmsisdap: Adding command to batch: Read(port=DebugPort, addr=4)
DEBUG debug_port_start: probe_rs::probe::cmsisdap: 2 items in batch
DEBUG debug_port_start: probe_rs::probe::cmsisdap: Attempting batch of 2 items
DEBUG debug_port_start: probe_rs::probe::cmsisdap: 2 of batch of 2 items suceeded
DEBUG debug_port_start: probe_rs::architecture::arm::dp: Read    DP register CTRL/STAT, value=0xf8000000
DEBUG debug_port_start: probe_rs::architecture::arm::dp: Writing DP register CTRL/STAT, value=0x50000f00
DEBUG debug_port_start: probe_rs::probe::cmsisdap: Adding command to batch: Write(port=DebugPort, addr=4, data=0x50000f00
DEBUG debug_port_start: probe_rs::architecture::arm::dp: Reading DP register CTRL/STAT
DEBUG debug_port_start: probe_rs::probe::cmsisdap: Adding command to batch: Read(port=DebugPort, addr=4)
DEBUG debug_port_start: probe_rs::probe::cmsisdap: 2 items in batch
DEBUG debug_port_start: probe_rs::probe::cmsisdap: Attempting batch of 2 items
DEBUG debug_port_start: probe_rs::probe::cmsisdap: 2 of batch of 2 items suceeded
DEBUG debug_port_start: probe_rs::architecture::arm::dp: Read    DP register CTRL/STAT, value=0xf8000f00
DEBUG probe_rs::architecture::arm::communication_interface: Setting orun_detect: false
DEBUG probe_rs::architecture::arm::dp: Reading DP register CTRL/STAT
DEBUG probe_rs::probe::cmsisdap: Adding command to batch: Read(port=DebugPort, addr=4)
[... cut ...]
DEBUG debug_core_start: probe_rs::probe::cmsisdap: 6 of batch of 6 items suceeded
DEBUG probe_rs::probe::cmsisdap: Detaching from CMSIS-DAP probe
 INFO probe_rs_cli: Wrote log to "/home/neye/.local/share/probe-rs/1674228041567.log"
Error: Connecting to the chip was unsuccessful.

Caused by:
    0: A ARM specific error occured.
    1: Error using access port
    2: Failed to read register DRW at address 0x0000000c
    3: An error occured in the communication with an access port or debug port.
    4: Target device responded with a WAIT response to the request.
```
</details>

Tiwalun, 2023-01-20:
Thanks for checking again. But this seems to be with a CMSIS-DAP probe, right? I don't expect the ST-Link fixes to help there.

And please use the trace log level, with only debug the log unfortunately doesn't tell enough to figure out what's going wrong.

johannesneyer, 2023-01-24:
Ah, right. ST-Link does not work as well:

<details>
<summary>RUST_LOG=trace ./target/release/probe-rs-cli reset --chip STM32WL55JCIx</summary>

```
INFO probe_rs_cli: Writing log to "/home/neye/.local/share/probe-rs/1674570395701.log"
DEBUG probe_rs::config::registry: Searching registry for chip with name STM32WL55JCIx
DEBUG probe_rs::config::registry: Exact match for chip name: STM32WL55JCIx
DEBUG list_all:list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: Searching for CMSIS-DAP probes using libusb
DEBUG list_all:list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: Found 0 CMSIS-DAP probes using libusb, searching HID
DEBUG list_all:list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: Found 0 CMSIS-DAP probes total
DEBUG list_all:list_jlink_devices: jaylink: libusb 1.0.26.11724
DEBUG list_all:list_jlink_devices: jaylink: libusb has capability API: true
DEBUG list_all:list_jlink_devices: jaylink: libusb has HID access: true
DEBUG list_all:list_jlink_devices: jaylink: libusb has hotplug support: true
DEBUG list_all:list_jlink_devices: jaylink: libusb can detach kernel driver: true
TRACE open: probe_rs::probe::cmsisdap::tools: Attempting to open device matching 0483:374e:0023001A4741500320383733
TRACE open: probe_rs::probe::cmsisdap::tools: Trying device Bus 002 Device 005: ID 0451:8140
TRACE open: probe_rs::probe::cmsisdap::tools: Error opening: Access
TRACE open: probe_rs::probe::cmsisdap::tools: Trying device Bus 002 Device 003: ID 0451:8140
TRACE open: probe_rs::probe::cmsisdap::tools: Error opening: Access
TRACE open: probe_rs::probe::cmsisdap::tools: Trying device Bus 002 Device 004: ID 0451:8140
TRACE open: probe_rs::probe::cmsisdap::tools: Error opening: Access
TRACE open: probe_rs::probe::cmsisdap::tools: Trying device Bus 002 Device 002: ID 0451:8140
TRACE open: probe_rs::probe::cmsisdap::tools: Error opening: Access
TRACE open: probe_rs::probe::cmsisdap::tools: Trying device Bus 002 Device 001: ID 1d6b:0003
TRACE open: probe_rs::probe::cmsisdap::tools: Error opening: Access
TRACE open: probe_rs::probe::cmsisdap::tools: Trying device Bus 001 Device 012: ID 095d:3001
TRACE open: probe_rs::probe::cmsisdap::tools: Error opening: Access
TRACE open: probe_rs::probe::cmsisdap::tools: Trying device Bus 001 Device 009: ID 0451:8142
TRACE open: probe_rs::probe::cmsisdap::tools: Error opening: Access
TRACE open: probe_rs::probe::cmsisdap::tools: Trying device Bus 001 Device 005: ID 0451:8142
TRACE open: probe_rs::probe::cmsisdap::tools: Error opening: Access
TRACE open: probe_rs::probe::cmsisdap::tools: Trying device Bus 001 Device 011: ID 2357:0604
TRACE open: probe_rs::probe::cmsisdap::tools: Error opening: Access
TRACE open: probe_rs::probe::cmsisdap::tools: Trying device Bus 001 Device 008: ID 0451:8142
TRACE open: probe_rs::probe::cmsisdap::tools: Error opening: Access
TRACE open: probe_rs::probe::cmsisdap::tools: Trying device Bus 001 Device 018: ID 0483:374e
TRACE open: probe_rs::probe::cmsisdap::tools: Trying device Bus 001 Device 006: ID 0b0e:0301
TRACE open: probe_rs::probe::cmsisdap::tools: Error opening: Access
TRACE open: probe_rs::probe::cmsisdap::tools: Trying device Bus 001 Device 004: ID 0451:8142
TRACE open: probe_rs::probe::cmsisdap::tools: Error opening: Access
TRACE open: probe_rs::probe::cmsisdap::tools: Trying device Bus 001 Device 003: ID 444d:3536
TRACE open: probe_rs::probe::cmsisdap::tools: Error opening: Access
TRACE open: probe_rs::probe::cmsisdap::tools: Trying device Bus 001 Device 002: ID 046d:c52b
TRACE open: probe_rs::probe::cmsisdap::tools: Error opening: Access
TRACE open: probe_rs::probe::cmsisdap::tools: Trying device Bus 001 Device 015: ID 0835:8502
TRACE open: probe_rs::probe::cmsisdap::tools: Error opening: Access
TRACE open: probe_rs::probe::cmsisdap::tools: Trying device Bus 001 Device 014: ID 0835:8500
TRACE open: probe_rs::probe::cmsisdap::tools: Error opening: Access
TRACE open: probe_rs::probe::cmsisdap::tools: Trying device Bus 001 Device 013: ID 0835:8501
TRACE open: probe_rs::probe::cmsisdap::tools: Error opening: Access
TRACE open: probe_rs::probe::cmsisdap::tools: Trying device Bus 001 Device 010: ID 0835:8500
TRACE open: probe_rs::probe::cmsisdap::tools: Error opening: Access
TRACE open: probe_rs::probe::cmsisdap::tools: Trying device Bus 001 Device 007: ID 0b95:772b
TRACE open: probe_rs::probe::cmsisdap::tools: Error opening: Access
TRACE open: probe_rs::probe::cmsisdap::tools: Trying device Bus 001 Device 001: ID 1d6b:0002
TRACE open: probe_rs::probe::cmsisdap::tools: Error opening: Access
DEBUG open: probe_rs::probe::cmsisdap::tools: Attempting to open 0483:374e in CMSIS-DAP v1 mode
DEBUG open: probe_rs::probe::stlink::usb_interface: Acquired libusb context.
DEBUG open: probe_rs::probe::stlink::usb_interface: Aquired handle for probe
DEBUG open: probe_rs::probe::stlink::usb_interface: Active config descriptor: ConfigDescriptor { bLength: 9, bDescriptorType: 2, wTotalLength: 128, bNumInterfaces: 4, bConfigurationValue: 1, iConfiguration: 4, bmAttributes: 128, bMaxPower: 250, extra: [] }
DEBUG open: probe_rs::probe::stlink::usb_interface: Device descriptor: DeviceDescriptor { bLength: 18, bDescriptorType: 1, bcdUSB: 512, bDeviceClass: 239, bDeviceSubClass: 2, bDeviceProtocol: 1, bMaxPacketSize: 64, idVendor: 1155, idProduct: 14158, bcdDevice: 256, iManufacturer: 1, iProduct: 2, iSerialNumber: 3, bNumConfigurations: 1 }
DEBUG open: probe_rs::probe::stlink::usb_interface: Claimed interface 0 of USB device.
[... cut ...]
TRACE probe_rs::probe::stlink::usb_interface: Sending command [f5] to STLink, timeout: 1s
TRACE probe_rs::probe::stlink::usb_interface: Read 2 bytes, 0 bytes remaining
DEBUG probe_rs::probe::stlink: Current device mode: Jtag
 INFO probe_rs_cli: Wrote log to "/home/neye/.local/share/probe-rs/1674570395701.log"
Error: Connecting to the chip was unsuccessful.

Caused by:
    0: A ARM specific error occured.
    1: The debug probe encountered an error.
    2: An error specific to a probe type occurred
    3: Command failed with status SwdApWdataError
```
</details>

WesleyAC, 2023-08-04:
Any ideas on this? I am seeing this with the embedded STLink on the NUCLEO-WL55JC dev board. Happy to provide logs and try out patches if anyone has ideas.

<details><summary>RUST_LOG=trace ./target/debug/probe-rs reset --chip STM32WL55JCIx # current git 6e883249</summary>
<pre>
 INFO probe_rs: Writing log to "/home/wesleyac/.local/share/probe-rs/1691117466291.log"
DEBUG probe_rs::config::registry: Searching registry for chip with name STM32WL55JCIx
DEBUG probe_rs::config::registry: Exact match for chip name: STM32WL55JCIx
 WARN probe_rs::config::target: Using custom sequence for ARMv7 STM32WL55JCIx
DEBUG list_all:list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: Searching for CMSIS-DAP probes using libusb
DEBUG list_all:list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: Found 0 CMSIS-DAP probes using libusb, searching HID
DEBUG list_all:list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: Found 0 CMSIS-DAP probes total
DEBUG list_all:list_jlink_devices: jaylink: libusb 1.0.26.11724    
DEBUG list_all:list_jlink_devices: jaylink: libusb has capability API: true    
DEBUG list_all:list_jlink_devices: jaylink: libusb has HID access: true    
DEBUG list_all:list_jlink_devices: jaylink: libusb has hotplug support: true    
DEBUG list_all:list_jlink_devices: jaylink: libusb can detach kernel driver: true    
TRACE open: probe_rs::probe::cmsisdap::tools: Attempting to open device matching 0483:374e:0040001E4741500320383733
TRACE open: probe_rs::probe::cmsisdap::tools: Trying device Bus 002 Device 001: ID 1d6b:0003
TRACE open: probe_rs::probe::cmsisdap::tools: Error opening: Access
TRACE open: probe_rs::probe::cmsisdap::tools: Trying device Bus 001 Device 042: ID 138a:0097
TRACE open: probe_rs::probe::cmsisdap::tools: Error opening: Access
TRACE open: probe_rs::probe::cmsisdap::tools: Trying device Bus 001 Device 004: ID 8087:0a2b
TRACE open: probe_rs::probe::cmsisdap::tools: Error opening: Access
TRACE open: probe_rs::probe::cmsisdap::tools: Trying device Bus 001 Device 034: ID 0483:374e
TRACE open: probe_rs::probe::cmsisdap::tools: Trying device Bus 001 Device 001: ID 1d6b:0002
TRACE open: probe_rs::probe::cmsisdap::tools: Error opening: Access
DEBUG open: probe_rs::probe::cmsisdap::tools: Attempting to open 0483:374e in CMSIS-DAP v1 mode
DEBUG open: probe_rs::probe::stlink::usb_interface: Acquired libusb context.
DEBUG open: probe_rs::probe::stlink::usb_interface: Aquired handle for probe
DEBUG open: probe_rs::probe::stlink::usb_interface: Active config descriptor: ConfigDescriptor { bLength: 9, bDescriptorType: 2, wTotalLength: 128, bNumInterfaces: 4, bConfigurationValue: 1, iConfiguration: 4, bmAttributes: 128, bMaxPower: 250, extra: [] }
DEBUG open: probe_rs::probe::stlink::usb_interface: Device descriptor: DeviceDescriptor { bLength: 18, bDescriptorType: 1, bcdUSB: 512, bDeviceClass: 239, bDeviceSubClass: 2, bDeviceProtocol: 1, bMaxPacketSize: 64, idVendor: 1155, idProduct: 14158, bcdDevice: 256, iManufacturer: 1, iProduct: 2, iSerialNumber: 3, bNumConfigurations: 1 }
DEBUG open: probe_rs::probe::stlink::usb_interface: Claimed interface 0 of USB device.
DEBUG open: probe_rs::probe::stlink::usb_interface: Succesfully attached to STLink.
DEBUG open: probe_rs::probe::stlink: Initializing STLink...
TRACE open: probe_rs::probe::stlink: Getting current mode of device...
TRACE open: probe_rs::probe::stlink::usb_interface: Sending command [f5] to STLink, timeout: 1s
TRACE open: probe_rs::probe::stlink::usb_interface: Read 2 bytes, 0 bytes remaining
DEBUG open: probe_rs::probe::stlink: Current device mode: MassStorage
TRACE open: probe_rs::probe::stlink::usb_interface: Sending command [f1] to STLink, timeout: 1s
TRACE open: probe_rs::probe::stlink::usb_interface: Read 6 bytes, 0 bytes remaining
TRACE open: probe_rs::probe::stlink::usb_interface: Sending command [fb] to STLink, timeout: 1s
TRACE open: probe_rs::probe::stlink::usb_interface: Read 12 bytes, 0 bytes remaining
DEBUG open: probe_rs::probe::stlink: STLink version: (3, 7)
TRACE open: probe_rs::probe::stlink::usb_interface: Sending command [f2, 62, 0] to STLink, timeout: 1s
TRACE open: probe_rs::probe::stlink::usb_interface: Read 52 bytes, 0 bytes remaining
TRACE open: probe_rs::probe::stlink::usb_interface: Sending command [f2, 62, 1] to STLink, timeout: 1s
TRACE open: probe_rs::probe::stlink::usb_interface: Read 52 bytes, 0 bytes remaining
TRACE attach: probe_rs::probe::stlink: Getting current mode of device...
TRACE attach: probe_rs::probe::stlink::usb_interface: Sending command [f5] to STLink, timeout: 1s
TRACE attach: probe_rs::probe::stlink::usb_interface: Read 2 bytes, 0 bytes remaining
DEBUG attach: probe_rs::probe::stlink: Current device mode: MassStorage
DEBUG attach: probe_rs::probe::stlink: Switching protocol to SWD
TRACE attach: probe_rs::probe::stlink::usb_interface: Sending command [f7] to STLink, timeout: 1s
TRACE attach: probe_rs::probe::stlink::usb_interface: Read 8 bytes, 0 bytes remaining
 INFO attach: probe_rs::probe::stlink: Target voltage (VAPP): 3.28 V
TRACE attach: probe_rs::probe::stlink::usb_interface: Sending command [f2, 30, a3, 0] to STLink, timeout: 1s
TRACE attach: probe_rs::probe::stlink::usb_interface: Read 2 bytes, 0 bytes remaining
 WARN attach: probe_rs::probe::stlink: send_jtag_command 242 failed: JtagGetIdcodeError
TRACE probe_rs::probe::stlink: Getting current mode of device...
TRACE probe_rs::probe::stlink::usb_interface: Sending command [f5] to STLink, timeout: 1s
TRACE probe_rs::probe::stlink::usb_interface: Read 2 bytes, 0 bytes remaining
DEBUG probe_rs::probe::stlink: Current device mode: MassStorage
 INFO probe_rs: Wrote log to "/home/wesleyac/.local/share/probe-rs/1691117466291.log"
Error: Connecting to the chip was unsuccessful.

Caused by:
    0: An error with the usage of the probe occurred
    1: An error specific to a probe type occurred
    2: Command failed with status JtagGetIdcodeError
</pre>
</details>

Yatekii, 2023-08-04:
I think this might be caused when the chip is locked. Also, what's really odd is that we poll the device mode so many times.
Other than that I don't know atm :/

Yatekii, 2023-08-04:
It would be interesting what another probe has to say. If this is a STLink issue or if something is fishy with the chip.

newAM, 2023-08-04:
I'm using probe-run in CI with the embedded STLink on the nucleo boards for the stm32wlxx-hal, still works on v0.3.9.  I can try your specific command this weekend if I remember.

Yatekii, 2023-08-04:
That would be awesome, thank you!

newAM, 2023-08-05:
I was able to reproduce @WesleyAC 's problem.  It only occurs when using the `STM32WL55JCIx` target (multi-core variant), it works with the `STM32WLE5JCIx` target (single core), even though the nucleo board has the multi-core variant.  Same as the original issue then.  I haven't been keeping up with probe-rs, but I think multidrop targets are still a work-in-progress?

Yatekii, 2023-08-05:
Ah that makes sense!! We never put the correct IDCODE in the target file to connect to multidrop :) we only did that for the rp2040 I think :) but normal multicore should not require multidrop. only the peculiar rp2040 does I thought :/

Dirbaio, 2023-10-20:
I'm also running into this issue.

wl55 doesn't require multidrop, it has the standard layout of 1 DP with 2 APs, so the target yaml is correct. I think the issue is probe-rs is trying to access the 2nd core while it is "off". If you remove the 2nd core from the target yaml, probe-rs works flawlessly for debugging the 1st core.

I'm not sure what the right fix is. I think probe-rs in general should be more "resilient" to one of the cores being offline. Keep trying to attach periodically, but still allow the other core to be used. nrf53 has the same problemL on reset only the app core boots, the net core is held in reset, so probe-rs fails.

Yatekii, 2023-10-20:
I think @Tiwalun is working on an MR that should improve this.

TheNeikos, 2024-05-13:
I am also running into this 

Is there a way I could help getting this to work?

Running: 
```
❯ RUST_LOG=trace probe-rs reset --chip STM32WLE5JCIx --connect-under-reset
```

Has this output currently:

```
 INFO probe_rs::config::target: Using sequence Arm(Stm32Armv7)
 INFO probe_rs::session: Custom reset sequences are not supported on ST-Link V3.
 INFO probe_rs::session: Falling back to standard probe reset.
```

Which might be the cause of this problem?

```
❯ probe-rs --version
probe-rs 0.22.0
git commit: crates.io
```

Frostie314159, 2025-06-04:
Hi, are there any updates on how this could be fixed?

xobs, 2025-06-23:
Is this still an issue? I've managed to get it to work, and with #3417 flashing works again

Frostie314159, 2025-06-24:
I'll test it again today.

Frostie314159, 2025-06-24:
@xbos I can also confirm, this works.

xobs, 2025-10-29:
Is this on a Mac by any chance? I’ve seen issues with stlink parts when connected directly to a Mac. Going through a hub works.