probe-rs/probe-rs #1271: STM32G0B1xC / stm32g0b1cc missing

brandonros, 2022-11-01:
Not sure what the closest chip I can try is

brandonros, 2022-11-01:
[image]

brandonros, 2022-11-01:
looks like it would go here https://github.com/probe-rs/probe-rs/blob/master/probe-rs/targets/STM32G0_Series.yaml 

https://www.mouser.com/ProductDetail/STMicroelectronics/STM32G0B1CCT6?qs=CiayqK2gdcI5UzsaJZ9afA%3D%3D

brandonros, 2022-11-01:
[image]

brandonros, 2022-11-01:
```
STM32G0B1xB
STM32G0B1CB, STM32G0B1KB,
 STM32G0B1MB, STM32G0B1RB,
 STM32G0B1VB

STM32G0B1xC
STM32G0B1CC, STM32G0B1KC,
 STM32G0B1MC, STM32G0B1RC,
STM32G0B1VC

STM32G0B1xE
STM32G0B1CE, STM32G0B1KE,
 STM32G0B1ME, STM32G0B1NE,
 STM32G0B1RE, STM32G0B1VE
```

noppej, 2022-11-01:
Hi. This was actually a work in progress ( PR #1266 ) and is now merged into master. Can you please update from master and test again?

brandonros, 2022-11-02:
would be cool if a release could be done to cargo crates but otherwise got it working from git

thank you so much

closed 2022-11-02 by brandonros