probe-rs/probe-rs #1282: Cannot install the probe-rs-debugger server component

tsadowski, 2022-11-16:
**Describe the bug**
When I try to install the probe-rs-debugger server component according to the documentation I get:

[totty@y900 ~]$ cargo install --git https://github.com/probe-rs/probe-rs --force --branch master probe-rs-debugger
    Updating git repository `https://github.com/probe-rs/probe-rs`
error: could not find `probe-rs-debugger` in https://github.com/probe-rs/probe-rs?branch=master with version `*`

I am not sure if this is a cargo or a git problem. A simple Google search came up empty. Any hint is appreciated.

**Desktop (please complete the following information):**
  - Arch Linux

**Additional context**
cargo 1.61.0 (a028ae4 2022-04-29)

noppej, 2022-11-16:
I just did a cut and paste of your cargo install command, and it works perfectly (MacOS Ventura). 

Can you try to simplify with something like ... `cargo install --git https://github.com/probe-rs/probe-rs probe-rs-debugger`

tsadowski, 2022-11-17:
No, this has the same result. I was a bit puzzled because 
`cargo install --git https://github.com/probe-rs/probe-rs`
I get

    Updating git repository `https://github.com/probe-rs/probe-rs`
error: multiple packages with binaries found: binary_cargo_config, binary_cargo_config_toml, binary_project, workspace_bin, xtask. When installing a git repository, cargo will always search the entire repo for any Cargo.toml. Please specify which to install.

No probe-rs-debugger.

So I changed in to the .cargo git checkout directory and tried

`cargo install --path debugger`
result:
error: failed to parse manifest at `/home/totty/.cargo/git/checkouts/probe-rs-92d520f86d5696fc/375d46d/debugger/Cargo.toml`

Caused by:
  feature `workspace-inheritance` is required

  The package requires the Cargo feature called `workspace-inheritance`, but that feature is not stabilized in this version of Cargo (1.61.0 (a028ae4 2022-04-29)).
  Consider trying a newer version of Cargo (this may require the nightly release).
  See https://doc.rust-lang.org/nightly/cargo/reference/unstable.html#workspace-inheritance for more information about the status of this feature.

I can try to upgrade cargo.

Cheers,
Torsten

tsadowski, 2022-11-17:
Yes, that was the cause.
`rustup update`and it works.

Cheers, Torsten

closed 2022-11-17 by tsadowski