probe-rs/probe-rs #1283: bitvec v0.22.0 depends on funty = "~1.2 which has been yanked

Georges760, 2022-11-18:
**Describe the bug**

[funty 1.2 ](https://crates.io/crates/funty/1.2.0) has been yanked, so bitvec v0.22.0 and all the current probe-rs cannot be build

**To Reproduce**

fresh cargo build from probe-rs directory

**Expected behavior**

build OK

**Stacktrace**

**Desktop (please complete the following information):**

Linux Ubuntu

**Additional context**

Tiwalun, 2022-11-18:
Seems to work fine for me. Could you please post the exact error message you get when trying to build?

noppej, 2022-11-18:
The link posted in the OP says this ...

"This crate has been yanked, but it is still available for download for other crates that may be depending on it."

Tiwalun, 2022-11-18:
The issue seems to be [itm-decode](https://lib.rs/crates/itm-decode) in our case, which depends on the old bitvec version, which in itself depends on `funty`. There seems to be a new version in https://github.com/rtic-scope/itm, but that's not on crates.io yet unfortunately.

commit 49ed98f0 (2022-11-18): Merge #1284

1284: Add a patch for the yanked funty version r=Yatekii a=Tiwalun

This temporarily fixes #1283, but for a release it will not work.
PR #1284 (2022-11-18): Add a patch for the yanked funty version
This temporarily fixes #1283, but for a release it will not work.
closed 2022-11-18 by bors[bot]