probe-rs/probe-rs #1324: Can't make it run with NRF52840 - Error: Failed to get information for AP ApAddress { dp: Default, ap: 0 }

rracariu, 2022-12-13:
**Describe the bug**
Trying to run of flash a program on an NRF52840 - Adafruit Feather Sense -  will always fail with:

`Error: Failed to get information for AP ApAddress { dp: Default, ap: 0 }`

**To Reproduce**
Tested with embassy-rs nrf examples such as: 

`cargo run --bin blinky`

or 

`cargo flash --bin blinky --chip nRF52840_xxAA --connect-under-reset`

**Expected behavior**
Expected to work as JLink's proprietary tools work.

**Stacktrace**
<details>
  <summary>Debug output</summary>

```
(HOST) DEBUG J-Link interface is #0
└─ jaylink @ /.cargo/registry/src/github.com-1ecc6299db9ec823/jaylink-0.3.0/src/lib.rs:397
(HOST) TRACE write 1 bytes: [e8]
└─ jaylink @ /.cargo/registry/src/github.com-1ecc6299db9ec823/jaylink-0.3.0/src/lib.rs:521
(HOST) TRACE read 4 bytes: [33, 58, ea, 88]
└─ jaylink @ /.cargo/registry/src/github.com-1ecc6299db9ec823/jaylink-0.3.0/src/lib.rs:551
(HOST) DEBUG legacy caps: GetHwVersion | ReadConfig | WriteConfig | GetMaxBlockSize | GetHwInfo | ResetStopTimed | SelectIf | GetCounters | GetCpuCaps | ExecCpuCmd | Swo | Register | GetCapsEx
└─ jaylink @ /.cargo/registry/src/github.com-1ecc6299db9ec823/jaylink-0.3.0/src/lib.rs:450
(HOST) TRACE write 1 bytes: [ed]
└─ jaylink @ /.cargo/registry/src/github.com-1ecc6299db9ec823/jaylink-0.3.0/src/lib.rs:521
(HOST) TRACE read 32 bytes: [33, 58, ea, 88, c, 28, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
└─ jaylink @ /.cargo/registry/src/github.com-1ecc6299db9ec823/jaylink-0.3.0/src/lib.rs:551
(HOST) DEBUG unknown ext. capability bits: 0x280C88EA5833 truncated to 0x88EA5833 (Reserved0 | GetHwVersion | ReadConfig | WriteConfig | GetMaxBlockSize | GetHwInfo | ResetStopTimed | SelectIf | GetCounters | GetCpuCaps | ExecCpuCmd | Swo | Register | GetCapsEx)
└─ jaylink::capabilities @ /.cargo/registry/src/github.com-1ecc6299db9ec823/jaylink-0.3.0/src/capabilities.rs:96
(HOST) DEBUG extended caps: GetHwVersion | ReadConfig | WriteConfig | GetMaxBlockSize | GetHwInfo | ResetStopTimed | SelectIf | GetCounters | GetCpuCaps | ExecCpuCmd | Swo | Register | GetCapsEx
└─ jaylink @ /.cargo/registry/src/github.com-1ecc6299db9ec823/jaylink-0.3.0/src/lib.rs:467
(HOST) TRACE write 2 bytes: [c7, ff]
└─ jaylink @ /.cargo/registry/src/github.com-1ecc6299db9ec823/jaylink-0.3.0/src/lib.rs:521
(HOST) TRACE read 4 bytes: [3, 0, 0, 0]
└─ jaylink @ /.cargo/registry/src/github.com-1ecc6299db9ec823/jaylink-0.3.0/src/lib.rs:551
(HOST) DEBUG opened probe
└─ probe_run::probe @ /.cargo/registry/src/github.com-1ecc6299db9ec823/probe-run-0.3.5/src/probe.rs:33
(HOST) DEBUG Attaching to J-Link
└─ probe_rs::probe::jlink @ /.cargo/registry/src/github.com-1ecc6299db9ec823/probe-rs-0.13.0/src/probe/jlink/mod.rs:456
(HOST) TRACE write 2 bytes: [c7, 1]
└─ jaylink @ /.cargo/registry/src/github.com-1ecc6299db9ec823/jaylink-0.3.0/src/lib.rs:521
(HOST) TRACE read 4 bytes: [1, 0, 0, 0]
└─ jaylink @ /.cargo/registry/src/github.com-1ecc6299db9ec823/jaylink-0.3.0/src/lib.rs:551
(HOST) DEBUG Attaching with protocol 'SWD'
└─ probe_rs::probe::jlink @ /.cargo/registry/src/github.com-1ecc6299db9ec823/probe-rs-0.13.0/src/probe/jlink/mod.rs:476
(HOST) INFO  J-Link: S/N: 123456
└─ probe_rs::probe::jlink @ /.cargo/registry/src/github.com-1ecc6299db9ec823/probe-rs-0.13.0/src/probe/jlink/mod.rs:484
(HOST) DEBUG J-Link: Capabilities: GetHwVersion | ReadConfig | WriteConfig | GetMaxBlockSize | GetHwInfo | ResetStopTimed | SelectIf | GetCounters | GetCpuCaps | ExecCpuCmd | Swo | Register | GetCapsEx
└─ probe_rs::probe::jlink @ /.cargo/registry/src/github.com-1ecc6299db9ec823/probe-rs-0.13.0/src/probe/jlink/mod.rs:485
(HOST) TRACE write 1 bytes: [1]
└─ jaylink @ /.cargo/registry/src/github.com-1ecc6299db9ec823/jaylink-0.3.0/src/lib.rs:521
(HOST) TRACE read 2 bytes: [70, 0]
└─ jaylink @ /.cargo/registry/src/github.com-1ecc6299db9ec823/jaylink-0.3.0/src/lib.rs:551
(HOST) TRACE read 112 bytes: [4a, 2d, 4c, 69, 6e, 6b, 20, 41, 52, 4d, 2d, 4f, 42, 20, 53, 54, 4d, 33, 32, 20, 63, 6f, 6d, 70, 69, 6c, 65, 64, 20, 41, 75, 67, 20, 32, 32, 20, 32, 30, 31, 32, 20, 31, 39, 3a, 35, 32, 3a, 30, 34, 0, 43, 6f, 70, 79, 72, 69, 67, 68, 74, 20, 32, 30, 30, 33, 2d, 32, 30, 30, 38, 20, 53, 45, 47, 47, 45, 52, 3a, 20, 77, 77, 77, 2e, 73, 65, 67, 67, 65, 72, 2e, 63, 6f, 6d, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
└─ jaylink @ /.cargo/registry/src/github.com-1ecc6299db9ec823/jaylink-0.3.0/src/lib.rs:551
(HOST) INFO  J-Link: Firmware version: J-Link ARM-OB STM32 compiled Aug 22 2012 19:52:04
└─ probe_rs::probe::jlink @ /.cargo/registry/src/github.com-1ecc6299db9ec823/probe-rs-0.13.0/src/probe/jlink/mod.rs:490
(HOST) TRACE write 1 bytes: [f0]
└─ jaylink @ /.cargo/registry/src/github.com-1ecc6299db9ec823/jaylink-0.3.0/src/lib.rs:521
(HOST) TRACE read 4 bytes: [70, 11, 1, 0]
└─ jaylink @ /.cargo/registry/src/github.com-1ecc6299db9ec823/jaylink-0.3.0/src/lib.rs:551
(HOST) INFO  J-Link: Hardware version: J-Link 7.0.0
└─ probe_rs::probe::jlink @ /.cargo/registry/src/github.com-1ecc6299db9ec823/probe-rs-0.13.0/src/probe/jlink/mod.rs:492
[... 134 lines cut ...]
 - CoreSight SoC-400 or earlier
 - Scanning AP map to find all available APs
 - AP[2]: Stopped AP scan as end of AP map has been reached
 - AP[0]: AHB-AP (IDR: 0x24770011)
 - AP[1]: JTAG-AP (IDR: 0x02880000)
 - Iterating through AP map to find AHB-AP to use
 - AP[0]: Core found
 - AP[0]: AHB-AP ROM base: 0xE00FF000
 - CPUID register: 0x410FC241. Implementer code: 0x41 (ARM)
 - Found Cortex-M4 r0p1, Little endian.
 - FPUnit: 6 code (BP) slots and 2 literal slots
 - CoreSight components:
 - ROMTbl[0] @ E00FF000
 - [0][0]: E000E000 CID B105E00D PID 000BB00C SCS-M7
 - [0][1]: E0001000 CID B105E00D PID 003BB002 DWT
 - [0][2]: E0002000 CID B105E00D PID 002BB003 FPB
 - [0][3]: E0000000 CID B105E00D PID 003BB001 ITM
 - [0][4]: E0040000 CID B105900D PID 000BB9A1 TPIU
 - [0][5]: E0041000 CID B105900D PID 000BB925 ETM
 - Executing init sequence ...
  - Initialized successfully
 - Target interface speed: 4000 kHz (Fixed)
 - Found 1 JTAG device. Core ID: 0x2BA01477 (None)
 - Connected successfully
```

Yatekii, 2022-12-13:
You can use https://crates.io/crates/nrf-recover or the `nrfutil` to unlock the chip and then it should work just fine.

rracariu, 2022-12-13:
> You can use https://crates.io/crates/nrf-recover or the `nrfutil` to unlock the chip and then it should work just fine.

I tried that before with `nrfjprog --eraseall` and still get the same error. BTW `nrf-recorver` times out also.

Yatekii, 2022-12-13:
Is the reset line connected for `--connect-under-reset`?

Yatekii, 2022-12-13:
Also, is this on master? I think we have an issue with this fixed on master.

rracariu, 2022-12-13:
Reset line is connected.

I'm using `probe-rs-0.13.0`.

BTW - the bug report has the probe-rs logs if by any chance you missed that, just expand  `Debug output` section at the top.

rracariu, 2022-12-13:
Tried with master: `cargo install --git https://github.com/probe-rs/probe-rs --force --branch master probe-rs-cli`

And got the same error.

Yatekii, 2022-12-13:
Yeah, the log is unfortunately not very detailed. We are working on better logs. It has a ton of information, but it's not too useful. Normally the error you are seeing means the chip is locked.

Tiwalun, 2022-12-13:
@rracariu What kind of J-Link are you using? 

It seems that probe-rs is unable to connect at all, which usually indicates bad wiring. But if the J-Link software works, that means the connection is good... A bit mysterious.

Zagitta, 2022-12-17:
I've run into this exact bug a while back.
Chances are you have the newer revision that according to @Dirbaio automatically re-enables the write protection.
This is the fix he provided me with:
```rust
unsafe {
    let nvmc = &*pac::NVMC::ptr();

    // UICR.APPROTECT = HwDisabled
    if *(0x10001208 as *mut u32) != 0x0000_005a {
        nvmc.config.write(|w| w.wen().wen());
        while nvmc.ready.read().ready().is_busy() {}
        core::ptr::write_volatile(0x10001208 as *mut u32, 0x0000_005a);
        while nvmc.ready.read().ready().is_busy() {}
        nvmc.config.reset();
        while nvmc.ready.read().ready().is_busy() {}
        cortex_m::peripheral::SCB::sys_reset();
    }

    // APPROTECT.DISABLE = SwDisabled
    (0x4000_0558 as *mut u32).write_volatile(0x0000_005a);
}
```
You'll need to stick it at the top of your main function in YOUR app. That at least made it work reliably for me, your milage may vary :)

Here's a link to the matrix chat where @Dirbaio helped me solve the issue: https://matrix.to/#/!vhKMWjizPZBgKeknOo:matrix.org/$WwWkVyOhQUxjWSClmZFXeZLMSYLm79y2HW1qkI6Ovmc?via=matrix.org&via=chat.berline.rs&via=grusbv.com

Dirbaio, 2022-12-18:
Do not use --connect-under-reset for nrf chips. Unlike stm32, the reset pin also resets the SWD circuitry, so it doesn't respond at all while in reset.

Yatekii, 2022-12-18:
Is this really true @Dirbaio? I thought a discussion on Matrix proved this incorrect? Because I was under that belief as well :)

TheButlah, 2023-01-07:
Please note that as best as I can tell, the `APPROTECT` peripheral doesn't exist on boards older than revision 3. So if you are following [the comment](https://github.com/probe-rs/probe-rs/issues/1324#issuecomment-1356273774), you should be sure to check the revision of your board first, and only attempt this process if it matches that revision.

DeepSOIC, 2024-03-14:
this helped me, but i'm not sure if it will work with j-link too:
> okay, so adding --allow-erase-all to the runner command fixes it, closing the issue, sorry about the noise
> 
> .cargo/config.toml:
> ```
> ...
> runner = "probe-rs run --chip nRF52840_xxAA --allow-erase-all"
> ...
> ```