probe-rs/probe-rs #1325: Refactor Registry and Registry/Target relationship

willglynn, 2022-12-13:
Following up my thoughts in https://github.com/probe-rs/probe-rs/pull/1312#issuecomment-1344603214, I now have a concrete proposal.

# Observations

`Registry` assumes that [`Chip`+`ChipFamily`](https://github.com/probe-rs/probe-rs/blob/4f239d0f98e5cd75d421b19edb8adbc975cf5924/probe-rs/src/config/registry.rs#L279-L282) contain all the information needed to interact with a device. I think this assumption is false, that this assumption led to `Target::new()`'s [prodigious special casing by target name](https://github.com/probe-rs/probe-rs/blob/4f239d0f98e5cd75d421b19edb8adbc975cf5924/probe-rs/src/config/target.rs#L98-L134), and that it [misplaces `Chip`+`ChipFamily`-related validation](https://github.com/probe-rs/probe-rs/blob/4f239d0f98e5cd75d421b19edb8adbc975cf5924/probe-rs/src/config/target.rs#L72-L90) into `Target::new()`.

`Registry` assumes that a probe can produce a `ChipInfo` which can [subsequently be used for autodetection](https://github.com/probe-rs/probe-rs/blob/4f239d0f98e5cd75d421b19edb8adbc975cf5924/probe-rs/src/config/registry.rs#L237). I think this assumption is false, that the current implementation cannot detect any chips in practice (there are no `part`s which appear exactly once), and that this strategy cannot succeed in general without splitting responsibility for special cases across `Probe`, `ChipInfo`, and `Chip`/`ChipFamily`.

`Registry` as a singleton is [globally mutable shared state](https://github.com/probe-rs/probe-rs/blob/4f239d0f98e5cd75d421b19edb8adbc975cf5924/probe-rs/src/config/registry.rs#L320-L324). This is awkward to code (e.g. duplicated bare functions to hide the singleton) and leads to awkward restrictions (e.g. there's no way to undo adding a target from a YAML file).

# Proposal

## First pass

Define a group of `pub(crate)` object-safe traits:
* `Provider`: a thing with a `name()` and the ability to list `families()`
* `Family`: a thing with a `name()` and the ability to list `variants()`
* `Variant`: a thing with a `name()` and the ability to construct a `Target`

`ChipFamily` can implement `Family`. `(ChipFamily, Chip)` can implement `Variant`. These traits collectively cover `Registry`'s access patters, so `Registry` can now contain `Vec<Box<dyn Provider>>` without knowing about `ChipFamily` or `Chip`.

Move the generic targets out of `Registry` and into a `Generic` provider. Move the built-in targets out of `Registry` and into a `Builtin` provider, gated by the feature flag. Move file-based targets out of `Registry` and into a `File` provider. Validation needs for each of these differ, and validation can be moved from `Target::new()` into the provider constructor or into the test suite as appropriate. `Provider` is a new abstraction, but identifying three of them right away suggests it's valuable.

Autodetection is and should remain architecture-specific, but autodetection should happen while probing rather than being split between probe and lookup phases. Extend `Provider` with an `autodetect_arm(&ArmChipInfo, &mut Memory)` function returning an `Option<Box<dyn Variant>>`. Move existing `ChipFamily`-based logic to `ChipFamily::autodetect_arm()`, or remove it altogether if it can't work as designed.

Remove the singleton and have callers instantiate and use `Registry` as needed.

This is probably one big PR, with `Registry`'s exposure probably motivating a version bump.

## Subsequent refactoring

Each built-in `ChipFamily` with custom debug sequences or custom autodetection can and should implement a `Provider` instead of using `Builtin`. Break them out into vendor-specific `Provider`s, removing special cases from `Builtin` until they're all gone. Consider breaking out remaining families into `Provider`s and removing `Builtin` altogether.

The `Provider` trait is a way to co-locate target definitions with their debug sequences and autodetection routines, consolidating all three of those related concerns into a single modular entrypoint. In other words, `infineon.rs` should contain `pub struct Infineon;`, `impl Provider for Infineon`, and everything Infineon-related should only be accessible through that interface. `Registry` need not know or care how each `Provider` is implemented.

This keeps vendor-specific things in `probe-rs`, but it provides an interface through which vendor-specific behaviors can be expressed without scattering details throughout the codebase. That interface -- those traits -- remain `pub(crate)` and can freely evolve as `probe-rs` evolves.

This is probably a series of PRs, probably starting with build-time support.

## Potential future work

`Provider` can be exposed as a plug-in interface, e.g. to implement a `Provider` via a WebAssembly module. `infineon.rs` could become `infineon.wasm` – built, tested, versioned, and distributed separately. `probe-rs` need not know about Infineon at all.

This breaks vendor-specific things out of `probe-rs` completely, at the expense of committing to an external interface. This is a tradeoff which principally depends on the maturity of that interface.

This is probably off the roadmap/out of scope for the medium term, but it would at least be possible without touching all this again. `impl Provider for WasmModule` would let `Registry` add `.wasm` targets just like a YAML file.

willglynn, 2022-12-13:
It seems that `--chip-description-path` doesn't work; the only reference to [`maybe_load_chip_desc()`]( https://github.com/probe-rs/probe-rs/blob/c05cbc0ee3b195676889b05a4ad842e8158cd0b2/probe-rs-cli-util/src/common_options.rs#L174-L190) is a doctest. I discovered this while changing the CLI to take a `&Registry` in #1326. A `&Registry` argument makes the registry dependency explicit in the code, whereas the current globally mutable shared registry relies on programmers to notice the "do this before that" requirement in the `maybe_load_chip_desc()` documentation.

Yatekii, 2022-12-13:
I like the the outlined proposal very much! Thanks a lot for putting in all the effort!

I will add more comments on the Draft PR!