probe-rs/probe-rs #1336: ESP32C3 - lots of log data corrupts rtt control block

TheButlah, 2022-12-16:
**Describe the bug**
Logging lots of data on the esp32c3 often corrupts the RTT control block.

**To Reproduce**
log a large amount of data very fast, in a tight loop, to `defmt-rtt`

**Expected behavior**
data should be logged, either blocking or trimming depending on the settings.

**Stacktrace**
[image]

**Desktop (please complete the following information):**
 - macOS 12.6, apple m1 processor

noppej, 2022-12-29:
@TheButlah ... I'd like to test if this is an API or a tool thing, but cannot reproduce this on probe-rs-debugger. Can you share some parameters please?
- What is the connection speed to the probe?
- Are you writing small amounts of data, but very fast, or large (how large) blocks at a time?
- Anything you think that might be a contributing factor (other than the 'large amount of data very fast') ?

@Yatekii do you know off-hand if cargo-embed turns on "BlockIfFull" on the defmt channel when it connects?

Yatekii, 2022-12-29:
I think cargo-embed does not turn that on.

noppej, 2022-12-31:
@TheButlah I've setup cargo embed for my own test case, and although I'm dumping full buffers at zero wait time, I am unable to reproduce this.

Can you please provide a test case?

TheButlah, 2023-01-01:
sure! I was having a bit of trouble getting a minimally reproducible example. I'll try to get back to you after the holiday

TheButlah, 2023-01-05:
I tried this a bit ago but getting a basic esp32c3 no_std working was taking some time, and my freetime is quite bottlenecked rn with apartment hunting. Apologies for the delay 😅

TheButlah, 2023-01-05:
OK, on my minimally reproducible example, I could not get the error message above. But I did get the weird behavior I often observe when logging data, where I frequently have to reset the device a lot. I opened #1353 for that.

TheButlah, 2023-01-05:
Since my attempt to reproduce with a minimal example failed, here is a *very not minimal* example:
[recording](https://asciinema.org/a/8SeynYwrrwFr5AUIccmRRFZvf)
[very complex example (sorry! I will try to pare it down)](https://github.com/SlimeVR/SlimeVR-Rust/tree/2789cc089c38c2f412133e065d1f3c1f484e949e)
This recording was taken on windows 11, build 22623.875. Seems like the issue is cross platform, at least between windows and mac.

Make this change to cargo.toml to get your working tree in the same state that I used for the recording:
[image]

MabezDev, 2023-01-26:
FYI I've seen this on both RISC-V and ARM when the stack clobbers the RTT header, not sure this is the case here but just adding another data point.

noppej, 2023-01-26:
Silly question, but how do you know it is the stack that clobbers the RTT header, and not maybe the rtt app that misreports an error, or maybe overwrites the header when it updates the buffer pointers?  Just trying to get a full understanding of what others have seen/discovered.

MabezDev, 2023-01-26:
Well, I guess once a stack overflow happens and corrupts a part of memory, anything can happen, its UB. I saw the stack overflow by using a debug assist which allowed me to monitor the stack pointer. Not really much that can be done on the probe-rs side, just wanted to note a possible cause :).

noppej, 2023-01-27:
That is good info. I wonder if we could trap the error about the RTT header corruption, and test the stack pointer, for a more informative error. Knowing there was a stack overflow would be less likely to send users down the rathole of RTT header corruptions. Thoughts?

Yatekii, 2023-01-27:
Yeah I think there was the idea of a canary in the past :)

noppej, 2023-01-27:
My guess is it could be a relatively low-cost test as part of the RTT poll cycle.  Also valuable, because if reading RTT header information detects a stack overflow condition, then you can declare that everything else about your app's execution after that point is suspect.

That said, isn't that exactly what [flip-link](https://crates.io/crates/flip-link) was meant to protect against?

@TheButlah Is your original problem with or without flip-link?  Can you reproduce in both cases?

Yatekii, 2023-01-27:
Yes, flip-link should protect against exactly this :)

TheButlah, 2023-01-27:
> @TheButlah Is your original problem with or without flip-link?

I usually (always?) get this error on esp32c3. `flip-link` is only for cortex-m devices.

I tried setting up flip link for esp32c3 via my .cargo/config.toml, and tried it with both `direct-boot` and without in the esp hal. This is what I got :

direct-boot:
```
60abb.rlib" "/Users/ryan.butler/P/slimevr/rust/firmware/target/riscv32imc-unknown-none-elf/debug/deps/libcompiler_builtins-0
158787edf03555e.rlib" "-Bdynamic" "-znoexecstack" "-L" "/Users/ryan.butler/.rustup/toolchains/nightly-2022-12-18-aarch64-app
le-darwin/lib/rustlib/riscv32imc-unknown-none-elf/lib" "-o" "/Users/ryan.butler/P/slimevr/rust/firmware/target/riscv32imc-un
known-none-elf/debug/deps/firmware-1e28105c11103122" "--gc-sections" "-Tlinkall.x" "-Tdefmt.x"
  = note: rust-lld: warning: section type mismatch for .uninit.defmt-rtt.BUFFER
          >>> /Users/ryan.butler/P/slimevr/rust/firmware/target/riscv32imc-unknown-none-elf/debug/deps/libdefmt_rtt-0059e106
3e6f7d93.rlib(defmt_rtt-0059e1063e6f7d93.defmt_rtt.eeaf34bf-cgu.5.rcgu.o):(.uninit.defmt-rtt.BUFFER): SHT_PROGBITS
          >>> output section .uninit: SHT_NOBITS
          
          rust-lld: warning: address (0x42000008) of section .text is not a multiple of alignment (256)
          rust-lld: warning: address (0x3c041ece) of section .rodata is not a multiple of alignment (16)
          Error: "MEMORY.RAM not found after scanning linker scripts"
```

I'm not sure that flip link is an option for my use case.

TheButlah, 2023-01-27:
Note that even when not logging data fast, this bug occurs. I'm editing the original message now.

noppej, 2023-01-27:
I forgot that flip-link is only for ARM.  

If we want to test the theory that this error is caused by a stack overflow, do you think there is anything in your app that could cause that?

TheButlah, 2023-01-27:
honestly I don't think so. It happens with all imu drivers stubbed and the networking stubbed. At that point its just embassy and defmt logging.

TheButlah, 2023-01-27:
OK I just did another test, and realized the behavior differs depending on if I use direct-boot or not.

**Direct boot**: I do not observe the error message.
```
cargo espflash flash --format direct-boot --release
cargo embed monitor --release
```
OR 
```rust
cargo embed --release
```

**Without direct boot**: I observe the error message.
```
cargo espflash flash --release
cargo embed monitor --release
```

PR #2340 (2024-04-04): Validate RTT channel pointers during attach, and make the logs less chatty
> An attempt to solve the "Corrupted RTT channel" errors which sometimes appear when attaching to a RTT control block which hasn't been fully initialized.

Previously, probe-rs attached to partially initialized RTT control blocks. This meant that subsequent polls sometimes reported corrupted control block errors, because the read and/or write pointers were out of range. We now read and verify these pointers and delay initializing if they are clearly wrong.

This isn't a perfect solution. Problem is that we might still e [... cut]
closed 2024-04-04 by bugadani