probe-rs/probe-rs #1340: Can't flash STM32F103 which is using wfi instruction

Bobo1239, 2022-12-21:
**Describe the bug**
When a program using `wfi` instructions is flashed on an STM32F103(C8T8) then probe-rs (via probe-run) isn't able to connect to/flash the chip without `--connect-under-reset` and manually resetting.

Interestingly it seems like no tool is able to successfully connect to the chip. I've tried the official STM32CubeProgrammer (with various settings for mode and reset mode; "debug in low power mode" enabled) which just returns `Error: ST-LINK error (DEV_TARGET_CMD_ERR)`. And the FOSS [stlink tools](https://github.com/stlink-org/stlink) also can't establish a connection. (`st-info --probe` can't read any chip information)

<hr/>

With/Without #1337; no reset:
```
Error: An error with the usage of the probe occurred

Caused by:
    0: An error specific to a probe type occurred
    1: Command failed with status SwdDpWait
```
(We don't even get to flashing)

<hr/>

With #1337: manual reset + `--connect-under-reset`:
```
(HOST) INFO  flashing program (1 pages / 1.00 KiB)
(HOST) INFO  success!
RTT logs not available; blocking until the device halts..
────────────────────────────────────────────────────────────────────────────────
^C────────────────────────────────────────────────────────────────────────────────
stack backtrace:
   0: __wfi
        at ./asm/lib.rs:52:14
   1: stm32f1_wfi_repro::__cortex_m_rt_main
        at src/main.rs:21:5
   2: main
        at src/main.rs:19:1
   3: Reset
(HOST) INFO  device halted by user
```
(Flashing goes through and probe can connect.)

Afterwards flashing works without reset since the chip gets halted each time and (presumably) is not in a low power state.

<hr/>

Without #1337; manual reset + `--connect-under-reset`:
```
(HOST) INFO  flashing program (1 pages / 1.00 KiB)
(HOST) INFO  success!
RTT logs not available; blocking until the device halts..
────────────────────────────────────────────────────────────────────────────────
Error: An error with the usage of the probe occurred

Caused by:
    0: An error specific to a probe type occurred
    1: Command failed with status SwdDpWait
```
(Flashing goes through but the probe can't connect afterwards.)

Without #1337 the flashing succeeds but we can't establish a SWD connection.

**To Reproduce**
Steps to reproduce the behavior:

1. Clone https://github.com/Bobo1239/stm32f1_wfi_repro (also has a log output in `no_reset.log`)
2. `cargo build --release`
3. (patched) `probe-run -vv --chip STM32F103C8 target/thumbv7m-none-eabi/release/stm32f1_wfi_repro` (`--connect-under-reset`); The flashing fails with the output above

**Expected behavior**
Flashing should work.

**Desktop (please complete the following information):**

- Arch Linux

**Additional context**
Unfortunately I'm currently unable to check whether the STM32F103 is a genuine part or a clone. (Would use [this](https://mecrisp-stellaris-folkdoc.sourceforge.io/bluepill-diags-v1.640.html#diags-1-640) but I first need to fix an SMD resistor on the blue pill board...)

The probe I'm using is a cheap [ST-LINK V2 clone](https://stm32-base.org/boards/Debugger-STM32F101C8T6-STLINKV2). AFAICT hardware reset (the RST pin) on those isn't wired up correctly so the reset must be done manually with my setup.

burrbull, 2022-12-21:
https://github.com/probe-rs/probe-rs/pull/1337

Tiwalun, 2022-12-21:
When WFI is used, and no flags are set to enable debugging in sleep modes, it it expected that no connection is possible without `--connect-under-reset`. The debug module is powered down in this case, so no connection is possible.

Of couse, the connection *can* work if you happen to connect while the chip is not in sleep mode, but it will not work reliably. 

If the reset pin is connected, it should always be possible to connect with `--connect-under-reset`.

Bobo1239, 2022-12-22:
@Tiwalun: Hm, yeah that was my suspicion after digging into this. Thanks for the confirmation and I guess I can close this issue.

(also linking #350 if other people stumble upon this behavior)

closed 2022-12-22 by Bobo1239