probe-rs/probe-rs #1347: `probe-rs-cli info` fails on ST-Link probe with confusing message about RISC-V

noppej, 2023-01-03:
**Describe the bug**
```
probe-rs-cli info
Probing target via JTAG
No JTAG interface was found on the connected probe. Thus, RISC-V info cannot be printed.
Probing target via SWD
thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: Probe(ProbeSpecific(CommandFailed(Other(29))))', cli/src/info.rs:76:88
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

**To Reproduce**
The error happens against the following:
- STM32H745ZI-Q Nucleo (ST-Link v3)

The command works correctly against Micro:Bit 2.0 and RP2040.

Tiwalun, 2023-01-03:
Not sure what's confusing about the RISCV message? We currently don't support JTAG on ST-Links as far as I know, and I don't know if it even would be possible to support.

The other error looks more interesting, could you please post the backtrace?

And do mean that a ST-Link v3 works with a RP2040 and a Micro:Bit, or was that another probe?

noppej, 2023-01-03:
Each of those were using their own probe, so ST-Link v3 is only against the STM32H745 chip. 

The RISC-V error message is confusing, because I'm not asking for anything RISC-V related, nor am I asking for JTAG. I'm simply asking for a info on connected chips. I would expect only successes to be reported.

noppej, 2023-01-03:
```
RUST_BACKTRACE=full probe-rs-cli info
Probing target via JTAG
No JTAG interface was found on the connected probe. Thus, RISC-V info cannot be printed.
Probing target via SWD
thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: Probe(ProbeSpecific(CommandFailed(Other(29))))', cli/src/info.rs:76:88
stack backtrace:
   0:        0x104845664 - std::backtrace_rs::backtrace::libunwind::trace::h53e322ea3db1bb39
                               at /rustc/37d7de337903a558dbeb1e82c844fe915ab8ff25/library/std/src/../../backtrace/src/backtrace/libunwind.rs:93:5
   1:        0x104845664 - std::backtrace_rs::backtrace::trace_unsynchronized::hed588d6ba19923fc
                               at /rustc/37d7de337903a558dbeb1e82c844fe915ab8ff25/library/std/src/../../backtrace/src/backtrace/mod.rs:66:5
   2:        0x104845664 - std::sys_common::backtrace::_print_fmt::hb61a6d856473f48e
                               at /rustc/37d7de337903a558dbeb1e82c844fe915ab8ff25/library/std/src/sys_common/backtrace.rs:65:5
   3:        0x104845664 - <std::sys_common::backtrace::_print::DisplayBacktrace as core::fmt::Display>::fmt::h3afa7525ad35d5ca
                               at /rustc/37d7de337903a558dbeb1e82c844fe915ab8ff25/library/std/src/sys_common/backtrace.rs:44:22
   4:        0x1048628c4 - core::fmt::write::hd5a925881a3d0450
                               at /rustc/37d7de337903a558dbeb1e82c844fe915ab8ff25/library/core/src/fmt/mod.rs:1208:17
   5:        0x10483f40c - std::io::Write::write_fmt::hb312ea68e22acd06
                               at /rustc/37d7de337903a558dbeb1e82c844fe915ab8ff25/library/std/src/io/mod.rs:1682:15
   6:        0x104845478 - std::sys_common::backtrace::_print::haf7bc9ba09583eaa
                               at /rustc/37d7de337903a558dbeb1e82c844fe915ab8ff25/library/std/src/sys_common/backtrace.rs:47:5
   7:        0x104845478 - std::sys_common::backtrace::print::h42024cef2ac72e1c
                               at /rustc/37d7de337903a558dbeb1e82c844fe915ab8ff25/library/std/src/sys_common/backtrace.rs:34:9
   8:        0x104846bec - std::panicking::default_hook::{{closure}}::h3bcd32378faf1100
                               at /rustc/37d7de337903a558dbeb1e82c844fe915ab8ff25/library/std/src/panicking.rs:267:22
   9:        0x104846944 - std::panicking::default_hook::h8baac6582e23f278
                               at /rustc/37d7de337903a558dbeb1e82c844fe915ab8ff25/library/std/src/panicking.rs:286:9
  10:        0x104847210 - std::panicking::rust_panic_with_hook::h719cb65bf474ab26
                               at /rustc/37d7de337903a558dbeb1e82c844fe915ab8ff25/library/std/src/panicking.rs:688:13
  11:        0x104847030 - std::panicking::begin_panic_handler::{{closure}}::h17976f8dcfb388ae
                               at /rustc/37d7de337903a558dbeb1e82c844fe915ab8ff25/library/std/src/panicking.rs:579:13
  12:        0x104845acc - std::sys_common::backtrace::__rust_end_short_backtrace::hccf9f2e71d4c318f
                               at /rustc/37d7de337903a558dbeb1e82c844fe915ab8ff25/library/std/src/sys_common/backtrace.rs:137:18
  13:        0x104846d8c - rust_begin_unwind
                               at /rustc/37d7de337903a558dbeb1e82c844fe915ab8ff25/library/std/src/panicking.rs:575:5
  14:        0x10488696c - core::panicking::panic_fmt::h96965722d4967105
                               at /rustc/37d7de337903a558dbeb1e82c844fe915ab8ff25/library/core/src/panicking.rs:64:14
  15:        0x104886c8c - core::result::unwrap_failed::h6facee39b669ca69
                               at /rustc/37d7de337903a558dbeb1e82c844fe915ab8ff25/library/core/src/result.rs:1791:5
  16:        0x104254e88 - probe_rs_cli::info::show_info_of_device::h60602d4e086f80e4
  17:        0x10425dd38 - probe_rs_cli::main::h6690a7d5b5d0e4e4
  18:        0x104289de4 - std::sys_common::backtrace::__rust_begin_short_backtrace::he6f0c1a5d9644b8a
  19:        0x1042b4b7c - std::rt::lang_start::{{closure}}::hce85c4e4af42f855
  20:        0x104839470 - core::ops::function::impls::<impl core::ops::function::FnOnce<A> for &F>::call_once::h16d7e0144d7f6169
                               at /rustc/37d7de337903a558dbeb1e82c844fe915ab8ff25/library/core/src/ops/function.rs:606:13
  21:        0x104839470 - std::panicking::try::do_call::h53f496d3f7c10394
                               at /rustc/37d7de337903a558dbeb1e82c844fe915ab8ff25/library/std/src/panicking.rs:483:40
  22:        0x104839470 - std::panicking::try::h596a6728fc0baefb
                               at /rustc/37d7de337903a558dbeb1e82c844fe915ab8ff25/library/std/src/panicking.rs:447:19
  23:        0x104839470 - std::panic::catch_unwind::hb51dcc02d2b8d31b
                               at /rustc/37d7de337903a558dbeb1e82c844fe915ab8ff25/library/std/src/panic.rs:137:14
  24:        0x104839470 - std::rt::lang_start_internal::{{closure}}::hd76e4a81fdf0c7d5
                               at /rustc/37d7de337903a558dbeb1e82c844fe915ab8ff25/library/std/src/rt.rs:148:48
  25:        0x104839470 - std::panicking::try::do_call::h88162cdc06a24d43
                               at /rustc/37d7de337903a558dbeb1e82c844fe915ab8ff25/library/std/src/panicking.rs:483:40
  26:        0x104839470 - std::panicking::try::h3e4950d3b6007524
                               at /rustc/37d7de337903a558dbeb1e82c844fe915ab8ff25/library/std/src/panicking.rs:447:19
  27:        0x104839470 - std::panic::catch_unwind::hfb8e30fa351e41e3
                               at /rustc/37d7de337903a558dbeb1e82c844fe915ab8ff25/library/std/src/panic.rs:137:14
  28:        0x104839470 - std::rt::lang_start_internal::hcb71a3757dfca71d
                               at /rustc/37d7de337903a558dbeb1e82c844fe915ab8ff25/library/std/src/rt.rs:148:20
  29:        0x10426b398 - _main
```

Tiwalun, 2023-01-03:
> The RISC-V error message is confusing, because I'm not asking for anything RISC-V related, nor am I asking for JTAG. I'm simply asking for a info on connected chips. I would expect only successes to be reported.

The tool cannot know what kind of chip is connected, so I think it's completly reasonable to print that message. If you would have connected a RISC-V chip, that message is very helpful to diagnose that it is not able to discover any RISC-V based chips. Assuming everyone is only interested in ARM chips and SWD would make it unusable for RISC-V.

noppej, 2023-01-03:
I am in agreement, this is probably just a matter of personal preference, but perhaps something like "No RISC-V connected devices found." would feel more like information, rather than an error.

Tiwalun, 2023-01-03:
I think there is a difference between "No RISC-V devices found", and "Unable to check for RISC-V devices, because we cannot use JTAG with this probe". I think it's worth it to handle these cases in a different way, but I'm very open to improve the output if you think the current one sounds too much like an error.

In general, I think it's hard for these cases to know if it's an user error (wrong cable connection), a probe-rs error, or if it's a setup where the info command just can't work.

noppej, 2023-01-03:
I've just looked back at the output. What you have makes perfect sense, when the `info` command completes.

The fact that it panicked made me feel like the RISC-V info message was an error (or at least a contributor), which is why I was confused.

If we fix the panic, then I think the output is fine :)

commit 448238fd (2023-01-03): Merge #1348

1348: Don't panic when initialization fails r=Yatekii a=Tiwalun

Fix #1347
PR #1348 (2023-01-03): Don't panic when initialization fails
Fix #1347
closed 2023-01-03 by bors[bot]