probe-rs/probe-rs #1349: `probe-rs-cli info` fails to connect to Nucleo-STM32H745ZI-Q via onboard STLink 3

noppej, 2023-01-04:
When running from master, I get 
```
probe-rs-cli info                                                                            
Probing target via JTAG
No JTAG interface was found on the connected probe. Thus, RISC-V info cannot be printed.
Probing target via SWD
No JTAG interface was found on the connected probe. Thus, RISC-V info cannot be printed.
```

The same command on the released probe-rs v0.13 gives the following
```
probe-rs-cli info                 

Available Access Ports:
Class1RomTable(
    ComponentId {
        component_address: 0xe00fe000,
        class: RomTable,
        peripheral_id: PeripheralID {
            REVAND: 0x000000,
            CMOD: No,
            REVISION: 0x000000,
            JEP106: Some(
                JEP106Code({ cc: 0x00, id: 0x20 } => Some("STMicroelectronics")),
            ),
            PART: 0x000450,
            SIZE: 0x000001,
            dev_type: 0x000001,
            arch_id: 0x000000,
        },
    },
    RomTable {
        entries: [
            RomTableEntry {
                power_domain_id: 0x000000,
                power_domain_valid: false   ,
                format: true    ,
                component: CoresightComponent {
                    component: Class1RomTable(
                        ComponentId {
                            component_address: 0xe00ff000,
                            class: RomTable,
                            peripheral_id: PeripheralID {
                                REVAND: 0x000000,
                                CMOD: No,
                                REVISION: 0x000000,
                                JEP106: Some(
                                    JEP106Code({ cc: 0x04, id: 0x3b } => Some("ARM Ltd")),
                                ),
                                PART: 0x0004c7,
                                SIZE: 0x000001,
                                dev_type: 0x000001,
                                arch_id: 0x000000,
                            },
                        },
                        RomTable {
                            entries: [
                                RomTableEntry {
                                    power_domain_id: 0x000000,
                                    power_domain_valid: false   ,
                                    format: true    ,
                                    component: CoresightComponent {
                                        component: GenericIPComponent(
                                            ComponentId {
                                                component_address: 0xe000e000,
                                                class: GenericIPComponent,
                                                peripheral_id: PeripheralID {
                                                    REVAND: 0x000000,
                                                    CMOD: No,
                                                    REVISION: 0x000000,
[... 163 lines cut ...]
                                REVAND: 0x000000,
                                CMOD: No,
                                REVISION: 0x000004,
                                JEP106: Some(
                                    JEP106Code({ cc: 0x04, id: 0x3b } => Some("ARM Ltd")),
                                ),
                                PART: 0x000906,
                                SIZE: 0x000001,
                                dev_type: 0x000014,
                                arch_id: 0x000000,
                            },
                        },
                    ),
                    ap: MemoryAp {
                        address: ApAddress {
                            dp: Default,
                            ap: 0x000000,
                        },
                    },
                },
            },
        ],
    },
)
```

Tiwalun, 2023-01-04:
This seems to be a combination of a misleading error message and an actual bug. The JTAG specific stuff can probably be ignored.

Can you please upload the log output, e.g. with `RUST_LOG=trace`?

Tiwalun, 2023-01-04:
Not sure what exactly fails here. The output can be fixed to show the actual error, that's a minor thing.

It could be that the added JTAG support for ARM chips messes up something, AFAIK that's something new, and we didn't have that in 0.13.0. Might be worth a try to specify SWD using `--protocol SWD` and check if that helps.

If there is actually a problem with the ST-Link, I probably can't help a lot, since I don't have the necessary hardware.

noppej, 2023-01-04:
Adding the protocol specifier didn't help. Do you remember off-hand who worked on the JTAG stuff?

Tiwalun, 2023-01-04:
I remember this PR: https://github.com/probe-rs/probe-rs/pull/1087, but I don't know if anything was done with ST-Link and JTAG specifically.

Do the other commands work? The info command is quite different from most other things, so maybe only that is broken.

Yatekii, 2023-01-04:
> I remember this PR: https://github.com/probe-rs/probe-rs/pull/1087, but I don't know if anything was done with ST-Link and JTAG specifically.

ST-Link does not have a JTAG interface, so it hit's the branch where no riscv interface can be found and the error message is the same, no matter if the interface is unsupported or if the protocol was an unsupported one.

Tiwalun, 2023-01-04:
> ST-Link does not have a JTAG interface, so it hit's the branch where no riscv interface can be found and the error message is the same, no matter if the interface is unsupported or if the protocol was an unsupported one.

According to @noppej's log file, it seems that JTAG gets selected, and then we try to use JTAG to communicate with the chip. I don't know if that works or if the log is even correct, but it seems that we try both JTAG and SWD.

noppej, 2023-01-04:
> Do the other commands work? The info command is quite different from most other things, so maybe only that is broken.

@Tiwalun , the `list`, `download`, and `debug` commands all work.

PR #1471 (2023-01-30): Fix: Support reading banked DP register on ST-Link
`probe-rs-cli info` was updated to also read the DPIDR register, which led to errors with ST-Link probes.

Newer ST-Link firmware actually supports reading this register, so this PR adds a firmware version check.

This should fix #1349.
closed 2023-01-30 by Yatekii