probe-rs/probe-rs #1353: RTT output seems to freeze

TheButlah, 2023-01-05:
**Describe the bug**
You can see here that the output starts in a frozen state randomly, or that the output will still show old data.
[Recording](https://asciinema.org/a/efl9G9Ua6zaDhjyrUBjk7cOFc)
[Minimally reproducible example](https://github.com/TheButlah/bugs/tree/0667442c7573df8883f10afad9e51f20af6c259d)

**To Reproduce**
Steps to reproduce the behavior:

1. Use the linked minimally reproducible example
2. Plug in an esp32c3
3. Run `cargo embed` at least once.
4. run `cargo embed` or `cargo embed monitor` in various orders until you observe the bug.

**Expected behavior**
Output should not show old data, and should not freeze at boot.

**Stacktrace**
Cannot do `RUST_LOG` and `cargo embed`'s tui at the same time.

**Desktop (please complete the following information):**
Windows 11 pro, build 22623.875

**Additional context**
I was going to post this in #1336 but this seemed like a different issue.

Yatekii, 2023-01-05:
This is most likely because the RTT buffers are not initialized yet. Does this happen in release mode too?
What you could try is introduce a delay in `cargo-embed` between the reset and the RTT start.

Yatekii, 2023-01-05:
Do you have a reset line connected? It can happen that the buffers do not get reset properly and we read them before they are initialized properly and they still have the contents of the old run. This happens if the reset line is not connected and it only does a soft or no reset at all where RAM does not get cleared.

TheButlah, 2023-01-05:
The esp32c3 supports onboard JTAG over usb. So I presume that its reset is hard wired into the chip itself.
See section 29.3.4 of the [technical reference manual](https://www.espressif.com/sites/default/files/documentation/esp32-c3_technical_reference_manual_en.pdf).

I also tried this with a two second delay at the start of the program, and it still happens.

Yatekii, 2023-01-05:
I wrote that driver and I am pretty sure reset is not working properly.

TheButlah, 2023-01-05:
>  I am pretty sure reset is not working properly.
In the software driver or on my board? I can show you the schematics if that is helpful. But basically the D+ and D- lines are directly connected to GPIO18/19 of the esp32-c3-mini

Any other info I can provide to help you reproduce?

bugadani, 2024-04-03:
My guess is that this is a duplicate of #1374

bugadani, 2025-05-06:
I believe this issue has been fixed.

closed 2025-05-06 by bugadani