probe-rs/probe-rs #1374: Host CPU usage increases over time

9names, 2023-01-14:
**Describe the bug**
When printing via defmt once per millisecond, CPU usage on the host machine grows.
Print output starts super smooth, but updates are clearly taking longer over time.
Eventually defmt can't print in time and the defmt stream is corrupted

**To Reproduce**
Steps to reproduce the behavior:
1. Set up a target to print rapidly:
```rust
let mut counter = 0;
    loop {
        delay.delay_ms(1u32);
        debug!("{}", counter);
        counter += 1;
    }
```
2. 
```system
cargo embed --release
```
3. Watch CPU usage of cargo-embed grow by about 3% per second
4. No more messages printed once CPU usage hits 100%

**Expected behavior**
cargo-embed CPU usage should not be increasing over time

**Desktop (please complete the following information):**
 - Linux - Ubuntu 22.10

bugadani, 2024-03-30:
I'm not sure because it wasn't completely intended, but this issue may have been fixed by #2249

closed 2024-03-30 by bugadani

9names, 2024-03-30:
Yep, does appear to be fixed in my testing too. Thanks @bugadani!