probe-rs/probe-rs #1377: Allow ANSI (color) escape sequences in RTT output

arbrauns, 2023-01-14:


bugadani, 2024-03-30:
Since #2323 cargo embed strips the ANSI sequences from its output. I know this isn't the solution you are looking for, but getting sequences right with line wrapping is pretty tricky.

chrysn, 2025-04-27:
For context, the [current code](https://github.com/probe-rs/probe-rs/blob/3ddfd10870619fd104e7ed5999e2c5222749405d/probe-rs-tools/src/bin/probe-rs/cmd/cargo_embed/rttui/tab.rs#L124) says:

                // We strip ANSI sequences because they interfere with text wrapping.
                //  - It's not obvious how we could tell defmt_parser to not emit ANSI sequences.
                //  - Calling textwrap on a string with ANSI sequences may break a sequence into
                // multiple lines, which is incorrect.
                //  - We can only interpret the sequences by emitting ratatui span styles, but at
                // that point we can no longer wrap the text using textwrap.
                //  - Leaving sequences in the output intact is just a bad experience.

I've experimented a bit with using the [ansi_to_tui](https://docs.rs/ansi-to-tui/latest/ansi_to_tui/) crate -- that converts the ANSI codes into ratatui and works with the current code (in app.js. the List::new gets the IntoText output passed into a Paragraph::new), and while that can be made to line-wrap, the current scrolling logic is meant to deal with pre-wrapped lines. As the current code manually implements a scrolling view (for which there is [a crate](https://docs.rs/tui-widgets/0.4.1/tui_widgets/scrollview/struct.ScrollView.html#method.render_widget) too), I'm looking into whether there there is a working design that delegates both the scrolling and the ANSI code handling to the ready crates (and might be an easier change than fixing just this issue).