probe-rs/probe-rs #1387: Is there a way to direct RTT to stdout rather than rttui?

noppej, 2023-01-14:
Hi. I'm trying to integrate cargo-embed (with both RTT and GDB) into a VSCode debug extension. Without the RTT output, it behaves rather well, but as soon as I turn on RTT, it causes a problem.   

The rttui/app.rs makes calls to `crossterm::enable_raw_mode()`, which causes a panic  
` code: 102, kind: Other, message: "Operation not supported on socket"`

Is there a config (or other) way will allow me to bypass rttui, and have rtt output go directly to `stdout`?

Yatekii, 2023-09-12:
This is possible with `probe-rs run` and `rtthost`. Do we consider this done?

noppej, 2023-09-12:
Can rtt-host do this also?

Yatekii, 2023-09-12:
rtthost does nothing else rhan print to stdout :)

bugadani, 2024-03-30:
We can keep this issue open for a while. In my mind, both `probe-rs run/attach` and `cargo embed` should use the same RTT display implementation, with maybe different defaults (I think the current state of the world [probe-rs printing to stdout, cargo-embed using rttui] is not a terrible approach for now).

Yatekii, 2024-03-30:
I think vscode has full RTT integration anyways and everything asked in this issue is possible.

bugadani, 2024-03-30:
Yeah I kind of intended to repurpose the issue :) While the original question has been resolved, the question in the title has not: there's no way to redirect output currently. Soo, hijacking :)

Yatekii, 2024-04-27:
The output can be sent to a socket thanks to #2126