probe-rs/probe-rs #1404: Attempted to evaluate a Variable with no type or no memory location

Eplankton, 2023-01-16:
Try to debug with a variable of matrix type, which defined as following:
```
pub struct Matrix<T, const R: usize, const C: usize> {
    pub raw: [[T; C]; R],
}
```
Though everything seems to be just fine and the debugger ran smoothly, I got a warning says ` Error: This is a bug! Attempted to evaluate a Variable with no type or no memory location ` , which stop me from checking for internal value of certain structure element:
[image]

By the way, I find that sometimes during the debugging with `F11 single-step into`will accidentally jump into some lib files like ` gpio.rs ` while I don't expect to happen.

noppej, 2023-02-01:
@Eplankton If you feel up to it, you can try the fix in the #1484 branch. I tested it with ...
```
pub struct Matrix<T, const X: usize, const Y: usize, const Z: usize> {
    pub raw: [[[T; Y]; X]; Z],
}
```
and 
```
    let three_d_usize_array = Matrix {
        raw: [
            [[0, 1, 2], [3, 4, 5]],
            [[6, 7, 8], [9, 10, 11]],
            [[12, 13, 14], [15, 16, 17]],
            [[18, 19, 20], [21, 22, 23]],
        ],
    };
    let three_d_string_array = Matrix {
        raw: [
            [["Apple", "Banana", "Cherry"], ["Dog", "Elephant", "Fish"]],
            [
                ["Guitar", "Horse", "Ice Cream"],
                ["Jaguar", "Kangaroo", "Lion"],
            ],
            [["Moon", "Newton", "Owl"], ["Pencil", "Queen", "Rainbow"]],
            [
                ["Sun", "Tree", "Umbrella"],
                ["Violin", "Watch", "Xylophone"],
            ],
            [["Yellow", "Zebra", "Alpha"], ["Bravo", "Charlie", "Delta"]],
            [["Echo", "Foxtrot", "Golf"], ["Hotel", "India", "Juliet"]],
        ],
    };
```

closed 2023-02-03 by Yatekii

Eplankton, 2023-02-03:
As far as I had tested on the newest #1484 fix (which has been merged into master branch), everything‘s fine. 👌
Thanks for your work !