probe-rs/probe-rs #1407: DA14531 chip support

ImUrX, 2023-01-18:
I don't have any experience on what would be needed, I see that probe-rs uses CMSIS packs for getting kind of the gist of the chip but it also has a flashing method which I wouldn't know how to implement.
[There is an open-source implementation](https://github.com/ezflash/ezFlashCLI) on flashing Dialog Semiconductors Smartbond devices. Maybe that helps

Tiwalun, 2023-01-18:
We extract the flash algorithm from the CMSIS-pack, so if there is one available, we could use that. You can use https://developer.arm.com/tools-and-software/embedded/cmsis/cmsis-search to search for a pack, and then use the `target-gen` tool to generate the YAML files:

```
cargo run -p target-gen -- <pack_file>.zip out
```

See https://github.com/probe-rs/probe-rs/tree/master/target-gen for more info about target-gen.

ImUrX, 2023-01-18:
I tried running it and it didn’t include any flashing method, so it’s something that will have to get added in the code ;-;

ImUrX, 2023-01-18:
So I gave an empty flashing method and this still happens
```
$ probe-rs-cli info
Probing target via JTAG
ERROR probe_rs::probe::jlink::arm: Unexpected DAP response: 7
ERROR probe_rs::probe::jlink::arm: Unexpected DAP response: 7
ERROR probe_rs::probe::jlink::arm: Unexpected DAP response: 7
ERROR probe_rs::probe::jlink::arm: Unexpected DAP response: 7
ERROR probe_rs::probe::jlink::arm: Unexpected DAP response: 7
ERROR probe_rs::probe::jlink::arm: Unexpected DAP response: 7
ERROR probe_rs::probe::jlink::arm: Unexpected DAP response: 7
ERROR probe_rs::probe::jlink::arm: Unexpected DAP response: 7
ERROR probe_rs::probe::jlink::arm: Unexpected DAP response: 7
ERROR probe_rs::probe::jlink::arm: Unexpected DAP response: 7
ERROR probe_rs::probe::jlink::arm: Unexpected DAP response: 7
ERROR probe_rs::probe::jlink::arm: Unexpected DAP response: 7
ERROR probe_rs::probe::jlink::arm: Unexpected DAP response: 7
ERROR probe_rs::probe::jlink::arm: Unexpected DAP response: 7
ERROR probe_rs::probe::jlink::arm: Unexpected DAP response: 7
ERROR probe_rs::probe::jlink::arm: Unexpected DAP response: 7
ERROR probe_rs::probe::jlink::arm: Unexpected DAP response: 7
ERROR probe_rs::probe::jlink::arm: Unexpected DAP response: 7
ERROR probe_rs::probe::jlink::arm: Unexpected DAP response: 7
ERROR probe_rs::probe::jlink::arm: Unexpected DAP response: 7
ERROR probe_rs::probe::jlink::arm: Unexpected DAP response: 7
ERROR probe_rs::probe::jlink::arm: Unexpected DAP response: 7
ERROR probe_rs::probe::jlink::arm: Unexpected DAP response: 7
ERROR probe_rs::probe::jlink::arm: Unexpected DAP response: 7
ERROR probe_rs::probe::jlink::arm: Unexpected DAP response: 7
ERROR probe_rs::probe::jlink::arm: Unexpected DAP response: 7
ERROR probe_rs::probe::jlink::arm: Unexpected DAP response: 7
ERROR probe_rs::probe::jlink::arm: Unexpected DAP response: 7
ERROR probe_rs::probe::jlink::arm: Unexpected DAP response: 7
ERROR probe_rs::probe::jlink::arm: Unexpected DAP response: 7
Error showing ARM chip information: An error occured in the communication with an access port or debug port.
ERROR probe_rs_cli::info: Error: The version '15' of the debug transport module (DTM) is currently not supported.    
Probing target via SWD
Error showing ARM chip information: An error occured in the communication with an access port or debug port.
```

Tiwalun, 2023-01-18:
Please try running `cargo run -p probe-rs-cli -- debug --chip cortex-m0`. Adding a new target is not really necessary, just using a generic description should be enough to work.

The *info* command can sometimes be a bit misleading, because it tries to autodetect the chip. This can sometimes put the chip in a weird state, so trying another command might work better.

ImUrX, 2023-01-18:
Yeah that worked, what should I do now?

Tiwalun, 2023-01-19:
Without flash algorithm, you will be able to debug the chip, but you can't flash any new software to it.

Unfortunately, if there is no flash algorithm in the CMSIS pack, the only solution is to either write it manually, or find another source.

probe-rs can work with flash algorithms which follow this specification: https://open-cmsis-pack.github.io/Open-CMSIS-Pack-Spec/main/html/flashAlgorithm.html
An example for a flash algorithm written in Rust can be found here: https://github.com/rp-rs/flash-algo