probe-rs/probe-rs #1420: Add support for EFM32TG11B family

n8tlarsen, 2023-01-24:
**Is your feature request related to a problem? Please describe.**
There is no current support for efm32tg11b. The end goal is to connect to an efm32tg11b120 device via J-Link probe and probe-run.

**Describe the solution you'd like**
Help in deciphering the error messages I'm seeing. Target-gen is giving several warnings (I assume one per device in the pack) saying
```
[WARN] Failed to parse flash algorithm.
[WARN] Reason: Invalid magic number: 0x6576206c6d783f3c
```
Whenever I try to connect with probe-run (after installing with the patched probe-rs and new target yaml) it gives me the following message:
```
    Finished dev [optimized + debuginfo] target(s) in 0.04s
     Running `probe-run --chip EFM32TG11B120F128GM64 target/thumbv6m-none-eabi/debug/xxxxxxxxxxx
(HOST) INFO  flashing program (8 pages / 8.00 KiB)
(HOST) INFO  success!
Error: Error communicating with probe: A core architecture specific error occurred

Caused by:
    0: A core architecture specific error occurred
    1: Failed to write register CSW at address 0x00000000
    2: An error specific to the selected architecture occurred
    3: Target device responded with a FAULT response to the request.
```

**Describe alternatives you've considered**
Connecting to the target device using the J-Link flasher works fine, so no hardware issue here. I haven't tried arm-none-eabi-gdb yet, but I'd like to use defmt-rtt and probe-run so I don't think that would help.

**Additional context**
The device family pack can be found [here](https://www.silabs.com/documents/public/cmsis-packs/SiliconLabs.GeckoPlatform_EFM32TG11B_DFP.4.2.0.pack). Here's the generated yaml: 
[EFM32TG11B_Series.yaml.txt](https://github.com/probe-rs/probe-rs/files/10494866/EFM32TG11B_Series.yaml.txt). The project I'm compiling and attempting to run is based on [rtic-rs/defmt-app-template](https://github.com/rtic-rs/defmt-app-template).

Any help would be greatly appreciated!

Tiwalun, 2023-01-25:
The warning means that we try to extract a flash algorithm from a file which is not an ELF file. Looking at the pack, they seem to have flash algorithms in the *IAR* style, which we don't support. Would be nice to filter this out, but for your case the warning can be ignored. 

There seems to be a working flash algorithm in the generated yaml, and according to the log it seems to work.
It seems that something goes wrong after flashing. 

Please try running it again with the `-vvv`flags, that should enable more debug output. And which version of probe-run are you using?

Tiwalun, 2023-01-25:
The warning message can not be avoided currently, the necessary information is not exposed by the cmsis-pack crate right now. See https://github.com/pyocd/cmsis-pack-manager/issues/203.

n8tlarsen, 2023-01-25:
I've tried probe-run v0.3.4, v0.3.5, and just tried v0.3.6 which was released yesterday. The `-vvv` output is over 5000 lines long so here's the tail end:
<details>
  <summary>Click to expand</summary>

```
(HOST) DEBUG Writing register DRW, value=DRW { data: a05f0001 }
└─ probe_rs::architecture::arm::ap @ /home/nathan/github/probe-rs/probe-rs/src/architecture/arm/ap/mod.rs:161
(HOST) TRACE Adding 2 idle cycles after transfer!
└─ probe_rs::probe::jlink::arm @ /home/nathan/github/probe-rs/probe-rs/src/probe/jlink/arm.rs:457
(HOST) DEBUG Performing 2 transfers (1 additional transfers)
└─ probe_rs::probe::jlink::arm @ /home/nathan/github/probe-rs/probe-rs/src/probe/jlink/arm.rs:484
(HOST) TRACE write 34 bytes: [cf, 0, 72, 0, ff, 83, ff, ff, ff, ff, ff, ff, f, 0, 0, 0, 0, fc, 3, ec, 82, 0, 80, 2f, d0, 0, d0, b, 0, 0, 0, 0, 0, 0]
└─ jaylink @ /home/nathan/.cargo/registry/src/github.com-1ecc6299db9ec823/jaylink-0.3.0/src/lib.rs:521
(HOST) TRACE read 16 bytes: [ff, 4, ff, ff, ff, ff, ff, ff, 10, e3, 0, 0, 80, 3, 0, 0]
└─ jaylink @ /home/nathan/.cargo/registry/src/github.com-1ecc6299db9ec823/jaylink-0.3.0/src/lib.rs:551
(HOST) DEBUG Transfer result 0: Ok(0)
└─ probe_rs::probe::jlink::arm @ /home/nathan/github/probe-rs/probe-rs/src/probe/jlink/arm.rs:336
(HOST) TRACE DAP read 454.
└─ probe_rs::probe::jlink::arm @ /home/nathan/github/probe-rs/probe-rs/src/probe/jlink/arm.rs:798
(HOST) DEBUG Transfer result 1: Ok(1c6)
└─ probe_rs::probe::jlink::arm @ /home/nathan/github/probe-rs/probe-rs/src/probe/jlink/arm.rs:336
(HOST) DEBUG Scanning at exact address: 0x20007BA0
└─ probe_rs_rtt::rtt @ /home/nathan/.cargo/registry/src/github.com-1ecc6299db9ec823/probe-rs-rtt-0.14.2/src/rtt.rs:182
(HOST) DEBUG Writing register CSW, value=CSW { DbgSwEnable: 0, HNONSEC: 1, PROT: 2, CACHE: 3, SPIDEN: 0, _RES0: 0, MTE: 0, Type: 0, Mode: 0, TrinProg: 0, DeviceEn: 0, AddrInc: Single, _RES1: 0, SIZE: U32 }
└─ probe_rs::architecture::arm::ap @ /home/nathan/github/probe-rs/probe-rs/src/architecture/arm/ap/mod.rs:161
(HOST) TRACE Adding 2 idle cycles after transfer!
└─ probe_rs::probe::jlink::arm @ /home/nathan/github/probe-rs/probe-rs/src/probe/jlink/arm.rs:457
(HOST) DEBUG Performing 2 transfers (1 additional transfers)
└─ probe_rs::probe::jlink::arm @ /home/nathan/github/probe-rs/probe-rs/src/probe/jlink/arm.rs:484
(HOST) TRACE write 34 bytes: [cf, 0, 72, 0, ff, 83, ff, ff, ff, ff, ff, ff, f, 0, 0, 0, 0, fc, 3, 8c, 2, 9, 0, 80, 31, 0, d0, b, 0, 0, 0, 0, 0, 0]
└─ jaylink @ /home/nathan/.cargo/registry/src/github.com-1ecc6299db9ec823/jaylink-0.3.0/src/lib.rs:521
(HOST) TRACE read 16 bytes: [ff, 4, ff, ff, ff, ff, ff, ff, 20, 0, 0, 0, 0, 0, 0, 0]
└─ jaylink @ /home/nathan/.cargo/registry/src/github.com-1ecc6299db9ec823/jaylink-0.3.0/src/lib.rs:551
(HOST) DEBUG Transfer result 0: Ok(0)
└─ probe_rs::probe::jlink::arm @ /home/nathan/github/probe-rs/probe-rs/src/probe/jlink/arm.rs:336
(HOST) DEBUG Transfer result 1: Err(WaitResponse)
└─ probe_rs::probe::jlink::arm @ /home/nathan/github/probe-rs/probe-rs/src/probe/jlink/arm.rs:336
(HOST) DEBUG DAP WAIT, (write), retries remaining 1000.
└─ probe_rs::probe::jlink::arm @ /home/nathan/github/probe-rs/probe-rs/src/probe/jlink/arm.rs:1150
(HOST) TRACE Adding 2 idle cycles after transfer!
└─ probe_rs::probe::jlink::arm @ /home/nathan/github/probe-rs/probe-rs/src/probe/jlink/arm.rs:457
(HOST) DEBUG Performing 1 transfers (0 additional transfers)
└─ probe_rs::probe::jlink::arm @ /home/nathan/github/probe-rs/probe-rs/src/probe/jlink/arm.rs:484
(HOST) TRACE write 20 bytes: [cf, 0, 3a, 0, ff, 83, ff, ff, ff, ff, ff, 3, 4, 2, 8, 0, 0, 80, 0, 0]
└─ jaylink @ /home/nathan/.cargo/registry/src/github.com-1ecc6299db9ec823/jaylink-0.3.0/src/lib.rs:521
(HOST) TRACE read 9 bytes: [ff, 4, ff, ff, ff, ff, ff, 0, 0]
└─ jaylink @ /home/nathan/.cargo/registry/src/github.com-1ecc6299db9ec823/jaylink-0.3.0/src/lib.rs:551
(HOST) DEBUG Transfer result 0: Ok(0)
└─ probe_rs::probe::jlink::arm @ /home/nathan/github/probe-rs/probe-rs/src/probe/jlink/arm.rs:336
(HOST) DEBUG Cleared sticky overrun bit
└─ probe_rs::probe::jlink::arm @ /home/nathan/github/probe-rs/probe-rs/src/probe/jlink/arm.rs:1167
(HOST) TRACE Adding 4 idle cycles after transfer!
└─ probe_rs::probe::jlink::arm @ /home/nathan/github/probe-rs/probe-rs/src/probe/jlink/arm.rs:457
(HOST) DEBUG Performing 2 transfers (1 additional transfers)
└─ probe_rs::probe::jlink::arm @ /home/nathan/github/probe-rs/probe-rs/src/probe/jlink/arm.rs:484
(HOST) TRACE write 34 bytes: [cf, 0, 74, 0, ff, 83, ff, ff, ff, ff, ff, ff, 3f, 0, 0, 0, 0, f0, f, 8c, 2, 9, 0, 80, 31, 0, 40, 2f, 0, 0, 0, 0, 0, 0]
└─ jaylink @ /home/nathan/.cargo/registry/src/github.com-1ecc6299db9ec823/jaylink-0.3.0/src/lib.rs:521
(HOST) TRACE read 16 bytes: [ff, 4, ff, ff, ff, ff, ff, ff, 80, 0, 0, 0, 0, 0, 0, 0]
└─ jaylink @ /home/nathan/.cargo/registry/src/github.com-1ecc6299db9ec823/jaylink-0.3.0/src/lib.rs:551
(HOST) DEBUG Transfer result 0: Ok(0)
└─ probe_rs::probe::jlink::arm @ /home/nathan/github/probe-rs/probe-rs/src/probe/jlink/arm.rs:336
(HOST) DEBUG Transfer result 1: Err(WaitResponse)
└─ probe_rs::probe::jlink::arm @ /home/nathan/github/probe-rs/probe-rs/src/probe/jlink/arm.rs:336
(HOST) DEBUG DAP WAIT, (write), retries remaining 999.
└─ probe_rs::probe::jlink::arm @ /home/nathan/github/probe-rs/probe-rs/src/probe/jlink/arm.rs:1150
(HOST) TRACE Adding 2 idle cycles after transfer!
└─ probe_rs::probe::jlink::arm @ /home/nathan/github/probe-rs/probe-rs/src/probe/jlink/arm.rs:457
[... cut ...]
└─ probe_rs::session @ /home/nathan/github/probe-rs/probe-rs/src/session.rs:727
(HOST) TRACE -- debug_core_stop;
└─ probe_rs::session @ /home/nathan/github/probe-rs/probe-rs/src/session.rs:727
Error: Error communicating with probe: A ARM specific error occured.

Caused by:
    0: A ARM specific error occured.
    1: Error using access port
    2: Failed to write register CSW at address 0x00000000
    3: An error occured in the communication with an access port or debug port.
    4: Target device responded with a FAULT response to the request.
```

</details>

n8tlarsen, 2023-01-26:
I was able to confirm today that openOCD and GDB are able to connect to the device and debug without issue. I did notice when attempting to use probe-run, that after a failed attempt to connect, I must erase the contents of the flash to gain a connection again with openOCD. It seems like something probe-run/rs is doing is causing a real issue with the device.

Tiwalun, 2023-02-27:
From a quick look into the manual of the chip, it might be that it needs some additional steps to enable debugging. Some chips have a feature where debug access can be disabled, and some special steps are needed to enable it again.
This is not really possible to test without the actual chip, so you would probably have to do that yourself.

There is code for some chips in [probe-rs/src/architecture/sequences](https://github.com/probe-rs/probe-rs/tree/master/probe-rs/src/architecture/arm/sequences), which can be used as a starting point.

It might also be something else. The most helpful log would be the output from `probe-rs-cli`, which can be installed using `cargo install probe-rs-cli`. You can then run `RUST_LOG=trace probe-rs-cli run --log-file log.json` (with the approriate flags for your target and binary), and upload the log using http://logs.probe.rs/

PR #1549 (2023-02-28): Add support for EFM32TG11B family
Closes #1420. Not sure what changed, but after rebuilding with version 0.16.0 and the new target yaml everything works as expected.
closed 2023-02-28 by Tiwalun