probe-rs/probe-rs #1469: `FlashProgress::StartedErasing` maynot be emitted in all setups

ryan-summers, 2023-01-30:
**Describe the bug**
When `do_chip_erase` is enabled in the `DownloadOptions` during programming, the `FlashProgress::StartedErasing` event is not emitted during the flash process.

**To Reproduce**
Steps to reproduce the behavior:
1. Program the STM32H7 with `do_chip_erase` enabled in the options, log all `FlashProgress` events.
2. Notice that `StartedErasing` is skipped, but `FinishedErasing` is emitted.

**Expected behavior**
The `FlashProgress::StartedErasing` event should be emitted before the `FlashProgress::FinishedErasing` event is generated

**Stacktrace**
N/A

**Additional context**

Log of events:
```
Programming probe configured. Protocol speed: 3300 kHz
Flashing part 1/1 with 5793728 bytes to target
Finished erasing
Started filling
Finished filling
Started programming
```

There is a noticable delay between the "Flashing part 1/1 message (which then kicks off the `FlashLoader::commit()`) and the "Finished erasing", which indicates the chip definitely did erase (It's about 20 seconds)

Yatekii, 2023-01-30:
Yeah this might be a bug in probe-rs and I observed this before I think. Your chip must be huge if it takes 20seconds =)

ryan-summers, 2023-01-30:
The STM32H7 has 2MB of flash, and I think this is just half. I think in reality it's about 17 seconds :)

closed 2023-01-30 by Yatekii

ryan-summers, 2023-02-03:
@Yatekii The referenced PR made it so `FlashProgress::FinishedErasing` is _also_ no longer generated in my usecase (i.e. neither start nor stop is generated)

Yatekii, 2023-02-03:
I don't understand why that appens tho :/

PR #1470 (2023-01-30): probe-rs: Emit chip erase finished/failed event
Fixes #1469
commit 261abd79 (2023-01-30): probe-rs: Emit chip erase finished/failed event (#1470)

Fixes #1469
PR #1496 (2023-02-04): Fixing erase progress event emission
This PR fixes #1469 ( again) by updating emission of the flash erase start/stop events to not depend on the presence of the erase debug sequence.

CC @Yatekii
commit 1b50829f (2023-02-04): Fixing erase progress event emission (#1496)

This PR fixes #1469 (again) by updating emission of the flash erase
start/stop events to not depend on the presence of the erase debug
sequence.

---------