probe-rs/probe-rs #1477: Embed.toml: add option to set default binary

tgross35, 2023-01-31:
**Is your feature request related to a problem? Please describe.**

Sometimes crates have >1 binary, but it's common to work with only one at a time.

**Describe the solution you'd like**

Under the `[default.general]` section of `Embed.toml`, add a `default_bin` or `default_binary` key that should get used by default. If `--bin foo` is specified by the CLI, ignore this value

**Describe alternatives you've considered**

You can currently pass `--bin foo`, but it's just an added step. I haven't found a way to default this via configuration

jukrieger, 2024-04-03:
@bugadani This also seems like not too hard of a problem to fix. You can assign this to me as well if you like.

jukrieger, 2024-04-05:
@bugadani I've been trying to think of a better way to map the output stream generated by `cargo build`'s stderr to an Error, since I think it would be better for diagnostics if the config file is pointed out as the error source. Any ideas?

bugadani, 2024-04-05:
I'm not entirely sure I understand what you're trying to do. We pass `--message-format=json(-and-some-extras)` to cargo, does it even output anything to stderr, or do we receive `cargo_metadata::Message` objects? Take a look at `build_artifact` if you haven't already, it's where we process cargo's output.