probe-rs/probe-rs #1481: cargo-embed: enabling RTT results in error: ""

cstrahan, 2023-01-31:
**Describe the bug**

When ussing RTT, `cargo-embed` will error with "The system's UTC offset could not be determined".

Maybe related to this: https://github.com/time-rs/time/issues/457

**To Reproduce**

Run `cargo-embed` with a configuration including something like:

```toml
[default.rtt]
enabled = true
channels = [
    { up = 0, name = "Defmt Output", up_mode = "NoBlockTrim", format = "Defmt" },
]
```

**Expected behavior**

Should successfully start streaming RTT messages.

**Stacktrace**

N/A

**Desktop (please complete the following information):**

- Linux (Pop_OS 22.04 LTS)

**Additional context**

None that I can think of.

Yatekii, 2023-01-31:
This should be fixed in 0.16/master :) do you still have 0.15? :)

cstrahan, 2023-01-31:
Sorry, yes -- I was behind master by 22 commits. Didn't see any issues related issues (using the error message as a search query), so I suspected it was still an issue -- I just pulled master to work on a PR to fix the issue, and then realize the change I was going to make was already there :sweat_smile:. Sorry for the noise.

closed 2023-01-31 by cstrahan

Yatekii, 2023-01-31:
No worries :) I think there was no issue associated with it :)