probe-rs/probe-rs #1485: VsCode probe-rs debugger very slow

noppej, 2023-02-01:
### Discussed in https://github.com/probe-rs/probe-rs/discussions/1457

<div type='discussions-op-text'>

<sup>Originally posted by **BrunoMelaneo** January 29, 2023</sup>
Hey there,
Last year, I started using the probe-rs debugger v0.4.0 with VsCode and the responsiveness was acceptable.
Currently, I am migrating my development environment to a new machine and while doing so I've installed the latest version at this time which is v0.4.2.

Unfortunately, I am observing very poor responsiveness! User inputs (such as, Run, Pause, Step x, adding breakpoints, etc) may take more than 10 seconds to take effect!

Has this change in behaviour been observed from v0.4.0 to v0.4.2?
Any hints on what could be causing such different performance?

I am using the debugger with a Segger probe as well as an STlink and behaviour is similar.

Thanks in advance,
Kind regards.</div>

noppej, 2023-02-01:
@brunomelaneo I've created a new Issue from your discussion, so that we can fix it. Can you please test the branch of probe-rs-debugger from #1486 and let me know if this alleviates the performance degradation you observed.

Hint: You can install this branch as a local executable, with `cargo install --force --git https://github.com/probe-rs/probe-rs --branch vscode_fixes probe-rs-debugger`

closed 2023-02-01 by Tiwalun

BrunoMelaneo, 2023-02-02:
Could you check is the git branch exists? (... or has been pushed to the remote...)
I get the following error while running the that install command.

`$ cargo install --force --git https://github.com/probe-rs/probe-rs --branch vscode_fixes probe-rs-debugger
    Updating git repository `https://github.com/probe-rs/probe-rs`
error: failed to find branch `vscode_fixes`

Caused by:
  cannot locate remote-tracking branch 'origin/vscode_fixes'; class=Reference (4); code=NotFound (-3)
`

burrbull, 2023-02-02:
This branch is already merged
```
cargo install --force --git https://github.com/probe-rs/probe-rs probe-rs-debugger
```