probe-rs/probe-rs #1487: Problems programming RP2040 with jixin.pro CMSIS-DAP_LU

pferreir, 2023-02-02:
**Describe the bug**

I get `Failed to get information for AP ApAddress { dp: Multidrop(1002927), ap: 0 }` trying to program an RP2040 using a jixin.pro CMSIS-DAP_LU probe.

**To Reproduce**
Steps to reproduce the behavior:

1. Connect the probe to the computer and the MCU
2. Run e.g. `probe-rs-cli erase --chip RP2040`

**Expected behavior**
probe-rs-cli should be able to flash the chip (or show a meaningful error).

For instance, PyOCD shows me `pyocd.core.exceptions.TargetSupportError: RP2040 requires a debug probe with SWD sequence capability`, which seems to make more sense.

**Logs**

```
 DEBUG probe_rs::config::registry > Searching registry for chip with name RP2040
 DEBUG probe_rs::config::registry > Exact match for chip name: RP2040
 DEBUG probe_rs::config::registry > Partial match for chip name: RP2040_SELFDEBUG
 DEBUG probe_rs::probe::cmsisdap::tools > Searching for CMSIS-DAP probes using libusb
 DEBUG probe_rs::probe::cmsisdap::tools > Found 1 CMSIS-DAP probes using libusb, searching HID
 DEBUG probe_rs::probe::cmsisdap::tools > Found 1 CMSIS-DAP probes total
 DEBUG jaylink                          > libusb 1.0.24.11584
 DEBUG jaylink                          > libusb has capability API: true
 DEBUG jaylink                          > libusb has HID access: true
 DEBUG jaylink                          > libusb has hotplug support: true
 DEBUG jaylink                          > libusb can detach kernel driver: true
 DEBUG probe_rs::probe::cmsisdap::tools > Could not open c251:f001 in CMSIS-DAP v2 mode
 DEBUG probe_rs::probe::cmsisdap::tools > Attempting to open c251:f001 in CMSIS-DAP v1 mode
 DEBUG probe_rs::probe::cmsisdap::commands > Draining probe of any pending data.
 DEBUG probe_rs::probe::cmsisdap::commands > Attempt 1 to find packet size
 DEBUG probe_rs::probe::cmsisdap::commands > Success: packet size is 64
 DEBUG probe_rs::probe::cmsisdap::commands > Configuring probe to use packet size 64
 DEBUG probe_rs::probe::cmsisdap           > Detected probe capabilities: Capabilities { swd_implemented: true, jtag_implemented: true, swo_uart_implemented: false, swo_manchester_implemented: false, atomic_commands_implemented: false, test_domain_timer_implemented: false, swo_streaming_trace_implemented: false, uart_communication_port_implemented: false, uart_com_port_implemented: false }
 DEBUG probe_rs::probe::cmsisdap           > Attaching to target system (clock = 1000kHz)
 DEBUG probe_rs::probe::cmsisdap           > Adding command to batch: Read(port=DebugPort, addr=0)
 DEBUG probe_rs::probe::cmsisdap           > 1 items in batch
 DEBUG probe_rs::probe::cmsisdap           > Attempting batch of 1 items
 DEBUG probe_rs::probe::cmsisdap           > 0 of batch of 1 items suceeded
 DEBUG probe_rs::architecture::arm::communication_interface > Selecting DP Multidrop(1002927)
 DEBUG probe_rs::probe::cmsisdap                            > Adding command to batch: Read(port=DebugPort, addr=0)
 DEBUG probe_rs::probe::cmsisdap                            > 1 items in batch
 DEBUG probe_rs::probe::cmsisdap                            > Attempting batch of 1 items
 DEBUG probe_rs::probe::cmsisdap                            > 0 of batch of 1 items suceeded
 DEBUG probe_rs::probe::cmsisdap                            > DPIDR read failed, retrying. Error: ArchitectureSpecific(NoAcknowledge)
 DEBUG probe_rs::probe::cmsisdap                            > Adding command to batch: Read(port=DebugPort, addr=0)
 DEBUG probe_rs::probe::cmsisdap                            > 1 items in batch
 DEBUG probe_rs::probe::cmsisdap                            > Attempting batch of 1 items
 DEBUG probe_rs::probe::cmsisdap                            > 0 of batch of 1 items suceeded
 DEBUG probe_rs::probe::cmsisdap                            > DPIDR read failed, retrying. Error: ArchitectureSpecific(NoAcknowledge)
 DEBUG probe_rs::probe::cmsisdap                            > Adding command to batch: Read(port=DebugPort, addr=0)
 DEBUG probe_rs::probe::cmsisdap                            > 1 items in batch
 DEBUG probe_rs::probe::cmsisdap                            > Attempting batch of 1 items
 DEBUG probe_rs::probe::cmsisdap                            > 0 of batch of 1 items suceeded
 DEBUG probe_rs::probe::cmsisdap                            > DPIDR read failed, retrying. Error: ArchitectureSpecific(NoAcknowledge)
 DEBUG probe_rs::probe::cmsisdap                            > Adding command to batch: Read(port=DebugPort, addr=0)
 DEBUG probe_rs::probe::cmsisdap                            > 1 items in batch
 DEBUG probe_rs::probe::cmsisdap                            > Attempting batch of 1 items
 DEBUG probe_rs::probe::cmsisdap                            > 0 of batch of 1 items suceeded
 DEBUG probe_rs::probe::cmsisdap                            > DPIDR read failed, retrying. Error: ArchitectureSpecific(NoAcknowledge)
 DEBUG probe_rs::probe::cmsisdap                            > Adding command to batch: Read(port=DebugPort, addr=0)
 DEBUG probe_rs::probe::cmsisdap                            > 1 items in batch
 DEBUG probe_rs::probe::cmsisdap                            > Attempting batch of 1 items
 DEBUG probe_rs::probe::cmsisdap                            > 0 of batch of 1 items suceeded
 DEBUG probe_rs::probe::cmsisdap                            > DPIDR read failed, retrying. Error: ArchitectureSpecific(NoAcknowledge)
 WARN  probe_rs::probe::cmsisdap                            > Giving up on TARGETSEL, too many retries.
 DEBUG probe_rs::probe::cmsisdap                            > Detaching from CMSIS-DAP probe
Error: Connecting to the chip was unsuccessful.

Caused by:
    Failed to get information for AP ApAddress { dp: Multidrop(1002927), ap: 0 }
```

**Desktop (please complete the following information):**

- Linux (Fedora 36)

Marcuss2, 2024-01-10:
For people who search for the same issue, this also happens on WCH-LinkE in CMSIS-DAP mode.

It can be bypassed by first running `pyocd commander --target rp2040`

The current log is as follows: 
```
$ RUST_LOG=probe_rs=trace probe-rs run target/thumbv6m-none-eabi/debug/universe-dap --speed 100 --chip rp2040                      Wed Jan 10 18:49:17 2024
DEBUG probe_rs::config::registry: Searching registry for chip with name rp2040
DEBUG probe_rs::config::registry: Exact match for chip name: RP2040
DEBUG probe_rs::config::registry: Partial match for chip name: RP2040_SELFDEBUG
 INFO probe_rs::config::target: Using sequence Arm(DefaultArmSequence(()))
DEBUG list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: Searching for CMSIS-DAP probes using nusb
DEBUG list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: failed to open usb device to check for cmsisdap interfaces: Os { code: 13, kind: PermissionDenied, message: "Permission denied" } DeviceInfo { bus_number: 4, device_address: 5, vendor_id: 0x0BDA, product_id: 0x0411, device_version: 0x0104, class: 9, subclass: 0, protocol: 3, speed: Some(Super), manufacturer_string: Some("Generic"), product_string: Some("4-Port USB 3.0 Hub"), serial_number: None, path: SysfsPath("/sys/bus/usb/devices/4-2") }
DEBUG list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: failed to open usb device to check for cmsisdap interfaces: Os { code: 13, kind: PermissionDenied, message: "Permission denied" } DeviceInfo { bus_number: 3, device_address: 1, vendor_id: 0x1D6B, product_id: 0x0002, device_version: 0x0606, class: 9, subclass: 0, protocol: 1, speed: Some(High), manufacturer_string: Some("Linux 6.6.10-arch1-1 xhci-hcd"), product_string: Some("xHCI Host Controller"), serial_number: Some("0000:11:00.3"), path: SysfsPath("/sys/bus/usb/devices/usb3") }
DEBUG list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: failed to open usb device to check for cmsisdap interfaces: Os { code: 13, kind: PermissionDenied, message: "Permission denied" } DeviceInfo { bus_number: 3, device_address: 11, vendor_id: 0xFC32, product_id: 0x0287, device_version: 0x0001, class: 0, subclass: 0, protocol: 0, speed: Some(Full), manufacturer_string: Some("JosefAdamcik"), product_string: Some("Sofle"), serial_number: None, path: SysfsPath("/sys/bus/usb/devices/3-1.4") }
DEBUG list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: failed to open usb device to check for cmsisdap interfaces: Os { code: 13, kind: PermissionDenied, message: "Permission denied" } DeviceInfo { bus_number: 1, device_address: 1, vendor_id: 0x1D6B, product_id: 0x0002, device_version: 0x0606, class: 9, subclass: 0, protocol: 1, speed: Some(High), manufacturer_string: Some("Linux 6.6.10-arch1-1 xhci-hcd"), product_string: Some("xHCI Host Controller"), serial_number: Some("0000:03:00.0"), path: SysfsPath("/sys/bus/usb/devices/usb1") }
DEBUG list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: failed to open usb device to check for cmsisdap interfaces: Os { code: 13, kind: PermissionDenied, message: "Permission denied" } DeviceInfo { bus_number: 3, device_address: 10, vendor_id: 0x046D, product_id: 0x085B, device_version: 0x0016, class: 239, subclass: 2, protocol: 1, speed: Some(High), manufacturer_string: None, product_string: Some("Logitech Webcam C925e"), serial_number: Some("0E5A2D5F"), path: SysfsPath("/sys/bus/usb/devices/3-1.2") }
DEBUG list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: failed to open usb device to check for cmsisdap interfaces: Os { code: 13, kind: PermissionDenied, message: "Permission denied" } DeviceInfo { bus_number: 1, device_address: 3, vendor_id: 0x17A0, product_id: 0x0304, device_version: 0x1003, class: 0, subclass: 0, protocol: 0, speed: Some(Full), manufacturer_string: Some("Samson Technologies"), product_string: Some("Samson Q2U Microphone"), serial_number: None, path: SysfsPath("/sys/bus/usb/devices/1-10") }
DEBUG list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: failed to open usb device to check for cmsisdap interfaces: Os { code: 13, kind: PermissionDenied, message: "Permission denied" } DeviceInfo { bus_number: 3, device_address: 12, vendor_id: 0x0BDA, product_id: 0x5411, device_version: 0x0104, class: 9, subclass: 0, protocol: 2, speed: Some(High), manufacturer_string: Some("Generic"), product_string: Some("4-Port USB 2.0 Hub"), serial_number: None, path: SysfsPath("/sys/bus/usb/devices/3-2") }
DEBUG list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: failed to open usb device to check for cmsisdap interfaces: Os { code: 13, kind: PermissionDenied, message: "Permission denied" } DeviceInfo { bus_number: 1, device_address: 2, vendor_id: 0x0B05, product_id: 0x17CB, device_version: 0x0112, class: 255, subclass: 1, protocol: 1, speed: Some(Full), manufacturer_string: Some("Broadcom Corp"), product_string: Some("BCM20702A0"), serial_number: Some("5CF370828A60"), path: SysfsPath("/sys/bus/usb/devices/1-3") }
TRACE list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools:   Interface 0: WCH CMSIS-DAP
TRACE list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: Could not read string for interface 1, skipping
TRACE list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: Could not read string for interface 2, skipping
TRACE list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: WCH-Link: CMSIS-DAP device with 3 interfaces
TRACE list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: No HID interface for CMSIS-DAP found.
[... cut ...]
TRACE probe_rs::probe::cmsisdap::commands: Transmit buffer: [00, 03]...
TRACE probe_rs::probe::cmsisdap::commands: Receive buffer: [03, 00]...
TRACE probe_rs::probe::cmsisdap::commands: Transmit buffer: [00, 01, 00]...
TRACE probe_rs::probe::cmsisdap::commands: Receive buffer: [01, 00]...
Error: Connecting to the chip was unsuccessful.

Caused by:
    0: An ARM specific error occurred.
    1: An error occurred in the communication with an access port or debug port.
    2: Target device did not respond to request.
```

Marcuss2, 2024-01-10:
> @Marcuss2 Why do you think these issues are related? The original issue seems to be that we don't properly detect a missing capability of the probe, and also pyocd can only print an error.
> 
> But your issue seems to be something completly different.

Yes, you are correct. I will create a separate issue for it.

Tiwalun, 2024-01-10:
> Yes, you are correct. I will create a separate issue for it.

Thanks! It can become a bit confusing otherwise.