probe-rs/probe-rs #1517: cacheable bit should always be set in CSW

bcantrill, 2023-02-17:
Debugging with an MCU Link and an STM32H7 target, probe-rs yields frustratingly nearly-correct data.  For example, in Hubris, `TICKS` is expected to increment 1000 times per second:

```console
$ humility readvar TICKS
humility: attached via CMSIS-DAP
TICKS (0x24001308) = [
    AtomicU32 {
        v: UnsafeCell<u32> {
            value: 0x4b1c7b
        }
    },
    AtomicU32 {
        v: UnsafeCell<u32> {
            value: 0x0
        }
    }
]
$ humility readvar TICKS
humility: attached via CMSIS-DAP
TICKS (0x24001308) = [
    AtomicU32 {
        v: UnsafeCell<u32> {
            value: 0x4b1c7b
        }
    },
    AtomicU32 {
        v: UnsafeCell<u32> {
            value: 0x0
        }
    }
]
```

These values should not be the same on an otherwise functioning system!  If we use our OpenOCD target instead, we see the values increment as we expect:

```
$ humility readvar TICKS
humility: attached via OpenOCD
TICKS (0x24001308) = [
    AtomicU32 {
        v: UnsafeCell<u32> {
            value: 0x4d5cb0
        }
    },
    AtomicU32 {
        v: UnsafeCell<u32> {
            value: 0x0
        }
    }
]
$ humility readvar TICKS
humility: attached via OpenOCD
TICKS (0x24001308) = [
    AtomicU32 {
        v: UnsafeCell<u32> {
            value: 0x4d6026
        }
    },
    AtomicU32 {
        v: UnsafeCell<u32> {
            value: 0x0
        }
    }
]
```

Moreover this works as expected on an STLink.  The problem -- as @cbiffle observed -- is that we have hit the probe-rs manifestation of https://github.com/oxidecomputer/hubris/issues/1099:  probe-rs is not setting the (arguably misnamed) "cacheable" bit in the CSW, meaning that any memory that is dirty in the L1 will in fact be read incorrectly (the H7 has a writeback L1).

Fixing this is straightforward, and seems safe on all targets (this is against the [Oxide probe-rs branch](https://github.com/oxidecomputer/probe-rs/tree/oxide-v0.12.0) but should apply relatively cleanly to main):

```diff
diff --git a/probe-rs/src/architecture/arm/ap/memory_ap/mod.rs b/probe-rs/src/architecture/arm/ap/memory_ap/mod.rs
index b6c3a217..4170aa39 100644
--- a/probe-rs/src/architecture/arm/ap/memory_ap/mod.rs
+++ b/probe-rs/src/architecture/arm/ap/memory_ap/mod.rs
@@ -293,13 +293,13 @@ impl CSW {
         // The CACHE bits are set for the following AHB access:
         //   HPROT[0] == 1   - data           access
         //   HPROT[1] == 1   - privileged     access
-        //   HPROT[2] == 0   - non-cacheable  access
-        //   HPROT[3] == 0   - non-bufferable access
+        //   HPROT[2] == 0   - non-bufferable access
+        //   HPROT[3] == 1   - cacheable      access
         CSW {
             DbgSwEnable: 0b1,
             HNONSEC: 0b1,
             PROT: 0b110,
-            CACHE: 0b11,
+            CACHE: 0b1011,
             AddrInc: AddressIncrement::Single,
             SIZE: data_size,
             ..Default::default()
diff --git a/probe-rs/src/architecture/arm/memory/adi_v5_memory_interface.rs b/probe-rs/src/architecture/arm/memory/adi_v5_memory_interface.rs
index 66021d6f..286f7138 100644
--- a/probe-rs/src/architecture/arm/memory/adi_v5_memory_interface.rs
+++ b/probe-rs/src/architecture/arm/memory/adi_v5_memory_interface.rs
@@ -99,13 +99,13 @@ where
         // The CACHE bits are set for the following AHB access:
         //   HPROT[0] == 1   - data           access
         //   HPROT[1] == 1   - privileged     access
-        //   HPROT[2] == 0   - non-cacheable  access
-        //   HPROT[3] == 0   - non-bufferable access
+        //   HPROT[2] == 0   - non-bufferable access
+        //   HPROT[3] == 1   - cacheable      access
 
         CSW {
             HNONSEC: if self.supports_hnonsec { 0 } else { 1 },
             PROT: 0b10,
-            CACHE: 0b11,
+            CACHE: 0b1011,
             AddrInc: AddressIncrement::Single,
             SIZE: data_size,
             ..Default::default()
```

Running with this fix, we see correct results when using probe-rs on an MCU Link probe (and an LPC-Link 2) against an H7 target.

cbiffle, 2023-02-17:
> manifestation of oxidecomputer/hubris#1099

FWIW, I don't think the bit is misnamed -- it corresponds to the equivalent
mpu attribute, where non-cacheable means bypass cache and cacheable means
don't. You want to hit the cache if you want to see what the processor sees.

Ideally this setting would only be set on cores with cache and compatible
AHB/AXI APs but I don't think that information is currently available, and
I don't see a way to provide it as a user through all the abstraction. So
setting it all the time is *probably* harmless? Testing on v8-M+TZ and v6-M
would provide more evidence.

Yatekii, 2023-02-17:
Thanks a lot for the digging, the fix and the writeup!

I am okay fixing it like that, but I still wonder if I am missing an edge case where you'd not want to read from the cache. But I think for our usecases it is always favorable to do it? (Except if you wanna debug caching issues maybe? :P).

bcantrill, 2023-02-20:
> Thanks a lot for the digging, the fix and the writeup!
> 
> I am okay fixing it like that, but I still wonder if I am missing an edge case where you'd not want to read from the cache. But I think for our usecases it is always favorable to do it? (Except if you wanna debug caching issues maybe? :P).

Well, if you are connected via an STLink, the STLink itself sets this bit -- presumably because ST realized that they had broken every debugger out there.  So on ST parts, this fix is only serving to make CMSIS-DAP probes (and other non-STLink probes) behave as the STLink does.  (It's not clear to what degree non-ST parts even use this bit.)  Finally, it seems really, really unlikely that any read wishes to explicitly disregard the contents of the cache:  it is asking for a non-coherent read that is not architecturally visible.  This is really only useful for determining cache occupancy -- and even then, only by comparing the result of having this bit set versus not set.

adamgreig, 2023-02-20:
For what it's worth, I'm pretty confident that bit is treated as RAZ/WI on Cortex-M3, and presumably M0 and M4 too, so should be harmless to set on those architectures. So yea, I don't see any harm in always setting it.

PR #1883 (2023-11-16): Mark all ARM memory accesses as cacheable
Fixes #1517

Patch is from that issue, thanks to @bcantrill.
closed 2023-11-16 by Yatekii