probe-rs/probe-rs #1528: J-link remote server support

tgross35, 2023-02-22:
**Is your feature request related to a problem? Please describe.**

Using a J-Link via WSL has proven a bit tricky for me. I've used `usbipd` to bridge the connections but setup is a bit weird, and building on the Windows side isn't always as straightforward as with Linux/WSL.

**Describe the solution you'd like**

J-Link remote server seems to be their recommended way to use a jlink via WSL: https://wiki.segger.com/WSL. I don't know exactly what the interface is, but it seems like probe-rs might be able to pretty easily find out if a jlink is connected to localhost via this method, and list it as a usable probe.

(this may already be possible somehow, but I haven't found a way yet)

**Describe alternatives you've considered**

Any sort of docs update for a straightforward solution would be awesome. `usbipd` is pretty workable but has some weird permissions things that make it somewhat less user friendly.

Yatekii, 2023-02-22:
This will be tricky, because the remote server has a custom protocol that I am not even sure is open and it uses the Segger DLL/drivers which are horrible and you have to pay to use them programmatically (outside of their standard tools).
Furthermore, you need a special probe for that with an RJ65 which is more expensive.

All in all, just avoid Segger products however possible.

tgross35, 2023-02-22:
Hm, I didn’t realize that part of the API was also proprietary, that’s really unfortunate. It isn’t just for their RJ45 probes, you can start a server with their USB probes (presumably their networked probes just have basically the same thing running all the time?)

Yeah I’ve never been a huge fan, we just use them as our default at work.

eflukx, 2023-07-19:
@tgross35 Excuse me for necromancing, I ran into your request by accident just now... 

I'm using `probe-run` successfully from WSL by explicitly setting the runner to `probe-run.exe` (so including `.exe`) in `.cargo/config.toml`.

WSL lets you invoke native Windows binaries no problem and even translates paths between Win64 <=> WSL seamlessly. A feature that comes in quite handy here. You do need to install `probe-run` on the Windows side of things obviously. This approach works perfectly for me without fiddling USB forwarding etc. (At firstI also looked into using the J-Link server thing, which I have been using that with other (C) tools/tool chains earlier.. but then it struck me that running `probe-run.exe` 'natively' could be an option.. and indeed it was..)

my `.cargo/config.toml`

```toml
[target.'cfg(all(target_arch = "arm", target_os = "none"))']
runner = "probe-run.exe --chip nRF9160_xxAA"

rustflags = [
  "-C", "linker=flip-link",
  "-C", "link-arg=-Tlink.x",
  "-C", "link-arg=-Tdefmt.x",
  # This is needed if your flash or ram addresses are not aligned to 0x10000 in memory.x
  # See https://github.com/rust-embedded/cortex-m-quickstart/pull/95
  "-C", "link-arg=--nmagic",
]

[build]
target = "thumbv8m.main-none-eabihf"

[alias]
rb = "run --bin"
rrb = "run --release --bin"
```