probe-rs/probe-rs #1529: probe-rs-debugger Does not halt on breakpoints when using 'launch' and not flashing.

noppej, 2023-02-23:
**Describe the bug**
probe-rs-debugger Does not halt on breakpoints when using 'launch' and not flashing.

**To Reproduce**
Steps to reproduce the behavior:

1. Using VSCode, add a `launch.json` that does uses: ` "flashingConfig": {"flashingEnabled": false}`
2. Set a breakpoint in main(), before the loop'ing part of the function.
3. Build, and manually flash the binary to the target.
4. Start a debug session with `F5`.

**Expected behavior**
I would expect the target to launch the binary app on the target, and then to halt at the breakpoint.

**Additional context**
This used to work, so I can only conclude that one of the recent updates broke this functionality. To be investigated.

mubes, 2023-02-24:
@noppej Can you see if you're getting warnings in the Debug Console as outlined in #1532 ? I saw this symptom too.

noppej, 2023-02-28:
> @noppej Can you see if you're getting warnings in the Debug Console as outlined in #1532 ? I saw this symptom too.

@mubes, as per my comment in #1532, I think the two issues are unrelated.  Feel free to DM if you want to discuss collaborating on a fix for that one.

closed 2023-03-01 by Tiwalun