probe-rs/probe-rs #1532: Cannot debug C code

mubes, 2023-02-24:
**The bug**

When debugging C I get messages in the Debug Window similar to `Cannot set breakpoint here. Try reducing compile time-, and link time-, optimization in your build configuration, or choose a different source location: Could not find valid source statements for this address. Consider using instruction level stepping.`. This is with various settings of `-g` and `-gdwarf`.

**To Reproduce**
I have added code to re-create to the 'simple' example in the [Orbmule](https://github.com/orbcode/orbmule) repository. Just open that directory in vsc and try to compile/run.  It should work with any STM32F427-ish target, it uses very few resources.

After starting the application, try to set a breakpoint, e.g. in main.  You will get the above noted error.  I think this may relate to #1529 too.

Running on Ubuntu 22.04.

TechnoMancer, 2023-02-24:
This appears to be due to gcc not emitting end of prologue instructions in the line number program debug info.

https://gcc.gnu.org/legacy-ml/gcc-patches/2011-03/msg02106.html

This mailing list exchange may be of use to whoever works on this issue.

noppej, 2023-02-24:
@TechnoMancer is correct. The current probe-rs implementation relies on the `DW_LNS_set_prologue_end` in the line program, and if it isn't there, it will never  find a 'valid' place between the prologue and end of statement group to insert a breakpoint.

As discussed in the matrix chat, it is viable to use the debug info to determine if the binary was generated from a c program, and then to use the beginning of the next source program line as a proxy for the end of prologue. 

If anyone wants to PR this change, I'd be happy to review and test to make sure existing functionality remains intact.

commit 97c09a67 (2023-03-09): Cannot debug C code
Fixes #1532
PR #1558 (2023-03-16): Experimental support for debugging binaries compiled from `.c` files.
Fixes #1532

The immediate reported problem relates to setting breakpoints. Please see #1532 for details on suspected root cause.

The strategy to fix this will be as follows:
- Identify the `DW_AT_language` attribute of the source file. We do this at source file level, and not at the binary level, so that binaries that have a mix of C and Rust should be covered also.
  - NOTE: The language check implemented in this PR is for:
    - `DW_AT_language` == `DW_LANG_C99` (and `C11` and C17)
- For Rust files, continue to employ th [... cut]
closed 2023-03-16 by noppej