probe-rs/probe-rs #1582: make probe-rs-cli run feature par with probe-run

Yatekii, 2023-04-06:
```[tasklist]
-[ ] probe-rs-cli run should support target halt and runner exit on ctrl + c
-[ ] show a backtrace on crash or ctrl + c
-[ ] stack usage analysis
```

datdenkikniet, 2023-04-19:
Would like to add a point: setting command values of command line options without having to modify the command itself (e.g. environment variables, a `.env` file or equivalent) so that using it as a runner isn't as much of a hassle if `probe-rs-cli` cannot determine the exact probe to use for an operation.

Yatekii, 2023-04-19:
I think environment variables are a bigger hassle than just commandline flags for something you just config once, no?
`.env` file would work nicely I guess :)

datdenkikniet, 2023-04-19:
Yes, that is true! Yeah I'm mostly thinking of a use-case where you're swapping out the feature set of a `cargo run` semi-often while also wanting to specify `--probe`. In those cases being able to configure the probe selector without requiring that it comes from a command line option directly may be nice! I'm impartial to env variables or some sort of `.env` file, just so long as there is a possibility :)

Yatekii, 2023-04-19:
Ohhh, now I get it, yep, that would be cool!