probe-rs/probe-rs #1585: Reporting a vulnerability

igibek, 2023-04-10:
Hello! 

I hope you are doing well! 

We are a security research team. Our tool automatically detected a vulnerability in this repository. We want to disclose it responsibly.  GitHub has a feature called **Private vulnerability reporting**, which enables security research to privately disclose a vulnerability. Unfortunately, it is not enabled for this repository. 

Can you enable it, so that we can report it? 

Thanks in advance! 

PS: you can read about how to enable private vulnerability reporting here: https://docs.github.com/en/code-security/security-advisories/repository-security-advisories/configuring-private-vulnerability-reporting-for-a-repository

Yatekii, 2023-04-10:
Thanks a lot for reporting this! I just enabled the feature :) How did you come about scanning our repo? :)

Tiwalun, 2023-05-10:
Closing this as private vulnerability reporting is now enabled.

closed 2023-05-10 by Tiwalun