probe-rs/probe-rs #1590: probe-rs-cli run RTT throughput is lower than expected

9names, 2023-04-13:
**Describe the bug**
`probe-rs-cli run` does not send RTT data to stdout as fast as probe-run

**To Reproduce**
Create a firmware that generates lots of RTT data or [use my rp2040 one](https://github.com/9names/rtt-spam)
Have some way of measuring data piped from stdout or [use the one i hacked together](https://github.com/9names/stdoutbytecount)

Measure the RTT throughput
```console
$ probe-rs-cli run --chip RP2040 --speed 20000 target/thumbv6m-none-eabi/release/rtt-spam | stdoutbytecount
     Erasing sectors ✔ [00:00:00] [] 12.00 KiB/12.00 KiB @ 88.26 KiB/s (eta 0s )
 Programming pages   ✔ [00:00:00] [] 12.00 KiB/12.00 KiB @ 96.55 KiB/s (eta 0s )Bytes/sec: 6071, average: 6071
Bytes/sec: 9984, average: 8027
Bytes/sec: 8952, average: 8335
Bytes/sec: 9072, average: 8519
Bytes/sec: 10080, average: 8831
Bytes/sec: 8928, average: 8847
Bytes/sec: 9072, average: 8879
Bytes/sec: 10080, average: 9029
Bytes/sec: 8856, average: 9010
Bytes/sec: 9072, average: 9016
Bytes/sec: 9960, average: 9405
Bytes/sec: 9072, average: 9314
Bytes/sec: 9072, average: 9326

$ probe-run --chip RP2040 --speed 20000 target/thumbv6m-none-eabi/release/rtt-spam | stdoutbytecount
(HOST) INFO  flashing program (3 pages / 12.00 KiB)
(HOST) INFO  success!
────────────────────────────────────────────────────────────────────────────────
Bytes/sec: 90720, average: 90720
Bytes/sec: 147168, average: 118944
Bytes/sec: 146160, average: 128016
Bytes/sec: 148176, average: 133056
Bytes/sec: 146160, average: 135676
Bytes/sec: 146160, average: 137424
Bytes/sec: 146160, average: 138672
Bytes/sec: 148176, average: 139860
Bytes/sec: 147168, average: 140672
Bytes/sec: 146160, average: 141220
Bytes/sec: 149184, average: 147067
Bytes/sec: 145152, average: 146865

```

tested on ubuntu + windows, which has different perf numbers but same order of magnitude

9names, 2023-04-13:
hmm. a naive check looks correct, but i'm not sure about control codes, etc.
```system
$ probe-rs-cli run --chip RP2040 --speed 20000 target/thumbv6m-none-eabi/release/rtt-spam | head -n30
     Erasing sectors ✔ [00:00:00] [] 12.00 KiB/12.00 KiB @ 85.53 KiB/s (eta 0s )
 Programming pages   ✔ [00:00:00] [] 12.00 KiB/12.00 KiB @ 95.27 KiB/s (eta 0s )    Finished in 0.28s
aaaaaaaaaaaaaaaaaaaaaa
aaaaaaaaaaaaaaaaaaaaaa
aaaaaaaaaaaaaaaaaaaaaa
aaaaaaaaaaaaaaaaaaaaaa
aaaaaaaaaaaaaaaaaaaaaa
aaaaaaaaaaaaaaaaaaaaaa
aaaaaaaaaaaaaaaaaaaaaa
aaaaaaaaaaaaaaaaaaaaaa
aaaaaaaaaaaaaaaaaaaaaa
aaaaaaaaaaaaaaaaaaaaaa
aaaaaaaaaaaaaaaaaaaaaa
aaaaaaaaaaaaaaaaaaaaaa
aaaaaaaaaaaaaaaaaaaaaa
aaaaaaaaaaaaaaaaaaaaaa
aaaaaaaaaaaaaaaaaaaaaa
aaaaaaaaaaaaaaaaaaaaaa
aaaaaaaaaaaaaaaaaaaaaa
aaaaaaaaaaaaaaaaaaaaaa
aaaaaaaaaaaaaaaaaaaaaa
aaaaaaaaaaaaaaaaaaaaaa
aaaaaaaaaaaaaaaaaaaaaa
aaaaaaaaaaaaaaaaaaaaaa
aaaaaaaaaaaaaaaaaaaaaa
aaaaaaaaaaaaaaaaaaaaaa
aaaaaaaaaaaaaaaaaaaaaa
aaaaaaaaaaaaaaaaaaaaaa
aaaaaaaaaaaaaaaaaaaaaa
aaaaaaaaaaaaaaaaaaaaaa
aaaaaaaaaaaaaaaaaaaaaa
Error: Broken pipe (os error 32)
$ probe-run --chip RP2040 --speed 20000 target/thumbv6m-none-eabi/release/rtt-spam | head -n30
(HOST) INFO  flashing program (3 pages / 12.00 KiB)
(HOST) INFO  success!
────────────────────────────────────────────────────────────────────────────────
aaaaaaaaaaaaaaaaaaaaaa
aaaaaaaaaaaaaaaaaaaaaa
aaaaaaaaaaaaaaaaaaaaaa
aaaaaaaaaaaaaaaaaaaaaa
aaaaaaaaaaaaaaaaaaaaaa
aaaaaaaaaaaaaaaaaaaaaa
aaaaaaaaaaaaaaaaaaaaaa
aaaaaaaaaaaaaaaaaaaaaa
aaaaaaaaaaaaaaaaaaaaaa
aaaaaaaaaaaaaaaaaaaaaa
aaaaaaaaaaaaaaaaaaaaaa
aaaaaaaaaaaaaaaaaaaaaa
aaaaaaaaaaaaaaaaaaaaaa
aaaaaaaaaaaaaaaaaaaaaa
aaaaaaaaaaaaaaaaaaaaaa
aaaaaaaaaaaaaaaaaaaaaa
aaaaaaaaaaaaaaaaaaaaaa
aaaaaaaaaaaaaaaaaaaaaa
aaaaaaaaaaaaaaaaaaaaaa
aaaaaaaaaaaaaaaaaaaaaa
aaaaaaaaaaaaaaaaaaaaaa
aaaaaaaaaaaaaaaaaaaaaa
aaaaaaaaaaaaaaaaaaaaaa
aaaaaaaaaaaaaaaaaaaaaa
aaaaaaaaaaaaaaaaaaaaaa
aaaaaaaaaaaaaaaaaaaaaa
aaaaaaaaaaaaaaaaaaaaaa
aaaaaaaaaaaaaaaaaaaaaa
aaaaaaaaaaaaaaaaaaaaaa
aaaaaaaaaaaaaaaaaaaaaa
Error: Broken pipe (os error 32)
```

9names, 2023-04-13:
Piping to a file then opening with a hex editor only shows what I see there too

noppej, 2023-04-14:
Looking at [this code](https://github.com/probe-rs/probe-rs/blob/47a0c5e3654819df5fdbaa9556e9ca01849145d3/cli/src/run.rs#L73-L83) I am not surprised about the throughput.  One possible solution would be to only sleep if the last poll retrieved no data.

9names, 2023-04-15:
Tested by removing the sleep, this is definitely the cause of the majority of the performance issues.
It's not going to be very efficient in a superloop like this but it is certainly fast.
I wonder if the comment about polling rtt too fast was actually about masking a bug elsewhere in the system? Won't be easy to prove without some cross-platform long-term tests.

```console
$ probe-rs-cli run --chip RP2040 --speed 20000 target/thumbv6m-none-eabi/release/rtt-spam |stdoutbytecount
     Erasing sectors ✔ [00:00:00] [] 12.00 KiB/12.00 KiB @ 73.70 KiB/s (eta 0s )
 Programming pages   ✔ [00:00:00] [] 12.00 KiB/12.00 KiB @ 98.29 KiB/s (eta 0s )Bytes/sec: 23, average: 23
Bytes/sec: 134113, average: 67068
Bytes/sec: 143865, average: 92667
Bytes/sec: 140875, average: 104719
Bytes/sec: 143313, average: 112437
Bytes/sec: 141404, average: 117265
Bytes/sec: 140576, average: 120595
Bytes/sec: 142508, average: 123334
Bytes/sec: 143014, average: 125521
Bytes/sec: 141726, average: 127141
Bytes/sec: 140990, average: 141238
Bytes/sec: 140944, average: 141921
Bytes/sec: 141243, average: 141659
```

and with mcu-link

```console
$ probe-rs-cli run --chip RP2040 --speed 30000 target/thumbv6m-none-eabi/release/rtt-spam | stdoutbytecount | head -n10
     Erasing sectors ✔ [00:00:00] [] 12.00 KiB/12.00 KiB @ 77.26 KiB/s (eta 0s )
 Programming pages   ✔ [00:00:00] [] 12.00 KiB/12.00 KiB @ 219.46 KiB/s (eta 0s )Bytes/sec: 23, average: 23
Bytes/sec: 489325, average: 244674
Bytes/sec: 510186, average: 333178
Bytes/sec: 509312, average: 377211
Bytes/sec: 505149, average: 402799
Bytes/sec: 509519, average: 420585
Bytes/sec: 509404, average: 433274
Bytes/sec: 509358, average: 442784
Bytes/sec: 508208, average: 450053
Bytes/sec: 507840, average: 455832
thread 'main' panicked at 'failed printing to stdout: Broken pipe (os error 32)', library/std/src/io/stdio.rs:1008:9
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
Error: Broken pipe (os error 32)
$ probe-run --chip RP2040 --speed 30000 target/thumbv6m-none-eabi/release/rtt-spam | stdoutbytecount | head -n10
(HOST) INFO  flashing program (3 pages / 12.00 KiB)
(HOST) INFO  success!
────────────────────────────────────────────────────────────────────────────────
Bytes/sec: 1012, average: 1012
Bytes/sec: 567387, average: 284199
Bytes/sec: 575161, average: 381186
Bytes/sec: 574793, average: 429588
Bytes/sec: 574287, average: 458528
Bytes/sec: 573712, average: 477725
Bytes/sec: 575115, average: 491638
Bytes/sec: 573114, average: 501822
Bytes/sec: 573804, average: 509820
Bytes/sec: 573666, average: 516205
thread 'main' panicked at 'failed printing to stdout: Broken pipe (os error 32)', library/std/src/io/stdio.rs:1008:9
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
Error: Broken pipe (os error 32)
```

Yatekii, 2023-04-15:
How I remember it, is that some debug probes would choke on too many packets too fast :)

PR #1694 (2023-07-25): More CLI fixes
This PR generalizes the different CLI commands and reuses code more efficiently.

Fixes #1680.
Fixes #1590.
Fixes #1695.
closed 2023-07-25 by noppej