probe-rs/probe-rs #1594: probe-rs-cli run sometimes fails to connect to RTT after successfully flashing

9names, 2023-04-15:
**Describe the bug**
Sometimes probe-rs-cli will successfully flash firmware then fail to connect to RTT.
This "sometimes" becomes "every time" on my machine when launching via [gamemoderun](https://github.com/FeralInteractive/gamemode), which should only be affecting process priority and scheduling.

My assumption is that there is not sufficient delay when connecting to RTT for the firmware to finish initialising it, at least for sufficiently fast debug hosts and with firmware that take a while to get to RTT init code.
In my test case (RP2040) the MCU has to load firmware off of SPI flash and do a bunch of other initialisation before `rtt_init_print!` can be run, so there's not much that can be done to speed things up on the MCU side.

**To Reproduce**

```console
$ gamemoderun probe-rs-cli run --chip RP2040 --speed 30000 target/thumbv6m-none-eabi/release/rtt-spam
     Erasing sectors ✔ [00:00:00] [####################################] 12.00 KiB/12.00 KiB @ 80.46 KiB/s (eta 0s )
 Programming pages   ✔ [00:00:00] [###################################] 12.00 KiB/12.00 KiB @ 254.29 KiB/s (eta 0s )    Finished in 0.201s
ERROR probe_rs_cli::run: Error attempting to attach to RTT: RTT control block not found in target memory.
- Make sure RTT is initialized on the target, AND that there are NO target breakpoints before RTT initalization.
- For VSCode and probe-rs-debugger users, using `halt_after_reset:true` in your `launch.json` file will prevent RTT 
        initialization from happening on time.
- Depending on the target, sleep modes can interfere with RTT. Continuing without RTT...
```

**Expected behavior**
RTT output should be present

**Desktop (please complete the following information):**

- Linux (Ubuntu 22.10, lowlatency kernel), AMD 7700x processor

alepez, 2023-07-19:
I'm having this issue with `probe-rs 0.19.0` (commit 02d76ba) on STM32H753ZI, while it is working fine on STM32U575ZITx.

But current master (commit e273c987) is working fin on STM32H753ZI.

alepez, 2023-07-19:
With some bisect I've found it is fixed by b2c5f05c

alepez, 2023-07-19:
So this issue seems to be fixed by PR #1675

closed 2023-07-19 by Yatekii

davidmanukian, 2023-08-26:
I think this issue is not fixed since I downloaded 0.20 version and still see this issue.

davidmanukian, 2023-08-26:
```
    Finished release [optimized] target(s) in 0.12s
     Running `probe-rs run --chip STM32F411CEUx /Users/..../rust-programming/black-pill-rust/target/thumbv7em-none-eabihf/release/demo-blinky`
     Erasing sectors ✔ [00:00:00] [###################################################################################################################################################################################################################################] 16.00 KiB/16.00 KiB @ 36.65 KiB/s (eta 0s )
 Programming pages   ✔ [00:00:00] [###################################################################################################################################################################################################################################] 11.00 KiB/11.00 KiB @ 19.01 KiB/s (eta 0s )    Finished in 1.045s
ERROR probe_rs::cmd::run: Error attempting to attach to RTT: RTT control block not found in target memory.
- Make sure RTT is initialized on the target, AND that there are NO target breakpoints before RTT initalization.
- For VSCode and probe-rs-debugger users, using `halt_after_reset:true` in your `launch.json` file will prevent RTT 
        initialization from happening on time.
- Depending on the target, sleep modes can interfere with RTT. Continuing without RTT...     
```
my probe-rs version is
```
probe-rs --version 
probe-rs 0.20.0
git commit: crates.io
```

Yatekii, 2023-08-26:
This specific issue is fixed.

You can get the same error if you have your linker file and memory setup wrong :)

Best would be to resolve this in our Matrix channel  because it has nothing to do with probe-rs but your project setup :)

davidmanukian, 2023-08-26:
Got it. Then sorry for that.
Asked my question in Matrix channel. Thanks!

Yatekii, 2023-08-26:
Don't worry, we love to help =)