probe-rs/probe-rs #1599: Failed to erase flash sector at address 0x00000000 on ATSAMD21G18A

nvsd, 2023-04-28:
**Describe the bug**
I'm unable to use cargo-flash, probe-run, and cargo-embed due to the below error while using a Jlink EDU Mini with a feather M0 proto
```
Caused by:
    0: Failed to erase flash sector at address 0x00000000.
    1: The execution of 'erase_sector' failed with code 1. This might indicate a problem with the flash algorithm.
```

**To Reproduce**
Steps to reproduce the behavior:

1. Ensure Jlink and feather board are plugged in 
2. Clone [atsamd-rs board example](https://github.com/atsamd-rs/atsamd/tree/master/boards/feather_m0)
3. Run cargo run
4. observe the error

**Expected behavior**
Able to flash and debug device

**Stacktrace**
If applicable, add a stacktrace to help explain your problem.

**Desktop (please complete the following information):**

- macOS Venture (13.2.1 (22D68)) m1

**Additional context**
I assume I have a configuration issue, but I'm unsure. It's also possible the atsamd repo is a better fit for this issue. Let me know what else I can provide.

nvsd, 2023-04-29:
I tried to use cargo flash from within the [feather_m0 board project](https://github.com/atsamd-rs/atsamd/tree/master/boards/feather_m0) and got the same error. I'm assuming this means there's a misconfiguration in that project. I'm still trying to figure out where to look for a solution, though. I'd appreciate it if anyone could point me in the right direction to get me started.

nvsd, 2023-04-29:
This isn't related to that project at all. I get the same error if I run a bare project using the feather m0 board crate and try to flash with cargo embed. This leads me to believe that something is unique about the board or that the flash algorithm is incorrect. The former is more likely, as there would be many more issues.

Yatekii, 2023-04-29:
From what i recall, the ATSAMs have a default bootloader that is locked. So you cannot write to that sector. You would have to write to 0x800 (I think) and not 0x000 and adjust for that or unlock/erase the first sector. We currently only have sequences enabled for the 5 series and not for the 2 series. You could see if you can make it work by reusing the same sequence here: https://github.com/probe-rs/probe-rs/blob/master/probe-rs/src/config/target.rs#L162 or if you can alter the sequence to make it work for both chips. You then also need to give the flash process the correct permissions to erase the sector for you :)

nvsd, 2023-04-29:
Ok, that tracks with what I've been seeing. I'm new to rust and embedded, but this is just as good of a starting point as any, so I'll give this a shot. Thanks for pointing me in the right direction.

Yatekii, 2023-04-29:
Let me know if you need more info or drop by in the #probe-rs:matrix.org channel =)

nvsd, 2023-04-30:
I found some mention of this issue and a tutorial from Adafruit that might be useful. Just dropping those here so I (or someone else) can reference them later.

[adafruit bootloader flash](https://learn.adafruit.com/proper-step-debugging-atsamd21-arduino-zero-m0/restoring-bootloader)
[segger forum issue](https://forum.segger.com/index.php/Thread/5711-SOLVED-Problem-programming-ATSAMD21G18/)

ianrrees, 2023-10-13:
I believe this is a duplicate of #507.  A workaround for SAMD21 is to tell probe-rs (via command line, Embed.toml, etc) that a part with a smaller flash is in use.  For example, if using ATSAMD21G18AU, specify ATSAMD21E15A.

bugadani, 2024-03-28:
Closing this as a duplicate of #507.

closed 2024-03-28 by bugadani