probe-rs/probe-rs #1608: SIGSEGV & SIGKILL when installing probe-rs-cli & probe-run respectivelly

elpiel, 2023-05-08:
**Describe the bug**
I am trying to install `probe-run` and `probe-rs-cli` **on** my Raspberry Pi 3 but receiving build error:

**To Reproduce**
Steps to reproduce the behavior:

1. Use Raspbian OS 64bit (target: `aarch64-linux-gnu` )
2. install rust
3. try to install `probe-rs-cli` (`0.18`) or `probe-run` (`0.3.9`)

**Expected behavior**
I should be able to install both tools

**Stacktrace**

Probe-rs-cli:

```
error: could not compile `proc-macro2`

Caused by:
  process didn't exit successfully: `rustc --crate-name build_script_build --edition=2018 /home/pi/.cargo/registry/src/github.com-1ecc6299db9ec823/proc-macro2-1.0.56/build.rs --error-format=json --json=diagnostic-rendered-ansi,artifacts,future-incompat --diagnostic-width=131 --crate-type bin --emit=dep-info,link -C embed-bitcode=no -C debug-assertions=off --cfg 'feature="default"' --cfg 'feature="proc-macro"' -C metadata=5d6942da968683d2 -C extra-filename=-5d6942da968683d2 --out-dir /tmp/cargo-installjdKTdH/release/build/proc-macro2-5d6942da968683d2 -L dependency=/tmp/cargo-installjdKTdH/release/deps --cap-lints allow` (signal: 11, SIGSEGV: invalid memory reference)
warning: build failed, waiting for other jobs to finish...
error: failed to compile `probe-rs-cli v0.18.0`, intermediate artifacts can be found at `/tmp/cargo-installjdKTdH`
```

Probe-run:
```
error: could not compile `probe-rs`

Caused by:
  process didn't exit successfully: `rustc --crate-name probe_rs --edition=2021 /home/pi/.cargo/registry/src/github.com-1ecc6299db9ec823/probe-rs-0.18.0/src/lib.rs --error-format=json --json=diagnostic-rendered-ansi,artifacts,future-incompat --diagnostic-width=131 --crate-type lib --emit=dep-info,metadata,link -C opt-level=3 -C embed-bitcode=no --cfg 'feature="builtin-targets"' --cfg 'feature="default"' --cfg 'feature="kmp"' --cfg 'feature="rtt"' -C metadata=81198dfdb67541cd -C extra-filename=-81198dfdb67541cd --out-dir /tmp/cargo-installUEBvnT/release/deps -L dependency=/tmp/cargo-installUEBvnT/release/deps --extern anyhow=/tmp/cargo-installUEBvnT/release/deps/libanyhow-f452631d14cf78b7.rmeta --extern base64=/tmp/cargo-installUEBvnT/release/deps/libbase64-3d153bde6e690d9a.rmeta --extern bincode=/tmp/cargo-installUEBvnT/release/deps/libbincode-981ba0e17561dff1.rmeta --extern bitfield=/tmp/cargo-installUEBvnT/release/deps/libbitfield-b32b2e2c0095a5c4.rmeta --extern bitvec=/tmp/cargo-installUEBvnT/release/deps/libbitvec-33131fad61069219.rmeta --extern enum_primitive_derive=/tmp/cargo-installUEBvnT/release/deps/libenum_primitive_derive-2ac1649fa701eb47.so --extern gimli=/tmp/cargo-installUEBvnT/release/deps/libgimli-86133d92752f1ead.rmeta --extern hidapi=/tmp/cargo-installUEBvnT/release/deps/libhidapi-dd8dc20029d2de0e.rmeta --extern ihex=/tmp/cargo-installUEBvnT/release/deps/libihex-beea3f620fa50c7d.rmeta --extern jaylink=/tmp/cargo-installUEBvnT/release/deps/libjaylink-22b8a93125b99fbd.rmeta --extern jep106=/tmp/cargo-installUEBvnT/release/deps/libjep106-6333e4cbcb0df172.rmeta --extern kmp=/tmp/cargo-installUEBvnT/release/deps/libkmp-ab6ee211160b258b.rmeta --extern num_traits=/tmp/cargo-installUEBvnT/release/deps/libnum_traits-f92dd618f35145ac.rmeta --extern object=/tmp/cargo-installUEBvnT/release/deps/libobject-10c19451cb69add1.rmeta --extern once_cell=/tmp/cargo-installUEBvnT/release/deps/libonce_cell-c983b9cf77f19a1d.rmeta --extern probe_rs_target=/tmp/cargo-installUEBvnT/release/deps/libprobe_rs_target-1c85afb1a912e3a1.rmeta --extern rusb=/tmp/cargo-installUEBvnT/release/deps/librusb-7ac2f98c87890a56.rmeta --extern scroll=/tmp/cargo-installUEBvnT/release/deps/libscroll-cd883fc040cbadf7.rmeta --extern serde=/tmp/cargo-installUEBvnT/release/deps/libserde-2766a275d7bb3535.rmeta --extern serde_yaml=/tmp/cargo-installUEBvnT/release/deps/libserde_yaml-2268851c51b5923d.rmeta --extern static_assertions=/tmp/cargo-installUEBvnT/release/deps/libstatic_assertions-939c52b0947bdfde.rmeta --extern svg=/tmp/cargo-installUEBvnT/release/deps/libsvg-d5333ff49672862e.rmeta --extern thiserror=/tmp/cargo-installUEBvnT/release/deps/libthiserror-0521b3d02c09e491.rmeta --extern tracing=/tmp/cargo-installUEBvnT/release/deps/libtracing-e40f4e4abb1211c7.rmeta --cap-lints allow -L native=/usr/lib/aarch64-linux-gnu -L native=/tmp/cargo-installUEBvnT/release/build/hidapi-1967d131e2f8b8ad/out -L native=/usr/lib/aarch64-linux-gnu -L native=/tmp/cargo-installUEBvnT/release/build/libusb1-sys-1d8e05aba8b08acc/out` (signal: 9, SIGKILL: kill)
error: failed to compile `probe-run v0.3.9`, intermediate artifacts can be found at `/tmp/cargo-installUEBvnT`
```

**Desktop (please complete the following information):**

- Linux (Raspbian OS)

**Additional context**

I'm not sure if I should have shared the information about `probe-run` here but since probe-rs-cli itself fails, I though it's the appropriate place to start

elpiel, 2023-05-08:
Ok, I will open this issue if the issue still originates from probe-rs

elpiel, 2023-05-08:
And thank you!

closed 2023-05-08 by elpiel

Yatekii, 2023-05-09:
No worries, I am also fine leaving this open until you resolved it. But it really smells like a systems issue and not probe-rs to me :)