probe-rs/probe-rs #1609: Error: Failed to open the debug probe.

uhinze, 2023-05-08:
**Describe the bug**
I see there are a lot of similar issues, but this one seems to be different, it's caused by `Probe was not found` and it's on Mac.

In a nutshell:
```
> probe-rs-cli list
The following devices were found:
[0]: DAPLink CMSIS-DAP (VID: 0d28, PID: 0204, Serial: 102636021305473600000000000000000000000097969902, CmsisDap)

> probe-rs-cli download --chip nRF52840_xxAA --format hex my_file.hex
Error: Failed to open the debug probe.

Caused by:
    0: Probe could not be created
    1: Probe was not found.
```

**To Reproduce**
Steps to reproduce the behavior:

1. install probe-rs-cli (but same issue with probe-rs-debugger)
2. connect nRF52840 to Mac
3. `probe-rs-cli download --chip nRF52840_xxAA --format hex my_file.hex` or any other command connecting to the device

**Expected behavior**
Connect to the device, flash file

**My current troubleshooting**
(forgive beginner's mistakes, I'm new to Rust)

I've debugged the execution of probe-rs-cli to https://github.com/probe-rs/probe-rs/blob/master/probe-rs/src/probe/cmsisdap/tools.rs#L356 and following. From my understanding, what happens here is that the interface number of the device is 3 (the HID interface). But for some reason, the interface number of my device above is 0. Not sure where this comes from though.

When I comment out this section (L356-L360), the connection works and I can also run my code again using the VS Code debugger. The only thing that doesn't work now is setting breakpoints, for some reason, so I'm not 100% happy.

**Stacktrace**
<details>

```

 INFO probe_rs_cli: Writing log to "/Users/uhinze/Library/Application Support/rs.probe-rs.probe-rs/1683551163501.log"
DEBUG probe_rs::config::registry: Searching registry for chip with name nRF52840_xxAA
DEBUG probe_rs::config::registry: Exact match for chip name: nRF52840_xxAA
 WARN probe_rs::config::target: Using custom sequence for nRF52
DEBUG list_all:list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: Searching for CMSIS-DAP probes using libusb
TRACE list_all:list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: Could not read string for interface 0, skipping
TRACE list_all:list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: Could not read string for interface 1, skipping
TRACE list_all:list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: Could not read string for interface 0, skipping
TRACE list_all:list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: Could not read string for interface 0, skipping
TRACE list_all:list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: Could not read string for interface 0, skipping
TRACE list_all:list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools:   Interface 3: CMSIS-DAP v1
TRACE list_all:list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools:     HID interface found
TRACE list_all:list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools:   Interface 4: WebUSB: CMSIS-DAP
TRACE list_all:list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: DAPLink CMSIS-DAP: CMSIS-DAP device with 5 interfaces
TRACE list_all:list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: Will use interface number 3 for CMSIS-DAPv1
TRACE list_all:list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: Could not read string for interface 0, skipping
TRACE list_all:list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: Could not read string for interface 0, skipping
TRACE list_all:list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: Could not read string for interface 0, skipping
TRACE list_all:list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: Could not read string for interface 0, skipping
DEBUG list_all:list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: Found 1 CMSIS-DAP probes using libusb, searching HID
TRACE list_all:list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: CMSIS-DAP device with USB path: "DevSrvsID:4294970289"
TRACE list_all:list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools:                 product_string: "DAPLink CMSIS-DAP"
TRACE list_all:list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools:                      interface: 0
TRACE list_all:list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: Ignoring duplicate DAPLink CMSIS-DAP (VID: 0d28, PID: 0204, Serial: 102636021305473600000000000000000000000097969902, CmsisDap)
DEBUG list_all:list_cmsisdap_devices: probe_rs::probe::cmsisdap::tools: Found 1 CMSIS-DAP probes total
DEBUG list_all:list_jlink_devices: jaylink: libusb 1.0.26.11724    
DEBUG list_all:list_jlink_devices: jaylink: libusb has capability API: true    
DEBUG list_all:list_jlink_devices: jaylink: libusb has HID access: false    
DEBUG list_all:list_jlink_devices: jaylink: libusb has hotplug support: true    
DEBUG list_all:list_jlink_devices: jaylink: libusb can detach kernel driver: true    
TRACE open: probe_rs::probe::cmsisdap::tools: Attempting to open device matching 0d28:0204:102636021305473600000000000000000000000097969902
TRACE open: probe_rs::probe::cmsisdap::tools: Trying device Bus 000 Device 004: ID 1050:0403
TRACE open: probe_rs::probe::cmsisdap::tools: Trying device Bus 000 Device 002: ID 2109:0822
TRACE open: probe_rs::probe::cmsisdap::tools: Trying device Bus 000 Device 001: ID 2109:2822
TRACE open: probe_rs::probe::cmsisdap::tools: Trying device Bus 001 Device 005: ID 0d28:0204
TRACE open: probe_rs::probe::cmsisdap::tools:   Interface 3: CMSIS-DAP v1
TRACE open: probe_rs::probe::cmsisdap::tools:     HID interface found
TRACE open: probe_rs::probe::cmsisdap::tools:   Interface 4: WebUSB: CMSIS-DAP
TRACE open: probe_rs::probe::cmsisdap::tools: DAPLink CMSIS-DAP: CMSIS-DAP device with 5 interfaces
TRACE open: probe_rs::probe::cmsisdap::tools: Will use interface number 3 for CMSIS-DAPv1
DEBUG open: probe_rs::probe::cmsisdap::tools: Could not open 0d28:0204 in CMSIS-DAP v2 mode
TRACE open: probe_rs::probe::cmsisdap::tools: Trying device Bus 001 Device 004: ID 0424:2422
TRACE open: probe_rs::probe::cmsisdap::tools: Error getting languages: Pipe
TRACE open: probe_rs::probe::cmsisdap::tools: Trying device Bus 001 Device 003: ID 05ac:1463
TRACE open: probe_rs::probe::cmsisdap::tools: Trying device Bus 001 Device 002: ID 05ac:1017
TRACE open: probe_rs::probe::cmsisdap::tools: Trying device Bus 001 Device 001: ID 05ac:1018
DEBUG open: probe_rs::probe::cmsisdap::tools: Attempting to open 0d28:0204 in CMSIS-DAP v1 mode
DEBUG open: probe_rs::probe::stlink::usb_interface: Acquired libusb context.
DEBUG open: probe_rs::probe::espusbjtag::protocol: Acquired libusb context.
 INFO probe_rs_cli: Wrote log to "/Users/uhinze/Library/Application Support/rs.probe-rs.probe-rs/1683551163501.log"
Error: Failed to open the debug probe.

Caused by:
    0: Probe could not be created
    1: Probe was not found.
```
</details>

**Desktop (please complete the following information):**

- macOS Ventura 13.3.1

**Additional context**
I recently came back to using probe-rs on my Mac, using commands that worked before. This could have been introduced during the last couple of MacOS updates.

Yatekii, 2023-05-08:
Hmm I thought this is a bug in libusb that does not map the interface numbers correctly. But I cannot find the issue (we had one open).

Yatekii, 2023-05-28:
Question: Do you have an USB hub between your probe and your Mac?

uhinze, 2023-05-29:
Yes I do! Will try without and report back

uhinze, 2023-05-30:
Same behavior when I directly attach the device :-/

Yatekii, 2023-07-23:
Fixed in #1689.

closed 2023-07-23 by Yatekii