probe-rs/probe-rs #1616: View logs without reflashing device

onetonfoot, 2023-05-17:
**Is your feature request related to a problem? Please describe.**

I wanna be able to access the logs without having to reflash the whole program with something like `probe-rs-cli run --chip "RP2040"`

**Describe the solution you'd like**

Maybe another `log` subcommand that can be used to view the logs. 

**Describe alternatives you've considered**

It's pretty simple to add another `--no-flash`  flag to the `run` subcommand which could be used to access the logs without flashing again,  although, from an API perspective, this feels a little out of place. 

**Additional context**

I'm fairly new to the embedded space so would need a little guidance for something a little larger like a log subcommand, but would be happy to try to submit a pull request with the changes.  The `--no-flash` flag is simple to add so could do that fine.

little-arhat, 2023-07-16:
Also wanted to just attach to logs; added https://github.com/probe-rs/probe-rs/pull/1672

Yatekii, 2023-07-23:
Fixed in #1672

commit ef5ffaf1 (2023-07-21): Implement `attach` sumbcommand (closes #1616) (#1672)
closed 2023-07-23 by Yatekii