probe-rs/probe-rs #1617: Flashed program hangs after power reset

rihardsk, 2023-05-21:
I can't tell if this is an issue with `probe-rs` but I don't have any better options so I hope somebody can help me understand what's going on here.

**Describe the bug**
When I flash my code to the board (I'm using a nrf52840 DK), either by using the `probe-run` tool or `cargo flash`, it runs just fine for the first time. But if I reset the board by turning the power off and on, my program seems to hang in a seemingly arbitrary place somewhere near the beginning, e.g., if my code sets the color of some neopixels right after startup, the neopixels light up in the respective colors, but nothing after that gets executed (even if I just try to set different colors right after that).

The specific neopixel code in my example above doesn't seem to be at fault – if I remove anything noepixel related, it still hangs somewhere, but I can't tell where.

Interestingly, there's a previous revision of my code that doesn't trigger this behavior – it runs just fine even after a power cycle. Unfortunately, I can't tell what are the specific changes between the revisions that are causing the issue (the hardware part also changes so it's hard to bisect, reversing the most obvious changes in software doesn't help). Also, I think I've experienced this issue previously manifest and later disappear at an even earlier stage, so this maybe is somehow related some specific instructions that the compiler uses?

I also tried compiling the code in release and debug modes, but that didn't have an effect.

**To Reproduce**
I don't have a minimal repro code, unfortunately, but in my project, the steps that I take are as follows:

1. The project is initialized using https://github.com/knurling-rs/app-template, I've written a fair bit of code, I'm using a nrf52840 DK
2. Build the code using `cargo build --bin some-bin`
3. Flash the program using `probe-run --erase-all --chip nRF52840_xxAA target/thumbv7em-none-eabihf/debug/some-bin`
    - Alternatively `cargo flash --chip nRF52840_xxAA --allow-erase-all --bin some-bin` can be used
4. Observe that everything is running as expected (RGB LEDs light up in the selected colors, LCD shows text, etc.)
5. Turn off and on the power button on the nrf52840 DK to reset it.
6. Observe that program has stopped executing at some point near the beginning of the program (RGB LEDs light up in the selected colors for the first time, but nothing else happens, LCD doesn't show text, etc.)

**Expected behavior**
Step 6. produces the same result as step 4.

**Stacktrace**
N/A

**Desktop (please complete the following information):**
- Linux (Arch)

Yatekii, 2023-05-22:
Do you have the RTT buffer set to blocking or override? If it's set to blocking (which I think is default with defmt) it will not be able to write more logs after some time and lock up. Could that be what happens?

rihardsk, 2023-05-22:
It seems that defmt now defaults to non-blocking mode for RTT – https://github.com/knurling-rs/defmt/pull/174  (I'm using `defmt-rtt` 0.4.0, so I should be fine in this regard), also here's some discussion on how it all works together https://github.com/knurling-rs/defmt/issues/133

Also `cargo flash --chip nRF52840_xxAA --allow-erase-all --bin some-bin` works fine for me for the first time (things only break after I power cycle), and that doesn't output anything to the terminal, so it probably means that RTT buffers aren't causing the issue (at least not in an obvious way).

rihardsk, 2023-05-23:
It seems the issue is timing related. I managed to make my code run even after I power cycle the device (thereby fixing or, rather, sidestepping the issue) by adding a few delays in my code. Here's the commit that fixed the issue if anyone's interested https://github.com/rihardsk/airlog/commit/613b6b954488ab87f2b5d2ad7b3ccf1141a61585

I guess this makes it clear that `probe-rs` is not at fault here, but, rather, `probe-rs` maybe somehow introduced some subtle timing differences which made the bug not appear, though I'm not sure how. I get that flashing and running with `probe-run` could introduce such timing differences because of `defmt` blocking on RTT stuff underneath, but how is it possible with `cargo-flash` which doesn't output anything (and so it doesn't seem to be attaching any probes)?

One possibility that somebody suggested to me is that HFCLK could somehow be relevant here, but I don't yet understand what it is exactly and allegedly there were some evidence against this possibility. I still have to look into this.

I guess this means that this issue should be closed? If so, please feel free to do so. Though I still very much welcome any ideas on what might be going on here, i.e., what is `cargo-flash` doing that might make the timings different for the issue to go away (or maybe it's something different entirely?). I will close the issue later myself, if there's no activity here.