probe-rs/probe-rs #1623: Want to debug my ESP32-C3-DevKitM-1 over USB but not found.

teamplayer3, 2023-06-01:
**Describe the bug**
I want to debug the esp32-c3 mini 1 devKit, but when I list the attached porbes it can't be found. Found out that the probes are checked against their usb ID. I added the VID and PID (10c4:ea60) of the `Silicon Labs CP210x UART Bridge` [here](https://github.com/probe-rs/probe-rs/blob/d3346852a1e7bf0dc009f849edb7ac8a15768665/probe-rs/src/probe/espusbjtag/protocol.rs#L552) which is shown up when I connect the board. Now it can be found. But if I want to start the debugging, the following error occurs:

```
Error: Failed to open the debug probe.

Caused by:
    0: Probe could not be created
    1: An error specific to a probe type occurred: USB interface or endpoints could not be found.
    2: USB interface or endpoints could not be found.
```

Maybe it's related to [this](https://github.com/platformio/platform-espressif32/issues/651).

**Desktop:**
- Linux (Ubuntu)

**Additional context**
- probe-rs version: main branch

Yatekii, 2023-06-01:
What exactly do you run? List probes should never fail.

teamplayer3, 2023-06-02:
Sorry for not including. I run `probe-rs-cli info` when getting the error.

Tiwalun, 2023-06-02:
@teamplayer3 Did you use the normal USB port on the devkit? That seems to be just a USB to UART converter, so it can't be used for debugging.

If you want to use the built-in JTAG interface, you will have to attach a USB cable yourself, as described in https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-guides/jtag-debugging/configure-builtin-jtag.html

(I'm assuming you use this devkit https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/hw-reference/esp32c3/user-guide-devkitm-1.html)

teamplayer3, 2023-06-02:
> If you want to use the built-in JTAG interface, you will have to attach a USB cable yourself, as described in https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-guides/jtag-debugging/configure-builtin-jtag.html

Ok thanks. I tried this and it was the same. But I will try it again. Maybe I connected the pins the wrong way.

Tiwalun, 2023-06-02:
You should see the devkit when you run `lsusb`, you should an entry with vendor id 0x303A and product ID 0x1001.

Tiwalun, 2023-06-02:
You shouldn't need a USB bridge, just connecting a USB cable should be enough. This guide has a picture: https://visualgdb.com/tutorials/esp32/esp32-c3/

Yatekii, 2023-06-02:
I think you might need to close some solder bridges (at least I had to for my C3 devkit)

teamplayer3, 2023-06-06:
> You shouldn't need a USB bridge, just connecting a USB cable should be enough. This guide has a picture: https://visualgdb.com/tutorials/esp32/esp32-c3/

Connecting a raw USB cable to the pins works. The board is now detected as `Espressif USB JTAG/serial debug unit`. Now, if I want to start a debug session in VSCode with the `probe-rs` extension installed, I get the error:

```
Error while flashing Trying to write flash, but no suitable (default) flash loader algorithm is linked to the given target: esp32c3 .
```

teamplayer3, 2023-06-06:
What are the general steps to debug the esp32c3 with Rust?

I created a project with `esp-rs/esp-template`. Added the launch config file. Built the project with cargo. And started the debug session.

I tried to flash with `probe-rs` lib. My code (the elf I flash was built as described previous):

```rust
let mut session = Session::auto_attach("esp32c3", Permissions::default())?;
let mut flasher = session.target().flash_loader();
flasher.load_elf_data(&mut File::open("debug_test")?)?;
flasher.commit(&mut session, DownloadOptions::default())?;
```

```bash
[2023-06-06T09:24:40Z INFO  probe_rs::flashing::loader] Found 3 loadable sections:
[2023-06-06T09:24:40Z INFO  probe_rs::flashing::loader]     Multiple sections at 42000020 (146178 bytes)
[2023-06-06T09:24:40Z INFO  probe_rs::flashing::loader]     .rodata at 3C030020 (23364 bytes)
[2023-06-06T09:24:40Z INFO  probe_rs::flashing::loader]     .rwtext at 40380000 (2040 bytes)
[2023-06-06T09:24:40Z TRACE probe_rs::flashing::loader] Adding data at address 0x42000020 with size 146178 bytes
[2023-06-06T09:24:40Z TRACE probe_rs::flashing::loader] Adding data at address 0x3c030020 with size 23364 bytes
[2023-06-06T09:24:40Z TRACE probe_rs::flashing::loader] Adding data at address 0x40380000 with size 2040 bytes
[2023-06-06T09:24:40Z DEBUG probe_rs::flashing::loader] committing FlashLoader!
[2023-06-06T09:24:40Z DEBUG probe_rs::flashing::loader] Contents of builder:
[2023-06-06T09:24:40Z DEBUG probe_rs::flashing::loader]     data: 3c030020-3c035b64 (23364 bytes)
[2023-06-06T09:24:40Z DEBUG probe_rs::flashing::loader]     data: 40380000-403807f8 (2040 bytes)
[2023-06-06T09:24:40Z DEBUG probe_rs::flashing::loader]     data: 42000020-42023b22 (146178 bytes)
[2023-06-06T09:24:40Z DEBUG probe_rs::flashing::loader] Flash algorithms:
[2023-06-06T09:24:40Z DEBUG probe_rs::flashing::loader]     algo esp32c3-flashloader: 00000000-01000000 (16777216 bytes)
[2023-06-06T09:24:40Z DEBUG probe_rs::flashing::loader] Regions:
[2023-06-06T09:24:40Z DEBUG probe_rs::flashing::loader]     region: 00000000-01000000 (16777216 bytes)
[2023-06-06T09:24:40Z DEBUG probe_rs::flashing::loader]      -- empty, ignoring!
[2023-06-06T09:24:40Z DEBUG probe_rs::flashing::loader]     region: 42000000-42800000 (8388608 bytes)
Error: Trying to write flash, but no suitable (default) flash loader algorithm is linked to the given target: esp32c3 .
```

noppej, 2023-06-06:
@teamplayer3 I'm pretty sure you are running into the issue related to ESP's IDF boot loader. `probe-rs` currently only supports the `direct-boot` feature. [See this discussion for more info](https://github.com/esp-rs/esp-hal/tree/main/esp32c3-hal)

You have two choices here:
1) To use the IDF bootloader, you can use `espflash` to flash your chip, and then any of the `probe-rs` tools will be able to connect to and run and optionally debug your binary. You can even automate this with `probe-rs-debugger`, by adding a `preLaunchTask` in your `launch.json`, to do the build and flashing, and then the debugging will work as expected.
2) To use the Direct Boot feature, in your `Cargo.toml` file, make sure you have something like `esp32c3-hal = { version = "0.9.0", features = [ "direct-boot"]}`, then the `probe-rs` tools will be able to flash and run/debug your binary without the need for `espflash`.

If you are still having difficulty with this, please post a link to your repo, or maybe include your `Cargo.toml` and `launch.json` files, and I'll be happy to help.

MabezDev, 2023-06-06:
FYI I am happy to PR esp-idf bootloader format to probe-rs. I guess the main thing to overcome would this change be welcome.

I guess I can work on a draft of how this would look (when I get a few spare cycles) and we can evaluate from there?

noppej, 2023-06-06:
My vote is "Yes, please"

noppej, 2023-06-07:
Use their `espflash` to flash the image, and once it is flashed, it connect with something like `probe-rs-debugger` with `flashingEnabled = false`.

Yatekii, 2023-06-07:
Yeah but if they use `espflash` that does not require the debug connection, so after flashing it's not guaranteed they can debug.

noppej, 2023-06-07:
I see what you mean. The newer versions of their dev kit has wired the debug access to the USB, so you no longer need to do the special cable hack some of us (I think you too) had to do on the early version. 

I guess what I'm trying to say is that with what @MabezDev is offering (implement IDF bootloader for probe-rs), if you have a USB cable, a newer devkit (or old one with USB hack), then you can use probe-rs to flash, debug, RTT, etc.

That feels like a worth-while status to me. What am I missing?

MabezDev, 2023-06-07:
> I see what you mean. The newer versions of their dev kit has wired the debug access to the USB, so you no longer need to do the special cable hack some of us (I think you too) had to do on the early version.

To add to this, all devkits going forward, i.e the ESP32-C6, ESP32-H2 come with dual USB by default, giving debug access out of the box this no external hardware.

I'll work on a PR for a little bit and see where I get it to.

MabezDev, 2023-06-07:
Didn't take too long actually, draft here: https://github.com/probe-rs/probe-rs/pull/1629. Do bare in mind that the current flash algo for the esp32c3/esp32c6 are quite slow so flashing bigger applications might take a while. The flash algos existed before `esp-hal`, we can probably rewrite them using esp-hal to get much better performance.

Yatekii, 2023-07-23:
@teamplayer3 what's the status here? :) Can this be marked as resolved?

noppej, 2023-07-24:
@Yatekii Based on my involvement in this issue, I vote we close this as fixed in #1629 and #1647

closed 2023-07-24 by Yatekii