probe-rs/probe-rs #1634: probe-rs-cli log files do not rotate

MabezDev, 2023-06-11:
**Describe the bug**

Running many instances of probe-rs-cli will keep creating log files indefinitely, I just found 23GB worth over a few days of use.

**Expected behavior**

Imo the log file should only be written under these circumstances:

1. It's requested by the user explicitly
2. On panic or crash

noppej, 2023-06-12:
I have an interpretation/implementation of #1 above [in probe-rs-debugger](https://github.com/probe-rs/probe-rs/blob/34fcd647fb7eea042f6ea6c37cf228356af2cbaa/debugger/src/main.rs#L126-L171) ... let me know if you improve it so I can back-port it please :)

bugadani, 2024-01-02:
Even though we don't save logs on panic (because we don't retain them, because they are expensive by default), I'm going to mark this as closed by #1946

closed 2024-01-02 by bugadani