probe-rs/probe-rs #1641: Add cli access to write_word_32 (and similar)

haata, 2023-06-19:
**Is your feature request related to a problem? Please describe.**
I have a use-case where I would like to use dfu-util to flash the firmware (instead of probe-rs) as USB is much faster than my probe. However, when a probe is attached DEMCR is set which halts the MCU whenever jumping to the bootloader (which uses a physical nRST pin).

I currently have an xtask using probe-rs-cli-util that looks like this
```rust
    let common = ProbeOptions {
        allow_erase_all: false,
        chip: Some(probe_run_chip),
        chip_description_path: None,
        connect_under_reset: false,
        dry_run: false,
        probe_selector: None,
        protocol: None,
        speed: None,
    };

    let mut session = common.simple_attach()?;
    let mut core = session.core(0)?;

    // Disable halt on reset (DEMCR) and other reset debugging
    core.write_word_32(0xE000EDFC, 0)?;

    // Force nRST (RSTC_CR) - atsam4
    core.write_word_32(0x400E1400, 0xA500_0008)?;
```

While this does work, it is rather slow to recompile large portions of probe-rs for this relatively simple operation.

**Describe the solution you'd like**

I would like something like what Jlink and other probe utilities have (reading/writing specified word sizes).
```bash
probe-rs-cli write --32 <start address> [<word>...]
probe-rs-cil write --32 0x400E1400 0xA5000008
```
Support for 8, 16, 32 (and maybe 64 bit words) would make sense.

(and it would probably make sense to have an inverse write option)

**Describe alternatives you've considered**

It's not entirely clear if I can use `probe-rs-cli download` to do these operations (I need to make sure nothing else gets written). And the ergonomics of generating two binary files to write two registers is pretty bad.

**Additional context**
I'm happy to work on a PR if there is interest in this.

Yatekii, 2023-09-08:
I am happy to review PRs adding this functionality :)

haata, 2023-09-08:
I'll see if I can put something together.

PR #1746 (2023-09-10): Add read/write probe-rs cli commands
- Fixes #1641
- Simple tool to read/write from target memory addresses
- Handles 8,32,64 bit words

Example (using general purpose backup registers on ATSAM4S8B)

```bash
: probe-rs read --probe 1fc9:0143 b32 0x400E1490 2 --chip ATSAM4S8B
00000000 00000000
: probe-rs write --probe 1fc9:0143 b32 0x400E1490 0xDEADBEEF 0xBEEFDEAD --chip ATSAM4S8B
: probe-rs read --probe 1fc9:0143 b32 0x400E1490 2 --chip ATSAM4S8B
deadbeef beefdead
: probe-rs read --probe 1fc9:0143 b8 0x400E1490 2 --chip ATSAM4S8B
ef be
: probe-rs read --probe 1fc9:0143 b64 0x400E1490 2 --chip  [... cut]
closed 2023-09-10 by Yatekii