probe-rs/probe-rs #1669: Tracking Issue: Reduce "drift to the right" in code, i.e. depth of nesting and indentation in code.

noppej, 2023-07-13:
There are some parts of the code, which evolved over time, that employ deep nesting, and as such, the indentation drifts the code to the right to the point where it is difficult to read.

(Please add specific references where we would prioritize refactoring to improve code maintenance)

- [ ] Several functions in `probe-rs/src/debug/debug_info.rs`

cc: @Yatekii and @Tiwalun : Please provide some guidance on what you feel an 'acceptable maximum' number of indents that we want to aim for. I looked and could not find any `clippy` lints that will help us, but please feel free to add prospective file/function references above.

Yatekii, 2023-07-13:
Generally there is a clippy lint for code complexity and length of functions. I am not sure if we deliberately disabled them because they should trigger I think.
Also we do not enforce clippy atm. We should.

Yatekii, 2024-01-28:
I guess this one is really tough to track. I guess we can close this and just improve as we go, because this specific ticket will most likely never be finished, but it is also not done by all means just yet :)

closed 2024-01-28 by Yatekii