probe-rs/probe-rs #1670: Investigate replacing "homegrown" code, with the use of `addr2line`

noppej, 2023-07-13:
The `probe-rs::debug` module contains code which can likely be superseded by the use of the [addr2line](https://github.com/gimli-rs/addr2line) repo. 

We already rely heavily on the `gimli` crate from the same organization, and experience has shown that their api's are fast and reliable.

@Yatekii and @Tiwalun : Please let me have a 👍 or 👎 on this, along with other suggestions you may have.

Yatekii, 2023-07-13:
I think `addr2line` had some flaws back in the day, but I do not fully recall.

bugadani, 2024-02-25:
One such flaw with `addr2line` is that it's trying to resolve symbols a bit too eagerly. See this [espflash issue](https://github.com/esp-rs/espflash/issues/600) for an example

noppej, 2024-02-25:
Yeah, I literally looked at it as part of refactoring in #2180 , and had to back out because of what you said @bugadani , and also because we also need line-2-addr, with some very specific requirements. I'm closing this for now :)

closed 2024-02-25 by noppej