probe-rs/probe-rs #1682: Add a possibility to specify expected IR lengths for JTAG capabilities of debug probes

Yatekii, 2023-07-20:
See https://github.com/probe-rs/probe-rs/pull/1462#issuecomment-1642911723

seanmlyons22, 2023-08-25:
@Yatekii, how do you think this should be done? One idea I have is that something can be added to the `target.yaml` file that describes the scan chain. This could also be a solution for https://github.com/probe-rs/probe-rs/issues/1683 where the yaml file could contain the index of the wanted DAP.

Both could be optional, where if not provided we fall back to the default behavior as it is today.

Yatekii, 2023-08-25:
Yeah, I think the target specifications are the right place :) Is this a fixed value? We already have special values for denoting the DAP on some multidrop chips (rp2040) or unconventionaly enumerated DAPs :)
So this should fit right in :)

seanmlyons22, 2023-08-25:
> Is this a fixed value?

Yep! Its fixed in the HW, but not by the spec. Different vendors can implement different ir lengths.

I found a `scanchain` description as part of the [System Description File](https://open-cmsis-pack.github.io/Open-CMSIS-Pack-Spec/main/html/sdf_pg.html#sdf_element_scanchain) which looks like it would do exactly the trick.

Converting this to yaml would look like:

```yaml
scanchain:
    - !Dap
        name: core0
        type: Cortex-M0+
        ir_len: 4
        protocol: 
        - jtag
        - swd
    - !Device
        name: ICEPICK
        type: Other
        ir_len: 6
```

Following that format means I think we could try to automatically parse and populate this from target-gen if it exists in the pack. Then, if the target file has a `scanchain` specified, we could use that to specify the i`r_lengths` before connect.

Yatekii, 2023-08-26:
Looking great! I am not sure we should use terms such as Cortex-M0+ rather than thumbv6 because other stuff uses it that way, but other than that its nice!

seanmlyons22, 2023-08-26:
I have the `scanchain` added to the target now. 

Do you have any recommendations on how to connect that to [configure_jtag](https://github.com/probe-rs/probe-rs/blob/master/probe-rs/src/probe/cmsisdap/mod.rs#L1014-L1017)? 

Is it through session? I basically need something that knows about the `interface` and `target`.

Yatekii, 2023-08-26:
I would give it the `interface` and `target` as separate params directly :)

PR #1731 (2023-09-18): Allow JTAG scan information to be encoded in targets
The scan chain information structure is a direct implementation of [Scan Chain](https://open-cmsis-pack.github.io/Open-CMSIS-Pack-Spec/main/html/sdf_pg.html#sdf_element_scanchain) from CMSIS pack's System Description File (SDF)

It aims to resolve #1682 by adding scan chain information to the target, passing scan chain information to the probe, and finally allowing the `CmsisDap` probe to use scan chain information if its provided.

Having scan chain information opens the possibly for the following (these are open ideas, not yet implemented) [... cut]
closed 2023-09-18 by Yatekii