probe-rs/probe-rs #1686: `embassy` example app does nothing after being pushed to board using `probe-rs`

U007D, 2023-07-21:
**Describe the bug**
Running the `blinky` example from `embassy` on the STM32F303VCT does not work.

**To Reproduce**
I test drove the newest `probe-rs` yesterday using the `embassy` `blinky` example on an STM32F3 (STM32F303VCT).  It appeared to flash happily, but then nothing happened (tried another project, `button`, with the same result).

My efforts:

```bash
❯ cargo run --release --bin blinky
    Finished release [optimized + debuginfo] target(s) in 0.11s
     Running `probe-rs run --chip STM32F303ZETx target/thumbv7em-none-eabihf/release/blinky`
     Erasing sectors ✔ [00:00:00] [########################################] 18.00 KiB/18.00 KiB @ 35.40 KiB/s (eta 0s )
 Programming pages   ✔ [00:00:01] [########################################] 17.00 KiB/17.00 KiB @ 14.23 KiB/s (eta 0s )    Finished in 1.728s
Error: An ARM specific error occurred.

Caused by:
    0: The debug probe encountered an error.
    1: An error specific to a probe type occurred
    2: USB
    3: Other error
```

I realized the default MCU was not the one I have so I changed my `embassy-rs/embassy/examples/stm32f3/.cargo/config.toml`:

```toml
       │ File: .cargo/config.toml
───────┼────────────────────────────────────────────────────────────────────────────────────────────────────────────────
   1   │ [target.'cfg(all(target_arch = "arm", target_os = "none"))']
   2   │ # replace STM32F429ZITx with your chip as listed in `probe-rs chip list`
   3 ~ │ runner = "probe-rs run --chip STM32F303VCTx"
   4   │
   5   │ [build]
   6   │ target = "thumbv7em-none-eabihf"
   7   │
   8   │ [env]
   9   │ DEFMT_LOG = "trace"
```

```
▶ cargo run --release --bin blinky
   Compiling embassy-stm32f3-examples v0.1.0 (/Users/Shared/Development/3rd_party/embassy-rs/embassy/examples/stm32f3)
    Finished release [optimized + debuginfo] target(s) in 0.36s
     Running `probe-rs run --chip STM32F303VCTx target/thumbv7em-none-eabihf/release/blinky`
     Erasing sectors ✔ [00:00:00] [########################################] 18.00 KiB/18.00 KiB @ 35.23 KiB/s (eta 0s )
 Programming pages   ✔ [00:00:01] [########################################] 17.00 KiB/17.00 KiB @ 12.85 KiB/s (eta 0s )    Finished in 1.858s
=q^C⏎
```

Note the random characters on the final line of output--looks similar to a baud rate mismatch.

Regardless, there is no flashing LED on the board.  (I am able to flash blinky apps to the board using `OpenOCD`.)

I thought I'd try simplifying things and try with `cargo-flash` since my understanding is it also uses `probe-rs`:

```bash
❯ cargo flash --bin blinky --chip STM32f303VCTx
    Finished dev [unoptimized + debuginfo] target(s) in 0.16s
    Flashing /Users/Shared/Development/3rd_party/embassy-rs/embassy/examples/stm32f3/target/thumbv7em-none-eabihf/debug/blinky
        WARN probe_rs::config::target > Using custom sequence for ARMv7 STM32F303VCTx
     Erasing sectors ✔ [00:00:01] [########################################] 62.00 KiB/62.00 KiB @ 51.44 KiB/s (eta 0s )
 Programming pages   ✔ [00:00:03] [########################################] 62.00 KiB/62.00 KiB @ 19.41 KiB/s (eta 0s )    Finished in 4.42s
```

No messy characters this time as the final line of output, but also still no blinky activity on the board.  I am using the latest `probe-rs` commit from `master` as of yesterday morning:

```bash
▶ probe-rs --version
probe-rs 0.19.0
git commit: v0.19.0-22-gef5ffaf1
```

**Expected behavior**
Expected an LED to be flashing on the board.

**Stacktrace**
N/A

**Desktop (please complete the following information):**

- macOS Ventura

**Additional context**
Troubleshooting on Matrix with Noah, Noah suspects the linker script is set for the wrong chip.  I tried flashing the board with `probe-rs` for [this exercise](https://github.com/rust-embedded/discovery/tree/master/f3discovery/src/05-led-roulette) and `probe-rs` flashed and ran fine.

U007D, 2023-07-21:
[Additional context here](https://matrix.to/#/!vhKMWjizPZBgKeknOo:matrix.org/$RDlvY3hwko2hzhvT7bue8bf_MFv-WzJb1sf1i2mxazo?via=matrix.org&via=chat.berline.rs&via=grusbv.com).

Yatekii, 2023-07-21:
The

```
Caused by:
    0: The debug probe encountered an error.
    1: An error specific to a probe type occurred
    2: USB
    3: Other error
```

is the interesting part here. Why is the error so weird?

U007D, 2023-07-21:
I'm not sure I'd consider it "the interesting part here", given I still can't flash `embassy` `blinky` to my board, but given I was configured to the wrong processor, I just assume all bets are off.

Maybe "Are you configured for the correct processor?" should be one of those options?

Yatekii, 2023-07-21:
You can flash - probe-rs is working properly. you just don't build/flash what you need yet =)

But the first error you got should def display something else.

Yatekii, 2023-07-24:
So the flashing itself works. What I am interested in in here is the fact that the error shown is super weird. I took the liberty to adjust the title.