probe-rs/probe-rs #1692: Pico W and RasPi Debug probe:ECONNREFUSED when I press F5 on (slight modified)template project

Dicklessgreat, 2023-07-23:
**Describe the bug**
On Windows10, I want to debug Pico W with [RasPi Debug Probe](https://www.raspberrypi.com/products/debug-probe/) . Then VSCode shows Error dialog box while I try to start debugging(F5).
[image]
[image]

Note I'm totally noob. This is the first attempt to start embedded development, but the error message lead me here.

**To Reproduce**
Steps to reproduce the behavior(I installed no embedded environment ever.)

1. on Power Shell, `rustup target install thumbv6m-none-eabi`
2. `vcpkg install libftdi1:x64-windows libusb:x64-windows`, `set VCPKGRS_DYNAMIC=1`
3. `cargo install probe-rs --features cli`
4. install VSCode extention of "debugger for probe-rs" 
5. `git clone` template project
6. replace "probe-rs-debugger" with "probe-rs" of `runtimeExecutable` on `.vscode/launch.json` (there's no "probe-rs-debugger", right? When I hit F5 on template project of mint condition, it says "probe-rs-debugger not found". referring [official document](https://probe.rs/docs/tools/vscode/) and Ctr + F5 "runtimeExecutable")
7. press F5 on VSCode

**Expected behavior**
press F5 and start debugging

**Desktop (please complete the following information):**

- Windows 10

**Additional context**
In the same project, `cargo run` works fine.
[image]

Yatekii, 2023-07-23:
You need to install the newest version of the extension to work with the newest version of probe-rs :)

Dicklessgreat, 2023-07-23:
Hmm...I'm using probe-rs 0.19.0, and Debugger for probe-rs which is last released 6/29/2023, 22:53:36.
Seems newest?

Yatekii, 2023-07-23:
You need the newest version of the *extension* not the debugger. The extension still tries to find the old binary name.

Dicklessgreat, 2023-07-24:
I mean...I'm using "Debugger for probe-rs" extension last released 6/29/23, 22:53:36. :)
I installed it yesterday from VSCode Marketplace...So I think it is the newest?

noppej, 2023-07-24:
@Dicklessgreat From your description, it appears as if you are following the correct steps, so the devil might be in the details. Can you please do the following: 
- Post your `launch.json`
- At the point where you get the dialog with the `ECONNREFUSED` error, can you please open another window and look to see if any `probe-rs` processes have been launched in the background?

Dicklessgreat, 2023-07-24:
@noppej Thanks reply!

I closed all windows that corresponds to development but the Pico debugging VSCode window, and hit F5, same dialog show up. At that moment, I run `Get-Process` on Power Shell and I can't find `probe-rs` processes.

there're no changes from [original template](https://github.com/rp-rs/rp2040-project-template)  but "runtimeExecutable" 
```
// The format of this file is specified in https://probe.rs/docs/tools/vscode/#start-a-debug-session-with-minimum-configuration
{
    "version": "0.2.0",
    "configurations": [
        {
            "preLaunchTask": "rust: cargo build",
            "type": "probe-rs-debug",
            "request": "launch",
            "name": "rp2040-project",
            "cwd": "${workspaceFolder}",
            "chip": "rp2040",
            // RP2040 doesn't support connectUnderReset
            "connectUnderReset": false,
            "speed": 4000,
            "runtimeExecutable": "probe-rs",
            "runtimeArgs": [
                "debug"
            ],
            "flashingConfig": {
                "flashingEnabled": true,
                "resetAfterFlashing": true,
                "haltAfterReset": true,
            },
            "coreConfigs": [
                {
                    "coreIndex": 0,
                    "programBinary": "target/thumbv6m-none-eabi/debug/rp2040-project-template",
                    "chip": "RP2040",
                    // Uncomment this if you've downloaded the SVD from
                    // https://github.com/raspberrypi/pico-sdk/raw/1.3.1/src/rp2040/hardware_regs/rp2040.svd
                    // and placed it in the .vscode directory
                    // "svdFile": "./.vscode/rp2040.svd",
                    "rttEnabled": true,
                    "options": {
                        "env": {
                            "DEFMT_LOG": "debug"
                        }
                    },
                }
            ],
            "consoleLogLevel": "Info", //Error, Warn, Info, Debug, Trace
            "wireProtocol": "Swd"
        }
    ]
}
```

Dicklessgreat, 2023-07-24:
IT WORKS!!!!!!!
[image]
`1b` works for me, and also, `1a` is not mandatory, but works both with and without `runtimeExecutable`(but `probe-rs-debugger` did not work! only `probe-rs`work;)).
`2`also works.
`3`, sorry I don't fully understand what you intended to,(because I'm not sure even dap-server!!) but if you intend to check `probe-rs` works as dap-server, yes, it works.
[image]

Thank you! So now my issue has finished, I close this issue but in case  you think to improve error catching, feel free to reopen this issue and ask me to offer more context and information(if it's necessary)!

noppej, 2023-07-24:
Re-opening to address the error handling for future users.

noppej, 2023-07-26:
Closing as addressed in  https://github.com/probe-rs/vscode/pull/63 and https://github.com/probe-rs/probe-rs/pull/1699

closed 2023-07-26 by noppej