probe-rs/probe-rs #1700: Add support for EFR32MG24

jaxter184, 2023-07-25:
**Is your feature request related to a problem? Please describe.**
This is an issue for collecting my attempts to add support for the EFR32MG24 family of chips by Silicon Labs

**Additional context**
I used `target-gen` to generate a yaml file based on [the CMSIS pack](https://www.keil.arm.com/devices/silicon-labs-efr32mg24b220f1536im48/features/):
```
> target-gen pack ~/Downloads/SiliconLabs.GeckoPlatform_EFR32MG24_DFP.4.3.0.pack .
Generated 1 target definition(s):
	./EFR32MG24_Series.yaml
Finished in 15.242942ms
```
The generated file: [efr32mg24.zip](https://github.com/probe-rs/probe-rs/files/12166302/efr32mg24.zip)
Then, to flash:
```sh
> cargo run --release --bin probe-rs --features=cli -- run --chip efr32mg24b220F1536IM48 <path-to-binary>
    Finished release [optimized] target(s) in 0.11s
     Running `<path-to-probe-rs> run --chip efr32mg24b220F1536IM48 <path-to-binary>`
     Erasing sectors ⠁ [00:00:00] [#################]       0B/      0B @       0B/s (eta 0s )Error: The flashing procedure failed for '<path-to-binary>'.

Caused by:
    0: Something during the interaction with the core went wrong
    1: An ARM specific error occurred.
    2: Error using access port
    3: Failed to read register DRW at address 0x0000000c
    4: An error occurred in the communication with an access port or debug port.
    5: Target device responded with a FAULT response to the request.
```

Using `--log-file`, I got the following log: [efr-log.json.log](https://github.com/probe-rs/probe-rs/files/12166248/efr-log.json.log). On line 1333, it says "Reading DAP register failed".

One of the things I'm trying is using openocd and gdb to flash the binary. Right now, I'm getting an error (using ef**m**32.cfg because ef**r**32.cfg doesn't seem to be installed for openocd on my machine):
```
[efm32.cpu] halted due to debug-request, current mode: Thread
xPSR: 0x69000000 pc: 0x08007814 msp: 0x20001198
Error: Target is not Cortex-Mx Device
Error: auto_probe failed
Error: Connect failed. Consider setting up a gdb-attach event for the target to prepare target for GDB connect, or use 'gdb_memory_map disable'.
Error: attempted 'gdb' connection rejected
```

Also, I tried using a Mini J-Link instead of the built-in J-Link Pro (using an adapter, because apparently the pinout for Silicon Labs' Simplicity debugger is different from Mini J-Link despite being the same connector). It showed the following error, but I'm fairly certain this is a hardware/configuration problem and not really relevant to the problem above with the FAULT response:
```
    Finished release [optimized] target(s) in 0.11s
     Running `<path-to-probe-rs> run --chip efr32mg24b220F1536IM48 <path-to-binary> --probe '1366:0101'`
Error: Connecting to the chip was unsuccessful.

Caused by:
    0: An ARM specific error occurred.
    1: An error occurred in the communication with an access port or debug port.
    2: Target device did not respond to request.
```

noppej, 2023-07-26:
So not having a EFM32 to debug with, I'm looking at the code paths, and the two `probe-rs run` errors look related, with slight differences because the first one follows code to flash the target, and the second is a straight-up attach. 

Perhaps you can reach out to @corecode who implemented the [custom sequence](https://github.com/probe-rs/probe-rs/pull/1556) ... in case he has some pointers.

Also, did you read these application notes on their debug interface locking and security?
-  https://www.silabs.com/documents/public/application-notes/an1303-efr32-dci-swd-programming.pdf
- https://www.silabs.com/documents/public/application-notes/an1190-efr32-secure-debug.pdf

Finally, `target-gen` relies on information in the CMSIS-PACK to generate memory addresses and other values, but we've all seen examples where CMSIS-PACK files have bugs. It might be worth comparing the values in the generated target yaml to those in the docs and make sure there are no obvious discrepencies.

Are you able to run probe-rs in a debugger, and trace back to where exactly the error is happening?

corecode, 2023-07-26:
I don't have openocd working for my efr32bg22 - there is a fork somewhere on github that adds support for the series 2 chips, but I didn't bother.  JLinkGDBServer works.  I recently used probe-rs and it worked to flash, but failed on `monitor reset`.

jaxter184, 2023-07-26:
> did you read these application notes on their debug interface locking and security?

I haven't seen these yet! I'll give them a look, thanks. Some sort of security memory protection shenanigan is top on my list of suspects at the moment.

> It might be worth comparing the values in the generated target yaml to those in the docs

There aren't any blantant errors, but I'll definitely do a closer comparison at some point.

> Are you able to run probe-rs in a debugger, and trace back to where exactly the error is happening?

Hopefully this isn't too nooby of a question, but since the program doesn't panic (at least I think that's what's going on?), what's the easiest way to get a trace of where the error is happening? Usually I would use the `backtrace` gdb command, but since it just exits, I think that wouldn't work? The only other option I can think of is to set break points and/or step through carefully.

Also, I was able to get the chip flashed using Silicon Labs' [Simplicity Commander tool](https://www.silabs.com/documents/public/user-guides/ug162-simplicity-commander-reference-guide.pdf). I couldn't get the GPIOs working (probably something wrong with my PAC), but it correctly identifies RTT memory, so I think it's flashing correctly? Maybe it'd help to sniff the USB messages or something and see what the differences are?

corecode, 2023-07-26:
Probably much easier to sniff on the SWD pins.  I tried sniffing JLink 
messages, but at least JLinkExe uses undocumented USB messages, so it 
was tedious to figure out what was going on.

noppej, 2023-07-26:
> what's the easiest way to get a trace of where the error is happening?

If you can run probe-rs in a debugger, you can put a breakpoint on the error message and look at the stack trace, and/or step through the code immediately before the error.

corecode, 2023-07-26:
would it be possible for somebody to document the logging environment 
variables to see which SWD/AP transactions are happening?  i tried 
probe-run with very verbose, and there is a lot of chatter which makes 
it difficult to follow.

noppej, 2023-07-26:
I apologize if this is stating the obvious, but have you tried limiting to just the module you suspect ... e.g. `RUST_LOG="trace, probe_rs::probe"`

noppej, 2023-07-26:
Also, I know @Yatekii created [this log viewer](https://github.com/probe-rs/log-viewer), but I haven't spent much time with it ... I prefer good old `grep` LOL

jaxter184, 2023-08-05:
I finally got around to getting a debug trace:
```
Thread 1 "probe-rs" hit Breakpoint 4, probe_rs::probe::jlink::arm::parse_swd_response (
    response=&[bool](size=56) = {...}, direction=probe_rs::probe::jlink::arm::TransferDirection::Read)
    at probe-rs/src/probe/jlink/arm.rs:795
795	        return Err(DapError::FaultResponse);
(gdb) bt
#0  probe_rs::probe::jlink::arm::parse_swd_response (response=&[bool](size=56) = {...},
    direction=probe_rs::probe::jlink::arm::TransferDirection::Read) at probe-rs/src/probe/jlink/arm.rs:795
#1  0x00005555570a0ad8 in probe_rs::probe::jlink::arm::perform_swd_transfers<probe_rs::probe::jlink::JLink> (
    probe=0x55555974a790, transfers=&mut [probe_rs::probe::jlink::arm::DapTransfer](size=2) = {...})
    at probe-rs/src/probe/jlink/arm.rs:332
#2  0x00005555570a439a in probe_rs::probe::jlink::arm::perform_transfers<probe_rs::probe::jlink::JLink> (
    probe=0x55555974a790, transfers=&mut [probe_rs::probe::jlink::arm::DapTransfer](size=1) = {...}, idle_cycles=2)
    at probe-rs/src/probe/jlink/arm.rs:496
#3  0x0000555557445b63 in probe_rs::probe::jlink::arm::{impl#5}::raw_read_register<probe_rs::probe::jlink::JLink> (
    self=0x55555974a790, port=probe_rs::architecture::arm::traits::PortType::AccessPort, address=12)
    at probe-rs/src/probe/jlink/arm.rs:954
#4  0x00005555575ee5d7 in probe_rs::architecture::arm::communication_interface::{impl#14}::read_raw_ap_register (
    self=0x555559725090, ap=..., address=12) at probe-rs/src/architecture/arm/communication_interface.rs:689
#5  0x00005555575d28e1 in probe_rs::architecture::arm::ap::{impl#1}::read_ap_register<probe_rs::architecture::arm::communication_interface::ArmCommunicationInterface<probe_rs::architecture::arm::communication_interface::Initialized>, probe_rs::architecture::arm::ap::memory_ap::MemoryAp, probe_rs::architecture::arm::ap::memory_ap::DRW> (self=0x555559725090,
    port=...) at probe-rs/src/architecture/arm/ap/mod.rs:145
#6  0x0000555557651189 in probe_rs::architecture::arm::memory::adi_v5_memory_interface::ADIMemoryInterface<probe_rs::architecture::arm::communication_interface::ArmCommunicationInterface<probe_rs::architecture::arm::communication_interface::Initialized>>::read_ap_register<probe_rs::architecture::arm::communication_interface::ArmCommunicationInterface<probe_rs::architecture::arm::communication_interface::Initialized>, probe_rs::architecture::arm::ap::memory_ap::DRW> (
    self=0x555559753b00, access_port=...) at probe-rs/src/architecture/arm/memory/adi_v5_memory_interface.rs:266
#7  0x0000555557652b1f in probe_rs::architecture::arm::memory::adi_v5_memory_interface::ADIMemoryInterface<probe_rs::architecture::arm::communication_interface::ArmCommunicationInterface<probe_rs::architecture::arm::communication_interface::Initialized>>::read_word_32<probe_rs::architecture::arm::communication_interface::ArmCommunicationInterface<probe_rs::architecture::arm::communication_interface::Initialized>> (self=0x555559753b00, access_port=..., address=4)
    at probe-rs/src/architecture/arm/memory/adi_v5_memory_interface.rs:363
#8  0x0000555557673391 in probe_rs::architecture::arm::memory::adi_v5_memory_interface::{impl#3}::read_32<probe_rs::architecture::arm::communication_interface::ArmCommunicationInterface<probe_rs::architecture::arm::communication_interface::Initialized>> (self=0x555559753b00, address=4, data=&mut [u32](size=1) = {...})
    at probe-rs/src/architecture/arm/memory/adi_v5_memory_interface.rs:910
#9  0x000055555764ddde in probe_rs::architecture::arm::memory::adi_v5_memory_interface::ArmProbe::read_word_32<probe_rs::architecture::arm::memory::adi_v5_memory_interface::ADIMemoryInterface<probe_rs::architecture::arm::communication_interface::ArmCommunicationInterface<probe_rs::architecture::arm::communication_interface::Initialized>>> (
    self=0x555559753b00, address=4) at probe-rs/src/architecture/arm/memory/adi_v5_memory_interface.rs:32
#10 0x000055555704c5cf in probe_rs::architecture::arm::sequences::efm32xg2::{impl#1}::reset_catch_set (
    self=0x555559711e10, core=..., _core_type=probe_rs_target::chip_family::CoreType::Armv8m, _debug_base=...)
    at probe-rs/src/architecture/arm/sequences/efm32xg2.rs:35
#11 0x0000555557277b7f in probe_rs::architecture::arm::core::armv8m::{impl#1}::reset_catch_set (self=0x55555974fd60)
    at probe-rs/src/architecture/arm/core/armv8m.rs:446
#12 0x0000555557273f9b in probe_rs::architecture::arm::core::armv8m::{impl#1}::reset_and_halt (self=0x55555974fd60,
    _timeout=...) at probe-rs/src/architecture/arm/core/armv8m.rs:230
#13 0x0000555557031338 in probe_rs::core::Core::reset_and_halt (self=0x7ffffffe6650, timeout=...)
    at probe-rs/src/core.rs:412
#14 0x00005555572a8c19 in probe_rs::flashing::flasher::Flasher::load (self=0x7ffffffeaf00)
    at probe-rs/src/flashing/flasher.rs:134
#15 0x00005555572a5bec in probe_rs::flashing::flasher::Flasher::new (session=0x7fffffff8298, core_index=0,
    raw_flash_algorithm=0x7fffffff0a60, progress=...) at probe-rs/src/flashing/flasher.rs:104
#16 0x00005555574de61d in probe_rs::flashing::loader::FlashLoader::commit (self=0x7fffffff8d30, session=0x7fffffff8298,
    options=...) at probe-rs/src/flashing/loader.rs:344
#17 0x0000555555ac9a32 in probe_rs::util::flash::run_flash_download (session=0x7fffffff8298, path=...,
    download_options=0x7fffffffcce0, probe_options=0x7fffffff8350, loader=..., do_chip_erase=false)
    at probe-rs/src/bin/probe-rs/util/flash.rs:144
[... cut ...]
#35 0x00007ffff7423d8a in __libc_start_main () from /usr/lib/libc.so.6
#36 0x0000555555a33f35 in _start ()
```
([a recording of me collecting the trace](https://asciinema.org/a/MuZPcMOgDSg13aHIjBPqfRDXC). the actual trace is about halfway through the recording.)

My untrained eye doesn't really see anything here that looks like it'll directly help with this problem. I think my next step is going to be to look at the SWD lines through a logic analyzer and compare it to Simplicity Commander's output.

Yatekii, 2023-09-18:
There is not much of a trace :) Interesting would be the logs with `RUST_LOG=debug` :) The `IHI0031` manual describes reasons for FAULTs. I wonder if the sticky bit is set. I think we would find that in the log trace.

Yatekii, 2023-09-26:
Any chance to post a trace @jaxter184? :)

jaxter184, 2023-09-26:
Ah, sorry, thanks for the reminder; I'll try to do it first thing tomorrow

jaxter184, 2023-09-26:
> Interesting would be the logs with `RUST_LOG=debug`

When you say "logs" here, do you mean the stdout? Here is the output: [stdout.log](https://github.com/probe-rs/probe-rs/files/12730189/stdout.log), but it seems to be more or less the same information that's in the `--log-file` output. Although it is easier to read, and I think you're right about the sticky bit being set, if I'm interpreting things right.

Yatekii, 2023-09-26:
So it is indeed an overrun. So we were issuing commands too fast. Did you try with a slower clockspeed for good measure?

jaxter184, 2023-09-27:
I'm unfortunately not totally certain, but I *think* I disturbed something by flashing the chip using Simplicity Commander. Rather than give the errors from before, it now times out before it even starts flashing the code. Hopefully the logs help:
[stdout.log](https://github.com/probe-rs/probe-rs/files/12732603/stdout.log)
[test.log](https://github.com/probe-rs/probe-rs/files/12732605/test.log)h

htcrane, 2025-03-17:
Resurrecting this from the dead. Some quick context of where I am at on this...

- Hardware: EFR32xG24 Explorer Kit (fresh out the box).
- I have generated the new target files successfully (will PR once we figure out this current issue) from the latest CMSIS pack.
- The error output to the terminal is very similar to what is reported above ([templog.log](https://github.com/user-attachments/files/19276249/templog.log)).

```bash
hello_world2 on  master [?] is 📦 v0.1.0 via 🦀 v1.85.0 via 🅒 base 
❯ probe-rs run --chip EFR32MG24B210F1536IM48 target/thumbv8m.main-none-eabi/debug/examples/hello --chip-description-path /Users/traskcrane/Documents/Github/probe-rs/probe-rs/targets/EFR32MG24_Series.yaml --speed 125 --log-file templog.log
      Erasing ✔   0% [--------------------]
  Programming ✔ 100% [--------------------]                                                   
Error: The flashing procedure failed for 'target/thumbv8m.main-none-eabi/debug/examples/hello'.

Caused by:
    0: Failed to reset, and then halt the CPU.
    1: An ARM specific error occurred.
    2: Error using access port FullyQualifiedApAddress { dp: Default, ap: V1(0) }.
    3: Failed to read register DRW at address 0xd0c
    4: An error occurred in the communication with an access port or debug port.
    5: Target device responded with a FAULT response to the request.
```

It is important to note that even without the `--speed 125` and setting other speeds the jlink accepts, the error remains. 

### Experimenting
**Segger Software**
Using the segger J-Flash software I pulled this report (just open up a project, select your device, and connect).

```
Connecting ...
 - Connecting via USB to probe/ programmer device 0
 - Probe/ Programmer firmware: Silicon Labs J-Link OB compiled Mar 15 2022 19:47:37
 - Probe/ Programmer S/N: 440333609
 - Device "EFR32MG24BXXXF1536" selected.
 - Target interface speed: 1900 kHz (Fixed)
 - VTarget = 3.290V
 - ConfigTargetSettings() start
 - ConfigTargetSettings() end - Took 239us
 - InitTarget() start
 - SWD selected. Executing JTAG -> SWD switching sequence.
 - InitTarget() end - Took 31.9ms
 - Found SW-DP with ID 0x6BA02477
 - DPv0 detected
 - CoreSight SoC-400 or earlier
 - AP map detection skipped. Manually configured AP map found.
 - AP[0]: AHB-AP (IDR: Not set)
 - AP[1]: MEM-AP (IDR: Not set)
 - AP[0]: Core found
 - AP[0]: AHB-AP ROM base: 0xE00FE000
 - CPUID register: 0x410FD214. Implementer code: 0x41 (ARM)
 - Feature set: Mainline
 - Cache: No cache

 - Found Cortex-M33 r0p4, Little endian.
 - Cortex-M (ARMv8-M and later): The connected J-Link (S/N 440333609) uses an old firmware module that does not handle I/D-cache correctly. Proper debugging functionality cannot be guaranteed if cache is enabled
 - FPUnit: 8 code (BP) slots and 0 literal slots
 - Security extension: implemented
 - Secure debug: enabled
 - CoreSight components:
 - ROMTbl[0] @ E00FE000
 - [0][0]: E00FF000 CID B105100D PID 000BB4C9 ROM Table
 - ROMTbl[1] @ E00FF000
 - [1][0]: E000E000 CID B105900D PID 000BBD21 DEVARCH 47702A04 DEVTYPE 00 Cortex-M33
 - [1][1]: E0001000 CID B105900D PID 000BBD21 DEVARCH 47701A02 DEVTYPE 00 DWT
 - [1][2]: E0002000 CID B105900D PID 000BBD21 DEVARCH 47701A03 DEVTYPE 00 FPB
 - [1][3]: E0000000 CID B105900D PID 000BBD21 DEVARCH 47701A01 DEVTYPE 43 ITM
 - [1][5]: E0041000 CID B105900D PID 002BBD21 DEVARCH 47724A13 DEVTYPE 13 ETM
 - [1][6]: E0042000 CID B105900D PID 000BBD21 DEVARCH 47701A14 DEVTYPE 14 CSS600-CTI
 - [0][1]: E0040000 CID B105900D PID 000BBD21 DEVARCH 00000000 DEVTYPE 11 TPIU
 - [0][2]: E00FD000 CID B105F00D PID 001BB101 TSG
```

What is interesting here is that 1900kHz is the communication rate and secure debug is enabled. I assume this means that just the hardware has such a part. In Simplicity Studio I can see the debug is unlocked so normal operations should work. In their IDE everything is set to 1900khz comm speed. I have tried that on the probe-rs and still the same result.

Interestingly, even though the terminal output says erasing and programming 'worked' it did not as the blinky program was not overwritten and continued to blink (rust project did not have said blinking code). 

@jaxter184 did you have any sucess/updates? @Yatekii I have tried the speed change as you requested. Any other ideas?

wtodd1, 2025-07-05:
I fixed the full chip erase as well. Now all the core functionality works, including reset, chip erase, download, and debug. Tested on the Seeed Studio XIAO MG24 board, using the built-in CMSIS-DAP debug probe.

corecode, 2025-07-05:
@wtodd1 where are those flash requirements documented?  would this apply to other gecko chips by silabs as well?

wtodd1, 2025-07-08:
@corecode The flash commands are from "EFR32xG24 Wireless SoC Reference Manual". It looks like the procedure is similar for other series 2 gecko chips, but I don't have any to test at the moment.

closed 2025-07-10 by jaxter184