probe-rs/probe-rs #1723: Linking with `cc` failed error while trying to install

adrien3d, 2023-08-17:
**Describe the bug**
Impossible to install probe-run due to probe-rs compilation error: linking with `cc` failed error

**To Reproduce**
`$ cargo install probe-rs --features cli`

**Expected behavior**
Having probe-run installed

**Stacktrace**
```
 = note: ld: warning: ignoring file /opt/homebrew/Cellar/libusb/1.0.26/lib/libusb-1.0.dylib, building for macOS-x86_64 but attempting to link with file built for macOS-arm64
          Undefined symbols for architecture x86_64:
            "_libusb_bulk_transfer", referenced from:
                probe_rs::probe::espusbjtag::protocol::ProtocolHandler::send_buffer::ha3bc8dd387ca4552 in libprobe_rs-c6f57a622b9d1bab.rlib(probe_rs-c6f57a622b9d1bab.probe_rs.3cb7a3f60b6fb250-cgu.7.rcgu.o)
                probe_rs::probe::espusbjtag::protocol::ProtocolHandler::receive_buffer::h614740c2ddc3a64e in libprobe_rs-c6f57a622b9d1bab.rlib(probe_rs-c6f57a622b9d1bab.probe_rs.3cb7a3f60b6fb250-cgu.7.rcgu.o)
                _$LT$probe_rs..probe..stlink..usb_interface..StLinkUsbDevice$u20$as$u20$probe_rs..probe..stlink..usb_interface..StLinkUsb$GT$::write::h30704cbd1debdc99 in libprobe_rs-c6f57a622b9d1bab.rlib(probe_rs-c6f57a622b9d1bab.probe_rs.3cb7a3f60b6fb250-cgu.7.rcgu.o)
                _$LT$probe_rs..probe..stlink..usb_interface..StLinkUsbDevice$u20$as$u20$probe_rs..probe..stlink..usb_interface..StLinkUsb$GT$::read_swo::h1a9f73557b04d6f6 in libprobe_rs-c6f57a622b9d1bab.rlib(probe_rs-c6f57a622b9d1bab.probe_rs.3cb7a3f60b6fb250-cgu.7.rcgu.o)
                probe_rs::probe::cmsisdap::commands::CmsisDapDevice::drain::h33c06acce7103112 in libprobe_rs-c6f57a622b9d1bab.rlib(probe_rs-c6f57a622b9d1bab.probe_rs.3cb7a3f60b6fb250-cgu.14.rcgu.o)
                probe_rs::probe::cmsisdap::commands::CmsisDapDevice::find_packet_size::h4ce5483b626ef2bd in libprobe_rs-c6f57a622b9d1bab.rlib(probe_rs-c6f57a622b9d1bab.probe_rs.3cb7a3f60b6fb250-cgu.14.rcgu.o)
                probe_rs::probe::cmsisdap::commands::CmsisDapDevice::read_swo_stream::h6c3dc4752e251648 in libprobe_rs-c6f57a622b9d1bab.rlib(probe_rs-c6f57a622b9d1bab.probe_rs.3cb7a3f60b6fb250-cgu.14.rcgu.o)
                ...
            "_libusb_claim_interface", referenced from:
                rusb::device_handle::DeviceHandle$LT$T$GT$::claim_interface::h224d6051429624ce in probe_rs-9a159bd93a3cc2d6.probe_rs.bd4b82fd42884923-cgu.12.rcgu.o
                probe_rs::probe::espusbjtag::protocol::ProtocolHandler::new_from_selector::hdeceb129898361a2 in probe_rs-9a159bd93a3cc2d6.probe_rs.bd4b82fd42884923-cgu.2.rcgu.o
                probe_rs::probe::espusbjtag::protocol::ProtocolHandler::new_from_selector::hf743d260116038ed in probe_rs-9a159bd93a3cc2d6.probe_rs.bd4b82fd42884923-cgu.2.rcgu.o
                probe_rs::probe::cmsisdap::tools::open_v2_device::hf7fe9c29f20ed519 in libprobe_rs-c6f57a622b9d1bab.rlib(probe_rs-c6f57a622b9d1bab.probe_rs.3cb7a3f60b6fb250-cgu.6.rcgu.o)
                rusb::device_handle::DeviceHandle$LT$T$GT$::claim_interface::h9e2c345d24f18baf in libjaylink-3999bb48f4f7fbb1.rlib(jaylink-3999bb48f4f7fbb1.jaylink.1b3c1867b86ffb9-cgu.5.rcgu.o)
            "_libusb_close", referenced from:
                core::ptr::drop_in_place$LT$probe_rs..probe..jlink..JLink$GT$::hb5644cdcccba19ba in probe_rs-9a159bd93a3cc2d6.probe_rs.bd4b82fd42884923-cgu.12.rcgu.o
                core::ptr::drop_in_place$LT$rusb..device_handle..DeviceHandle$LT$rusb..context..Context$GT$$GT$::h046da3d4903abddf in probe_rs-9a159bd93a3cc2d6.probe_rs.bd4b82fd42884923-cgu.12.rcgu.o
                _$LT$rusb..device_handle..DeviceHandle$LT$T$GT$$u20$as$u20$core..ops..drop..Drop$GT$::drop::h19097bee8a20f18a in probe_rs-9a159bd93a3cc2d6.probe_rs.bd4b82fd42884923-cgu.12.rcgu.o
                _$LT$rusb..device_handle..DeviceHandle$LT$T$GT$$u20$as$u20$core..ops..drop..Drop$GT$::drop::h251e029e350b23fa in probe_rs-9a159bd93a3cc2d6.probe_rs.bd4b82fd42884923-cgu.12.rcgu.o
                _$LT$rusb..device_handle..DeviceHandle$LT$T$GT$$u20$as$u20$core..ops..drop..Drop$GT$::drop::h0f27b1abacf67063 in libprobe_rs-c6f57a622b9d1bab.rlib(probe_rs-c6f57a622b9d1bab.probe_rs.3cb7a3f60b6fb250-cgu.3.rcgu.o)
                _$LT$rusb..device_handle..DeviceHandle$LT$T$GT$$u20$as$u20$core..ops..drop..Drop$GT$::drop::h123bf98b9f227c5e in libprobe_rs-c6f57a622b9d1bab.rlib(probe_rs-c6f57a622b9d1bab.probe_rs.3cb7a3f60b6fb250-cgu.3.rcgu.o)
                _$LT$rusb..device_handle..DeviceHandle$LT$T$GT$$u20$as$u20$core..ops..drop..Drop$GT$::drop::h138c3de003cc7d43 in libjaylink-3999bb48f4f7fbb1.rlib(jaylink-3999bb48f4f7fbb1.jaylink.1b3c1867b86ffb9-cgu.5.rcgu.o)
                ...
            "_libusb_control_transfer", referenced from:
                rusb::device_handle::DeviceHandle$LT$T$GT$::read_control::h7fe038c24e8a48c3 in probe_rs-9a159bd93a3cc2d6.probe_rs.bd4b82fd42884923-cgu.12.rcgu.o
                rusb::device_handle::DeviceHandle$LT$T$GT$::read_languages::h54bd6b12b96a136f in probe_rs-9a159bd93a3cc2d6.probe_rs.bd4b82fd42884923-cgu.12.rcgu.o
                rusb::device_handle::DeviceHandle$LT$T$GT$::read_serial_number_string::h094699383d032161 in probe_rs-9a159bd93a3cc2d6.probe_rs.bd4b82fd42884923-cgu.12.rcgu.o
                rusb::device_handle::DeviceHandle$LT$T$GT$::read_languages::hd597e58e200003ec in libprobe_rs-c6f57a622b9d1bab.rlib(probe_rs-c6f57a622b9d1bab.probe_rs.3cb7a3f60b6fb250-cgu.3.rcgu.o)
                rusb::device_handle::DeviceHandle$LT$T$GT$::read_string_descriptor::h979d941486940c39 (.llvm.3176623628611668140) in libprobe_rs-c6f57a622b9d1bab.rlib(probe_rs-c6f57a622b9d1bab.probe_rs.3cb7a3f60b6fb250-cgu.3.rcgu.o)
            "_libusb_exit", referenced from:
                core::ptr::drop_in_place$LT$alloc..sync..ArcInner$LT$rusb..context..ContextInner$GT$$GT$::h387d8a9ae9d7980b in librusb-b2c163eefe2c23bc.rlib(rusb-b2c163eefe2c23bc.rusb.b06f8be4e1bbc4ec-cgu.5.rcgu.o)
                _$LT$rusb..context..ContextInner$u20$as$u20$core..ops..drop..Drop$GT$::drop::he320a7c2cff0fb68 in librusb-b2c163eefe2c23bc.rlib(rusb-b2c163eefe2c23bc.rusb.b06f8be4e1bbc4ec-cgu.5.rcgu.o)
            "_libusb_free_config_descriptor", referenced from:
                _$LT$rusb..config_descriptor..ConfigDescriptor$u20$as$u20$core..ops..drop..Drop$GT$::drop::h93f8a8154741b4e0 in librusb-b2c163eefe2c23bc.rlib(rusb-b2c163eefe2c23bc.rusb.b06f8be4e1bbc4ec-cgu.3.rcgu.o)
            "_libusb_free_device_list", referenced from:
                core::ptr::drop_in_place$LT$rusb..device_list..DeviceList$LT$rusb..context..Context$GT$$GT$::hf8371a29d8d59662 in probe_rs-9a159bd93a3cc2d6.probe_rs.bd4b82fd42884923-cgu.15.rcgu.o
                probe_rs::probe::cmsisdap::tools::open_device_from_selector::h436359fd0d3f476e in probe_rs-9a159bd93a3cc2d6.probe_rs.bd4b82fd42884923-cgu.15.rcgu.o
                probe_rs::probe::cmsisdap::tools::open_device_from_selector::hcfe72ef7d1a63e9b in probe_rs-9a159bd93a3cc2d6.probe_rs.bd4b82fd42884923-cgu.15.rcgu.o
                _$LT$probe_rs..probe..stlink..StLink$LT$probe_rs..probe..stlink..usb_interface..StLinkUsbDevice$GT$$u20$as$u20$probe_rs..probe..DebugProbe$GT$::new_from_selector::h0a926276b47d5cc5 in probe_rs-9a159bd93a3cc2d6.probe_rs.bd4b82fd42884923-cgu.2.rcgu.o
                _$LT$probe_rs..probe..stlink..StLink$LT$probe_rs..probe..stlink..usb_interface..StLinkUsbDevice$GT$$u20$as$u20$probe_rs..probe..DebugProbe$GT$::new_from_selector::h170b78a27ada9369 in probe_rs-9a159bd93a3cc2d6.probe_rs.bd4b82fd42884923-cgu.2.rcgu.o
                core::ptr::drop_in_place$LT$rusb..device_list..DeviceList$LT$rusb..context..Context$GT$$GT$::hf8371a29d8d59662 in probe_rs-9a159bd93a3cc2d6.probe_rs.bd4b82fd42884923-cgu.2.rcgu.o
                probe_rs::probe::espusbjtag::protocol::ProtocolHandler::new_from_selector::hdeceb129898361a2 in probe_rs-9a159bd93a3cc2d6.probe_rs.bd4b82fd42884923-cgu.2.rcgu.o
                ...
            "_libusb_get_active_config_descriptor", referenced from:
                _$LT$probe_rs..probe..stlink..StLink$LT$probe_rs..probe..stlink..usb_interface..StLinkUsbDevice$GT$$u20$as$u20$probe_rs..probe..DebugProbe$GT$::new_from_selector::h0a926276b47d5cc5 in probe_rs-9a159bd93a3cc2d6.probe_rs.bd4b82fd42884923-cgu.2.rcgu.o
                _$LT$probe_rs..probe..stlink..StLink$LT$probe_rs..probe..stlink..usb_interface..StLinkUsbDevice$GT$$u20$as$u20$probe_rs..probe..DebugProbe$GT$::new_from_selector::h170b78a27ada9369 in probe_rs-9a159bd93a3cc2d6.probe_rs.bd4b82fd42884923-cgu.2.rcgu.o
[... cut ...]
          ld: symbol(s) not found for architecture x86_64
          clang: error: linker command failed with exit code 1 (use -v to see invocation)
          

error: could not compile `probe-rs` (bin "probe-rs") due to previous error
error: failed to compile `probe-rs v0.20.0`, intermediate artifacts can be found at `/var/folders/ww/lc242p6n5b96bbzp12rwffk40000gn/T/cargo-installQ4YOuf`
```

**Desktop (please complete the following information):**

- macOS Ventura 13.5
- cargo 1.71.1 (7f1d04c00 2023-07-29)
- rustc 1.71.1 (eb26296b5 2023-08-03)

Yatekii, 2023-08-17:
Something is seriously off. Your linker looks up x86 symbols when you apparently are on an ARM Mac (M1). Additionally, it tries to link against a brew libusb which should not be required at all. Something with your brew seems off, but this is definitely not a probe-rs issue. It builds just fine with the exact same setup on my end :)

Yatekii, 2023-08-17:
(I just closed this because it's not a probe-rs issue - feel free to comment here and I am glad to help - also on Matrix it might be a bit easier to help)

closed 2023-08-17 by Yatekii

adrien3d, 2023-08-18:
Just for the record, I think it was due to a past installation of libusb, uninstalling it and reinstalling it fixed the issue.