probe-rs/probe-rs #1724: cargo embed RTT invalid chars and incomplete output

avlec, 2023-08-19:
**Describe the bug**
While playing around with the example code for the RP2040 https://github.com/embassy-rs/embassy (see examples/rp directory) the wifi tcp server example. I have noticed there's some issues that are popping up from using the probe-rs `cargo embed` command to connect to the debug port with RTT enabled. When I run the project normally with cargo run with the custom runner it outputs without the weird symbols below. I have an example below from trying to use a modified version of the previously mentioned example to use reqwless as a HTTP client, the error isn't related to this problem, see the output I have in the collapsible regions below to see what I mean. It could be that I am doing something wrong, feel free to ask for more information.

<details>
<summary>
cargo embed (way less output and broken characters)
</summary>

```
19:41:07.514 Q�r��hb�lb�iubI���B�b��lb�t�b��b�D;bJ�w~�CbԈEb�wtbM(Whndarm_armr addr:
0�18003000, cr4_idx: 0���vbMW000000.001 @�<xbMFWRTE (SDIO-CDC) 7.95.62 (b03806e
CY) on BCM43439 r5 @ 37.4/81.6/81.6MHz��q{bM1W000000.003 sdpcmdcdc0: Broadcom
SDPCMD CDC driver���}bM>W000000.008 reclaim section!0: Ret�rned 46156 bytes to
the heap��߀bM*W000000.012 wlc_bmac_info_ioit: ho�t_enab 1��H�bMSW000000.037 wl0:
wlc_ampdu_tx_set: AGG Mode = MAC+Ucod� txmaxpkts 64 txmaxpkts_agg 0���bMTW000000.049
wl0: wlc_channels_com�it: no valid channel for "#n" nbands 1 bandlocked
0���bMQW000000.053 wl0: Broadcom BCM43439 802.11 Wireless Controller 7.95.62 (b03806e
CY)��o�bM&W000000.056 TCAM: 256 used: 77 exceed:0�雐bM>W000000.058 recl�im section 1:
Returned 86104 bytes to the heap�ꐞb���b���b뼱b��L bMWclmload_status�n bݎo bM
19:41:09.664 Wbus:txglomP@OhH}�� bMWapstaPO"HW�/� bMWcur_etheraddr�؝� bPOH:�:�"e�
bMWcountryPOH]XX����XXx�8�"bM
19:41:09.778 Wbus:txglomP@OhH}���$bMWampdu_ba_wsizeP�OhHu��H&bM
19:41:09.986 Wampdu_mpduP@OhHuݐ�'bMWbsscfg:event_msgsP�OhH}
�������������������������)bMW000000.763 wl0: wl_open��4�)b�U9-bݓ:-bMWpm2_sleep_retP�
OhH�u��]-bM
19:41:10.452 Wbcn_li_bcnP@OhHu��-bMWbcn_li_dtimP�OhHu�-bMWassoc_listenP�OhH
19:41:10.496 uS'�-b��WH�:�
19:41:10.496 "exU~�-b��-bMWampdu_ba_wsizeP�OhHu��
19:41:10.545 .bP�W�����0���VH�:�"eP�tP�/P�zFF7P�kF?F�KH�:�
19:41:10.545 "eFF0OiHFFIK�FFXFOm�w��U.bMWbsscfg:sup_wpaP�OhH}
~�^y.bMWbsscfg:sup_wpa2_eapverP�OhH}����p�,�.bMWbsscfg:sup_wpa_tmoP�OhH}�|�g�cb�Psn�?
An<�wl0�POH_��cb�Psow;l0wl0~P/OtHCool-Fi|�F�cbJWJkV��cbJW��cbMWampdu_ba_wsizeP�
OhHu�Y�cbMWbsscfg:sup_wpaP�OhH}~ݨdbMWbsscfg:sup_wpa2_eapverP�OhH}
����p�j5dbMWbsscfg:sup_wpa_tmoP�OhH}�|�z{�b�XPsW~K1;���ycwl0�POKH]Cool-
Fi����$0Hl2`!$0��a�0��Hx�P�}�;��b�Ps~�?An<�wl0�POH_�M��bMW000007.745 wl0:
link up (wl0)��Z��b�YPsX~�{wl0~P/O�tH����$20H`lFsЈ!@������`����`l���?
An<���P��@�'�LBC^b2/�����HP�J ;I7*J |�ǔb�Ps~��;An<�wl0�PO�H]1�����$20H`lFsЌ!
@������`����`l���?An<���P��@�'�LBC^b2/�����HP�J ;I7*J |�Քb�PS~�;An<�wl0�POH]0��
��ܔb�/Ps.n�?�'۬ewl0�POH_��ޔb��i�b�Ps~�?An<�wl0�POH_��p�b�Ps�;An<�wl0�POH]Cool-
Fi��ބ�bW���bXj��bYχ�bZ6��b\d��bpanicked at src/main.rs:146:14:
19:41:17.335 called `Result::unwrap()` on an `Err` value: Network(Other)��
```

</details>

<details>
<summary>
cargo run
</summary>

```
0.001921 INFO Hello World!
└─ src/main.rs:51
0.289081 DEBUG waiting for clock...
└─ /home/avlec/rp2040/embassy/cyw43/src/fmt.rs:127
0.290013 DEBUG clock ok
└─ /home/avlec/rp2040/embassy/cyw43/src/fmt.rs:127
0.292322 DEBUG chip ID: 43439
└─ /home/avlec/rp2040/embassy/cyw43/src/fmt.rs:127
0.303088 DEBUG loading fw
└─ /home/avlec/rp2040/embassy/cyw43/src/fmt.rs:127
1.928603 DEBUG loading nvram
└─ /home/avlec/rp2040/embassy/cyw43/src/fmt.rs:127
1.935383 DEBUG starting up core...
└─ /home/avlec/rp2040/embassy/cyw43/src/fmt.rs:127
1.961724 DEBUG waiting for wifi init...
└─ /home/avlec/rp2040/embassy/cyw43/src/fmt.rs:127
2.046840 DEBUG shared_addr 000177ac
└─ /home/avlec/rp2040/embassy/cyw43/src/fmt.rs:127
2.048843 DEBUG wifi init done
[... 149 lines cut ...]

</details>

**To Reproduce**
Steps to reproduce the behavior:

1. update the runner to use probe-rs
```toml
[target.'cfg(all(target_arch = "arm", target_os = "none"))']
runner = "probe-rs run --chip RP2040"
```
2. enable rtt in the Embed.toml
3. run `cargo embed` which flashes the program to the device and attaches to it

**Expected behavior**
Expect to see the same characters that are displayed when running via `cargo run`

**Desktop (please complete the following information):**

- Device Raspberry Pi Pico WH (2022)
- Flashing with Raspberry Pi Debug Probe
- Arch Linux
- rust versions
  - rustc 1.73.0-nightly (076887268 2023-08-17)
  - cargo 1.73.0-nightly (7c3904d6c 2023-08-14)

Yatekii, 2023-09-08:
You need to tell cargo-embed to use defmt for cahnnel 0 via the config :)

closed 2023-09-08 by Yatekii

avlec, 2023-09-18:
Yes, this worked. I had to explicitly specify the up/down numbers
```toml
[default.rtt]
enabled = true
channels = [
    { up = 0, down = 0, name = "debug", format = "Defmt" }
]
```