probe-rs/probe-rs #1729: Add support for Texas Instruments CC13xx/CC26xx devices

seanmlyons22, 2023-08-25:
**Describe the solution you'd like**
This issue covers adding support of the CC13xx/CC26xx family of devices to `probe-rs`. It would greatly improve their user experience in the rust ecosystem by adding support for common tools such as `cargo run`, `defmt`, and others.

This will build heavily on the work done by @cstrahan in this pull request that [added JTAG support][4].

**Describe alternatives you've considered**
#### Debug Access
These devices have a fairly contrived connection sequence for the following reasons:

1. They operate in cJTAG (two pin) mode out of boot, and require special sequencing to switch to four pin mode. Refer to the [Technical Reference Manual, Section 6 for more details][1]
2. The the CPU DAP is not part of the scan chain out of boot, and requires interaction with the TI's JTAG controller called icepick to be included in the scan chain. Its also important to note that each time test logic reset (TLR) is entered, the CPU DAP is dropped from the scan chain. Refer to the [ICEPICK documentation][3] for more details.

This is possible by hooking into [DebugPortSetup](https://open-cmsis-pack.github.io/Open-CMSIS-Pack-Spec/main/html/debug_description.html#debugPortSetup) and implementing the steps above.

#### Flash Support
Flashing is another consideration since these devices do not have a CMSIS pack and while their flash loader is proprietary.

So, in order to get flash support, a new flash algorithm must be implemented. Some devices in the sub-family will need a different flash loader. For example the CC13x4/CC26x4 devices have CCFG at a different flash region. See the [CC13x4/CC26x4 TRM section 3][2] for more details.

**Additional context**

Within the CC26xx/CC26xx device family, there are multiple types of cores and flash memory layouts, but their debug subsystem is identical. Likely this effort should be split into multiple pull-requests. I am planning to start with CC13x2/CC26x2.

[1]: https://www.ti.com/lit/ug/swcu185f/swcu185f.pdf
[2]: https://www.ti.com/lit/ug/swcu194/swcu194.pdf
[3]: https://software-dl.ti.com/ccs/esd/documents/xdsdebugprobes/emu_icepick.html
[4]: https://github.com/probe-rs/probe-rs/pull/1462

seanmlyons22, 2023-08-25:
I am working on this now. So far, I have the debug access part working with some hacks, but haven't started yet on the flash support yet.

With said hacks, I can connect and debug an existing program that is flashed via openOCD.

Before starting flash support I think I will need to tackle https://github.com/probe-rs/probe-rs/issues/1682 first to make my connect solution less hacky, because at this point I have commented out @cstrahan 's scan implementation and am providing the vector of `irlens` myself in a hardcoded manner like this:

```rust
// TODO: allow user to specify expected IR lengths
// let chain = self.jtag_scan(None)?; --> this will call test logic reset which disconnects CPU DAP.
let chain = JtagChain{ irlens: vec![4, 6] };   // This is based on https://software-dl.ti.com/ccs/esd/documents/xdsdebugprobes/files/Router_Scan_Sequence.pdf
let ir_lengths = chain.irlens.iter().map(|len| *len as u8).collect();
self.send_jtag_configure(JtagConfigureRequest::new(ir_lengths)?)?;
```

Yatekii, 2023-08-25:
Sounds like a great plan! Thank you very much for working on this!

PR #1771 (2024-05-02): Add cc13xx_cc26xx support within probe-rs
 * Doesn't include flash support
 * Includes cJTAG to JTAG switch logic
 * Includes ICEPICK setup logic
 * Define JTAG scan chain in CC13xxCC26xx target

Resolves: #1729
closed 2024-05-02 by bugadani