probe-rs/probe-rs #1748: RTT/defmt broken since #1738

jannic, 2023-09-10:
**Describe the bug**
On current master (commit 3b55f404), defmt logging doesn't work. `git bisect` points to commit `f77360e2` ("rtt: Allow targets to constrain the ranges of RTT scanning (#1738)"). Reverting that commit fixes the issue.

**To Reproduce**
Steps to reproduce the behavior:

1. Install current probe-rs binary (`cargo install --locked --path . --features cli` from commit 3b55f404)
2. Run `probe-rs run --chip RP2040 --protocol swd target/thumbv6m-none-eabi/release/blink` where the installed binary is compiled from the unmodified template from https://github.com/rp-rs/rp2040-project-template/
3. Receive the following output:

```
     Erasing sectors ✔ [00:00:00] [###################################################################################################################################] 8.00 KiB/8.00 KiB @ 43.68 KiB/s (eta 0s )
 Programming pages   ✔ [00:00:00] [###################################################################################################################################] 8.00 KiB/8.00 KiB @ 22.37 KiB/s (eta 0s )    Finished in 0.584s
~~
~
~
~
~
~
```

(The lines with the `~` characters show up with the expected delay between the lines, so it looks like probe-rs correctly detects when new RTT output is available, it just doesn't interpret it correctly.)

**Expected behavior**
Same as above, but with readable output:
```
    Finished release [optimized + debuginfo] target(s) in 0.03s
     Running `probe-rs run --chip RP2040 --protocol swd --log-file /dev/null target/thumbv6m-none-eabi/release/blink`
     Erasing sectors ✔ [00:00:00] [###################################################################################################################################] 8.00 KiB/8.00 KiB @ 46.45 KiB/s (eta 0s )
 Programming pages   ✔ [00:00:00] [###################################################################################################################################] 8.00 KiB/8.00 KiB @ 22.01 KiB/s (eta 0s )    Finished in 0.577s
INFO Program start
└─ src/main.rs:27
INFO on!
└─ src/main.rs:64
INFO off!
└─ src/main.rs:67
INFO on!
└─ src/main.rs:64
INFO off!
└─ src/main.rs:67
```

**Desktop (please complete the following information):**

- Linux (Debian buster)

**Additional context**
The probe used is a stock Raspberry Pi Debug Probe. The device flashed is a Raspberry Pico.
The same issue was initially reported by esden on the matrix channel. From the screenshot posted it looks like in this case, it was on an STM32L552ZETxQ. The scrambled output looked different (several garbage characters instead of just `~`).

9names, 2023-09-10:
The new code is working fine for me with programs using RTT, but defmt is broken as above.
it does seem look like this breaks defmt detection or defmt feature selection somehow.

jannic, 2023-09-10:
I think I found the cause:

```
-    match Rtt::attach_region(core, memory_map, &rtt_header_address) {
+    match Rtt::attach_region(core, &[], &rtt_header_address) {
```

While `attach_region` in this case doesn't need the memory_map to find the correct region (`rtt_header_address` contains the necessary information), it still uses the memory map internally in `read_c_string` when reading the channel name. And the channel name being equal to defmt is the signal to activate defmt support.

PR #1749 (2023-09-10): Fix defmt decoding
This reverts parts of f77360e2 without removing the new feature. This causes some redundancy in the passed values, though. (memory_map and rtt_scan_regions contain similar information).

Closes #1748
closed 2023-09-10 by Yatekii