probe-rs/probe-rs #1757: Supporting CMSIS-DAP DAP_PACKET_COUNT to enhance speed.

zh522130, 2023-09-14:
**Is your feature request related to a problem? Please describe.**
Using the CMSIS-DAP V2 to download a program of the same size, Keil has a faster download speed. Keil takes about 5.22 seconds, while Probe-rs takes about 8.53 seconds

**Describe the solution you'd like**
By capturing USB communication information through Bus-Hound, I found that Keil makes good use of the DAP_PACKET_COUNT parameter of CMSIS-DAP. During the sending and receiving phases, it will send or receive multiple data packets consecutively. In my tests, Keil sent up to 18 requests in a row when downloading (my DAP_PACKET_COUNT is 32), while Probe-rs operates on a one-request, one-response basis. I compared IAR and Keil in the same way and came to the same conclusion; IAR sends a maximum of two requests in a row. If Probe-rs could adjust the number of consecutive request sends or response receipts by obtaining the DAP_PACKET_COUNT value, the speed would be faster.

**Describe alternatives you've considered**

**Additional context**
[image]

[image]

Yatekii, 2023-09-18:
Interesting! Seems worth looking into :)

cc @adamgreig

adamgreig, 2023-09-18:
Interesting comparison, thank you!

My recollection from last time I looked into this was that it seemed like we should be able to get most of the benefit by just ensuring we put multiple transfers into each `DAP_Transfer` command. The current batch logic queues up all writes until either `flush()` is called, a read command is issued (since we need to return a response to the calling code), or a `DAP_Transfer` packet is filled up. If memory serves we do issue a bunch of writes even during flashing to check on status etc, but I might be wrong here.

The problem then with having multiple packets in-flight is that the higher level code can't handle sending new commands without waiting for a result from the old ones... but in the case where it's only issuing a lot of writes without reads, where currently we queue them until we fill one `DAP_Transfer` and then send it once full, it's conceivable we could immediately start sending another before getting the response back. For HS USB probes I'm not sure how much difference it would make but I can see it helping a bit with FS USB. Certainly if we could get closer to the Keil transfer rates that would be a very nice improvement!

You can see the current batch processing code here: https://github.com/probe-rs/probe-rs/blob/master/probe-rs/src/probe/cmsisdap/mod.rs#L439

I'm not really sure how we could re-architect the bottom layers of the probe code to handle sending a request without waiting for a response, and then processing multiple responses later on -- [`process_batch`](https://github.com/probe-rs/probe-rs/blob/master/probe-rs/src/probe/cmsisdap/mod.rs#L439) there calls [`send_command`](https://github.com/probe-rs/probe-rs/blob/master/probe-rs/src/probe/cmsisdap/commands/mod.rs#L338) which calls `device.write()` and `device.read()`. I guess we could have a special `send_command` that doesn't immediately read, but instead stores some reference to the sent request, and later/at the end of a large transfer, `process_batch` could call some new function to read all pending responses and return results.

One catch here is that none of the probes I was testing at the time supported multiple in-flight transfers - which probe are you using?

zh522130, 2023-09-19:
> Interesting comparison, thank you!
>
> My recollection from last time I looked into this was that it seemed like we should be able to get most of the benefit by just ensuring we put multiple transfers [...]
@adamgreig 
 
Thank you for your answer. I am using a self-made USB-FS CMSIS-DAP V2 and I am a beginner in Rust. I may not fully understand the code you provided, but I will take some time to study it. For comparative testing, I tested another MCU with only 128KBytes of FLASH. I found that on the STM32F743VGT6 (with a FLASH sector size of 128KBytes, total FLASH size of 2MBytes, and I only programmed about 512KBytes), it's the same as the screenshot I sent before, where Keil sends 18 request packets at once, and the actual flash data is 964 bytes.

For another MCU (with a sector size of 512Bytes, total FLASH size of 128KBytes, and a programmed size of about 128KBytes), Keil sends 8 request packets at once, with actual flash data being 402Bytes. Even if I increase the DAP_PACKET_COUNT to 64, the test result is the same. If I reduce the DAP_PACKET_COUNT, I tested the cases where DAP_PACKET_COUNT is 15, 16, 17. When the DAP_PACKET_COUNT is less than or equal to 18, the maximum number of request packets actually sent by the STM32F743VGT6 is equal to the DAP_PACKET_COUNT.

From the data captured by Bus Hound, it appears that Keil sends out all the request packets before any response packets appear, regardless of how many packets are requested. Summarizing the above information, I guess that Keil will calculate the maximum number of request packets that can be sent at once according to the sector size and DAP_PACKET_COUNT, take the smaller of the two, then split the firmware, and after sending a batch of request packets, it will read the response data from the bulk endpoint for verification. If the response does not match the request, it is considered a transmission failure.

[image]

Yatekii, 2023-09-19:
I guess we could also already improve by not flushing automatically when the packet is full but when DAP_PACKET_COUNT packets are full. And then we issue all of them. Should be the same behavior with bigger packets.

zh522130, 2023-09-19:
@Yatekii 

My understanding is that the multiple requests that are sent must be separated, rather than combined into one large data packet. This is because the underlying logic of CMSIS-DAP does not have code to extract individual requests from a large data packet, unless implemented manually. This is also why the request size captured by Bus Hound is 61 bytes, not 64 bytes (USB FS pack size).

zh522130, 2023-09-19:
My English is not very good, and this content has been translated using Chat GPT. If there's anything that is not clearly expressed, please let me know.

rgrr, 2024-03-11:
I do not want to advertise for my own probe, but which probe are you using?  E.g. when doing `probe-rs benchmark --address 0x20000000 --chip nrf52840_xxaa --speed 10000` with the official picoprobe, the max throughput is around 107KByte/s, when using [my probe FW](https://github.com/rgrr/yapicoprobe) max speed is 174KByte/s.  Just to give numbers.

And of course using "DAP_PACKET_COUNT > 1" increases the effect as well.  But it is not the only parameter if you are looking for speed.

zh522130, 2024-03-12:
> I do not want to advertise for my own probe, but which probe are you using? E.g. when doing `probe-rs benchmark --address 0x20000000 --chip nrf52840_xxaa --speed 10000` with the official picoprobe, the max throughput is around 107KByte/s, when using [my probe FW](https://github.com/rgrr/yapicoprobe) max speed is 174KByte/s. Just to give numbers.我不想为我自己的探针做广告，但是您使用的是哪种探针？例如。当使用官方 picoprobe 进行 `probe-rs benchmark --address 0x20000000 --chip nrf52840_xxaa --speed 10000` 时，最大吞吐量约为 107KByte/s，使用我的探针 FW 时最大速度为 174KByte/s。只是为了给出数字。
> 
> And of course using "DAP_PACKET_COUNT > 1" increases the effect as well. But it is not the only parameter if you are looking for speed.当然，使用“DAP_PACKET_COUNT > 1”也会增加效果。但如果您追求速度，它并不是唯一的参数。

I am using a custom probe that I modified from the one available at https://github.com/ataradov/free-dap, to which I added the functionalities I needed. I understand that using PIO can enhance speed, but pioprobe has only implemented SWD, and I am currently unable to implement the JTAG part myself. I used probe-rs benchmark --address 0x20000000 --chip STM32H743VGTx --speed 10000 and obtained the following results:

probe-rs benchmark --address 0x20000000 --chip STM32H743VGTx --speed 10000
Probe: Probe type CMSIS-DAP, debug interface not specified, target chip STM32H743VGTx

Test: Speed 10000, Word size 32bit, Data length 4 bytes, Number of iterations 5
Results: Read: 10461.04 bytes/s Std Dev 1084.36, Write: 553454.21 bytes/s Std Dev 70311.43

Test: Speed 10000, Word size 32bit, Data length 32 bytes, Number of iterations 5
Results: Read: 50189.17 bytes/s Std Dev 10040.47, Write: 38842.05 bytes/s Std Dev 5974.69

Test: Speed 10000, Word size 32bit, Data length 128 bytes, Number of iterations 5
Results: Read: 94418.22 bytes/s Std Dev 4258.57, Write: 88512.52 bytes/s Std Dev 8958.54

Test: Speed 10000, Word size 32bit, Data length 2048 bytes, Number of iterations 5
Results: Read: 129597.29 bytes/s Std Dev 13375.66, Write: 123760.49 bytes/s Std Dev 14233.32

Test: Speed 10000, Word size 32bit, Data length 32768 bytes, Number of iterations 5
Results: Read: 138249.18 bytes/s Std Dev 2835.41, Write: 131775.39 bytes/s Std Dev 1910.84

 the probe-rs benchmark --address 0x20000000 --chip nrf52840_xxaa --speed 10000 operates by reading and writing to RAM only, which is why it achieves higher speeds. When it comes to reading and writing to flash, the main time consumption is due to the erase and write cycles of the flash memory. If a mode that supports DAP_PACKET_COUNT > 1 is available, it could reduce some of the waiting times associated with erasing and writing by allowing data transfer caching to occur simultaneously with the erasing process, thereby enhancing overall speed.

rgrr, 2024-03-12:
Yes, I understand the advantage of DAP_PACKET_COUNT and I also understand that RAM operation more or less tests only USB interface and SWD speed.

Concerning effects of DAP_PACKET_COUNT: I'm looking for a simple benchmark (just like probe-rs benchmark) to make tests with pyocd or openocd.  Any suggestions?

Unfortunately I have no JTAG target in my collection to implement it with my probe FW.

My results:
```
hardy@ntbox:~/.nrfutil/bin$ probe-rs benchmark --address 0x20000000 --chip nrf52840_xxaa --speed 10000
Probe: Probe type CMSIS-DAP, debug interface not specified, target chip nRF52840_xxAA

Test: Speed 10000, Word size 32bit, Data length 4 bytes, Number of iterations 5
Results: Read: 7907.10 bytes/s Std Dev 400.48, Write: 2142253.15 bytes/s Std Dev 398875.39

Test: Speed 10000, Word size 32bit, Data length 32 bytes, Number of iterations 5
Results: Read: 31266.28 bytes/s Std Dev 217.73, Write: 32703.56 bytes/s Std Dev 339.94

Test: Speed 10000, Word size 32bit, Data length 128 bytes, Number of iterations 5
Results: Read: 85303.85 bytes/s Std Dev 4174.34, Write: 92183.64 bytes/s Std Dev 7050.71

Test: Speed 10000, Word size 32bit, Data length 2048 bytes, Number of iterations 5
Results: Read: 178641.02 bytes/s Std Dev 1402.97, Write: 174117.01 bytes/s Std Dev 1723.15

Test: Speed 10000, Word size 32bit, Data length 32768 bytes, Number of iterations 5
Results: Read: 177721.37 bytes/s Std Dev 1302.46, Write: 174133.18 bytes/s Std Dev 1223.52
```

zh522130, 2024-03-12:
> Yes, I understand the advantage of DAP_PACKET_COUNT and I also understand that RAM operation more or less tests only USB interface and SWD speed.
>
> Concerning effects of DAP_PACKET_COUNT: I'm looking [...]
The testing method I am currently considering involves using only Keil to measure the download time for the same firmware under different DAP_PACKET_COUNT settings. I remember conducting such tests before, but I've forgotten the specific data. I'll retest when I have the time and share the results. At the moment, I don't have the time to delve into how to implement JTAG with PIO, but I will give it a try when I can. Alternatively, I might first replace SWD with PIO and retain the conventional IO implementation for the JTAG part for now.

elfmimi, 2024-03-12:
@zh522130 If you are interested in JTAG and SWD with PIO , I suggest you to take a look at [rust-dap](https://github.com/ciniml/rust-dap) .
https://github.com/ciniml/rust-dap/blob/main/rust-dap-rp2040/src/pio/jtag.rs
https://github.com/ciniml/rust-dap/blob/main/rust-dap-rp2040/src/pio/swd.rs

zh522130, 2024-03-12:
> @zh522130 If you are interested in JTAG and SWD with PIO , I suggest you to take a look at [rust-dap](https://github.com/ciniml/rust-dap) .
> https://github.com/ciniml/rust-dap/blob/main/rust-dap-rp2040/src/pio/jtag.rs
> https://github.com/ciniml/rust-dap/blob/main/rust-dap-rp2040/src/pio/swd.rs

Thank you very much for sharing this incredibly valuable open-source project with me. I will take a good look at this repository. It has been a long time since I last used PIO, so I'll need to take some time to re-learn it.

zh522130, 2024-03-14:
@Yatekii @adamgreig I'm not sure if I should close this issue. Today, I planned to test the impact of DAP_PACKET_COUNT on the Keil software. I found that the download speed of probe-rs is almost equivalent to that of Keil, or even a bit faster. I thought I had provided incorrect test data before, so I retested all versions from v0.20.0 to v0.23.0, and the results are as follows:

It seems that the previously slow download speed was not caused by the lack of support for DAP_PACKET_COUNT. There was a noticeable improvement in speed after version 0.22.0. I carefully reviewed the changelog for version 0.22.0 (without looking at the source code) and am unsure where the improvement came from. I tested the download speeds with the Keil software's DAP_PACKET_COUNT set to 1 and 32, under the conditions of choosing to erase without verification, and the results are as follows:
If not erasing or verifying, the test results are as follows:

If probe-rs chooses not to erase, the programming speed is 107KB/s, which shows that reducing the impact of flash programming can improve speed with DAP_PACKET_COUNT.
I apologize for my previous inaccurate testing (not solely testing the impact of DAP_PACKET_COUNT on Keil). Based on current findings, the slow download issue seems to be resolved. Whether DAP_PACKET_COUNT needs to be supported may require further assessment.I'm not sure if I should close this issue. Today, I planned to test the impact of DAP_PACKET_COUNT on the Keil software. I found that the download speed of probe-rs is almost equivalent to that of Keil, or even a bit faster. I thought I had provided incorrect test data before, so I retested all versions from v0.20.0 to v0.23.0, and the results are as follows:

Command： probe-rs download .\STM32H743VIT6.hex  --format hex --chip STM32H743VGTx --protocol swd --speed 10000

V0.20.0
Total Time：24.82s
Erase Speed：71.31KB/S
Program Speed：32.66KiB/S

V0.21.0
Total Time：25.39s
Erase Speed：71.06KB/S
Program Speed：31.71KiB/S

V0.21.1
Total Time：24.781s
Erase Speed：71.19KB/S
Program Speed：32.79KiB/S

V0.22.0
Total Time：16.868s
Erase Speed：71.83KB/S
Program Speed：65.52KiB/S

V0.23.0
Total Time：16.873s
Erase Speed：71.4KB/S
Program Speed：65.65KiB/S

It seems that the previously slow download speed was not caused by the lack of support for DAP_PACKET_COUNT. There was a noticeable improvement in speed after version 0.22.0. I carefully reviewed the changelog for version 0.22.0 (without looking at the source code) and am unsure where the improvement came from. I tested the download speeds with the Keil software's DAP_PACKET_COUNT set to 1 and 32, under the conditions of choosing to erase without verification, and the results are as follows:

**DAP_PACKET_COUNT=32**
Total Time：16.178s

**DAP_PACKET_COUNT=1**
Total Time：16.966s

If not erasing or verifying, the test results are as follows:
**DAP_PACKET_COUNT=32**
Total Time：3.635s
Program Speed：142.165KiB/S

**DAP_PACKET_COUNT=1**
Total Time：4.209s
Program Speed：122.777KiB/S

If probe-rs chooses not to erase,not varify the programming speed is 107KB/s, 

**All my tests were conducted three times, selecting the highest speed value obtained.All my tests were conducted three times, selecting the highest speed value obtained.**

which shows that reducing the impact of flash programming can improve speed with DAP_PACKET_COUNT.

I apologize for my previous inaccurate testing (not solely testing the impact of DAP_PACKET_COUNT on Keil). Based on current findings, the slow download issue seems to be resolved. Whether DAP_PACKET_COUNT needs to be supported may require further assessment.

Yatekii, 2024-03-25:
Thanks a lot for the writeup and the testing!

Seems to me that the packet count still has a positive impact? SO we should aim to increase it?

Also, I am not sure I understand: why is just programming without erasing faster?

zh522130, 2024-03-26:
Based on the results, increasing the packet count does not significantly enhance the speed of programing the flash memory, mainly because the flash speed is too slow. However, if it involves operating RAM, increasing the packet count will improve the speed. 

As for the reason why programming without erasing is faster, I am not sure either. I have repeatedly tested it several times and the outcome is consistent—programming without erasing is indeed faster.

rgrr, 2024-03-26:
Erasing takes time.  Heavily depending on the chip.  For the nRF52840 a single 4K page needs around 85ms, while the nRF52832 requires around 6ms.  Total chip erasure is faster compared to iterating over the memory doing a page erase.  But still it needs some time.