probe-rs/probe-rs #1767: Parse JTAG scan chain from CMSIS-SDF file

seanmlyons22, 2023-09-18:
**Describe the solution you'd like**
[This pull request](https://github.com/probe-rs/probe-rs/pull/1731) added support for specifying the JTAG scan chain in the target file. An enhancement to that would be to update the target-gen process to try and parse scan chain from the System Description File (SDF) that is part of CMSIS pack.

**Describe alternatives you've considered**
 * Try to parse [SDF](https://open-cmsis-pack.github.io/Open-CMSIS-Pack-Spec/main/html/sdf_pg.html#sdf_element_scanchain) and autopopulate the scan chain in the target.yaml if it is present in the cmsis pack.

**Additional context**
 * Need to assess how many (if any) packs have one of these files. TI's MSP432 is a device that has a CMSIS pack and supports JTAG so could be a good candidate to check
* This is where the scan chain would need to be populated [see TODO in generate.rs](https://github.com/probe-rs/probe-rs/blob/master/target-gen/src/generate.rs#L162)

seanmlyons22, 2023-09-18:
@Yatekii, as we discussed.

I will first work on [Adding TI's CC13xx/CC26xx](https://github.com/probe-rs/probe-rs/issues/1729), but maybe can get to it after. We will see :) Making the issue so we don't lose it.

Yatekii, 2023-09-18:
Sounds great, thanks!

seanmlyons22, 2023-09-25:
Didn't mean to close this one.

cstrahan, 2023-10-04:
To toss in a couple interesting cases, the STM32F401xB/C and STM32F401xD/E MCUs have two TAPs, though (IIRC) the CMSIS packs don't declare them. From the manual:

>#### 23.5 STM32F401xB/C and STM32F401xD/E JTAG TAP connection
>The STM32F401xB/C and STM32F401xD/E MCUs integrate two serially connected JTAG TAPs, the boundary scan TAP (IR is 5-bit wide) and the Cortex ®-M4 with FPU TAP (IR is 4-bit wide).
>
>To access the TAP of the Cortex ®-M4 with FPU for debug purposes:
>
>1. First, it is necessary to shift the BYPASS instruction of the boundary scan TAP.
>2. Then, for each IR shift, the scan chain contains 9 bits (=5+4) and the unused TAP instruction must be shifted in using the BYPASS instruction.
>3. For each data shift, the unused TAP, which is in BYPASS mode, adds 1 extra data bit in the data scan chain.
>
> Note: Important: Once Serial-Wire is selected using the dedicated Arm® JTAG sequence, the
boundary scan TAP is automatically disabled (JTMS forced high).